realsenseai/realsense-ros #1926: No image seen in rviz after roslaunch

Abirami-Segar, 2021-06-11:
I am using roswrapper ros wrapper 2.0- build 2.3.0 and librealsense v2.45.0. The versions are compatible. when i comment 

roslaunch realsense2_camera rs_camera.launch

[image]
[image]

I ignored the warnings for two reasons. One of them is that the "control_transfer returned error, index: 768, error: No data available, number: 61" happened only 7 times. and the second reason being that I saw the "Realsense is up" comment once therefore, I proceeded on to rviz to see if I could see the camera streaming the images,

so i launched rviz 
[image]
and I could see no images in rviz.
[image]

Why does the warning say that there is no data available? Is it because of that I could see no image or any screening in rviz?
But when i launch the realsense viewer, it works absolutely fine. I could see all the depth, RGB,.

kindly help.

MartyG-RealSense, 2021-06-11:
Hi @Abirami-Segar   If the **control_transfer returned** warning only occurs a small number of times during launch and stops, or once or twice per minute after launch, then the warning can be safely ignored.

Does RViz provide any images if you change the **Fixed Frame** option from 'map' to **camera_link** please?

Abirami-Segar, 2021-06-11:
Many thanks for the quick response.

I unfortunately have no such option.
[image]

But when i choose add-->by topic-->camera-->compressed, I see an image like this.

[image]

MartyG-RealSense, 2021-06-11:
Is the RealSense Viewer closed when you are using RViz, please?  And which ROS branch are you using (Melodic, Noetic, Foxy, etc)

Abirami-Segar, 2021-06-11:
Yes i havent opened the real sense viewer at all. It is closed and i am using ROS kinetic.

MartyG-RealSense, 2021-06-11:
Thank you for the information.  ROS Kinetic recently reached End of Life (EOL) status, so I would recommend upgrading to at least ROS Melodic if that is possible for your project.

https://discourse.ros.org/t/preparing-for-final-kinetic-sync/19805

Could you check whether the depth topic is publishing data with a rostopic echo instruction:

**rostopic echo /camera/depth/image_rect_raw**

Abirami-Segar, 2021-06-11:
Does that mean kinetic would no longer support intel real sense?

 rostopic echo /camera/depth/image_rect_raw
WARNING: topic [/camera/depth/image_rect_raw] does not appear to be published yet

MartyG-RealSense, 2021-06-11:
The release notes for the ROS1 wrapper state that Kinetic is still supported by the wrapper at the time of writing this.  I believe it is more the case that Kinetic will no longer receive updates and will be frozen in its current state forever.

I cannot see any obvious problems in your launch log, and you stated that the control_transfer returned warnings only appeared a small number of times.  So they are not still generating continuously after launch?

Is there a positive change if you add an initial_reset instruction to the roslaunch instruction to reset the camera at launch:

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**

Abirami-Segar, 2021-06-11:
No, the control_transfer error appears after launch only a few times and then it stops. Then i dont see anything in that particular terminal. 

Yes, i did try "roslaunch realsense2_camera rs_camera.launch initial_reset:=true" and the result is [image]

Abirami-Segar, 2021-06-11:
so upgrading the ros and the ubuntu is the only solution?

MartyG-RealSense, 2021-06-11:
ROS Kinetic cases are very rare on this forum these days, as the minimum ROS1 branch that is used tends to be Melodic.  Having said that, the wrapper should be able to work with Kinetic, as listed in the release notes for wrapper 2.3.0.

The firmware version of the camera is listed in the log as **5.12.14.50** (the latest firmware at the time of writing this).  So that will likely be fine for use with SDK 2.45.0.

What is your Ubuntu version please?  My recollection is that Kinetic is suited to Ubuntu **16.04**

Abirami-Segar, 2021-06-11:
Yes I use ubuntu 16.04

MartyG-RealSense, 2021-06-11:
Since the camera is able to work with the RealSense Viewer, it is not likely to be a problem related to the camera hardware, firmware driver or USB cable.

Did you install the wrapper using Method 1 (librealsense and the wrapper together) or Method 2 (installing librealsense first and then building the ROS wrapper secondly from source code)

Abirami-Segar, 2021-06-14:
I installed using the method 1.

I want to set kimera-vio running on a live camera. I have tried to download and test the kimera vio on a rosbag and it worked absolutely fine. I think there is serious problem with the communication between camera and ros. PLease have a look at the following image.

[image]
[image]
[image]

sbrgrsh, 2021-06-14:
I also face the same problem. Rosbag perfectly works in kimera but I am not able to stream my camera live. My rviz output is blank

MartyG-RealSense, 2021-06-14:
I recall a case of another Kimera-VIO-ROS user who was able to use it with a rosbag but not with live camera data from a D435i.  I pointed them to advice about  trying to set Kimera's **online** mode to True in the launch file.

https://github.com/IntelRealSense/realsense-ros/issues/1713#issuecomment-789675009

Abirami-Segar, 2021-06-14:
Yes kimera online mode is true! even then, i am not able to see the outputfile:///home/abirami/Pictures/17.png
[image]

MartyG-RealSense, 2021-06-14:
@Abirami-Segar  In your earlier comment linked to below, you highlighted that **rostopic echo** was indicating that there was no depth-topic data being published.  You were able to access the color camera by adding the topic to RViz though.

https://github.com/IntelRealSense/realsense-ros/issues/1926#issuecomment-859506608

Is there a live-updating FPS speed for the depth stream after launch when checking it with **rostopic hz**?

**rostopic hz /camera/depth/image_rect_raw**

Abirami-Segar, 2021-06-14:
it says "no new messages"

MartyG-RealSense, 2021-06-14:
Would it be possible to go back to the Method 1 installation instructions for the wrapper, and instead of using this line:

sudo apt-get install ros-$ROS_DISTRO-realsense2-camera

Use the line below:

**sudo apt-get install ros-kinetic-realsense2-camera**

I thought of this because I cannot see a mention of the **rosdistro** line in the images of your log.  I also recall that the Method 1 instructions used to include instructions for defining **$ROS_VER** that are not present in the current instructions.

[image]

Abirami-Segar, 2021-06-15:
No when i command, 
roslaunch realsense2_camera rs_camera.launch
[image]

i could see rosdistro:kinetic

MartyG-RealSense, 2021-06-15:
The images that I could not see **rosdistro** on were from an **rs_camera** roslaunch log.  The image above that includes rosdistro is from a Kimera-VIO roslaunch.

My understanding of Kimera-VIO from its official instruction documentation about use of RealSense D435i is that to begin with, a **roscore** terminal shoud be launched.  After that, roslaunches for Kimera-VIO and the RealSense ROS wrapper are made in separate ROS terminals - the RealSense ROS wrapper first and Kimera-VIO second.  So in total you have 3 terminals open - roscore, the ROS Wrapper and Kimera-VIO.

https://github.com/MIT-SPARK/Kimera-VIO-ROS/blob/master/docs/hardware_setup.md
https://github.com/MIT-SPARK/Kimera-VIO-ROS/blob/master/docs/hardware_setup.md#realsense-d435i-infrared

MartyG-RealSense, 2021-06-21:
Hi @Abirami-Segar  Do you require further assistance with this case, please?  Thanks!

Abirami-Segar, 2021-06-22:
Hi

The problem still; exists. I have a roscore, kimera wrapper all of them in separate terminals and could not understand why I dont see any output in kimera. I am wondering because, the rosbag works fine, I could see the output. but why not with a real camera?

MartyG-RealSense, 2021-06-22:
I went back to the start of the case to review it again.  Essentially the problem is that Kimera-VIO-ROS and RealSense ROS appear to launch correctly but there is no data rendered in RViz.  Is that correct?

 I located a discussion where a person with a D435i (I could not find a D455 case) similarly could play a rosbag but could not render live camera data in RViz.  That person was provided with advice for displaying in RVIz.

https://ask.csdn.net/questions/4460365

The explanation in that advice is somewhat complicated.  There is a simpler and clearer version of the advice in the link below.

https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/78#issuecomment-662513135

MartyG-RealSense, 2021-07-04:
Case closed due to no further comments received.

Abirami-Segar, 2021-07-04:
Hello

I am sorry for the late reply. I am still trying to set up kimera on the
real sense D455. I have also upgraded my ubuntu 16 to ubuntu 18 and tried
it again, but the problem is still not resolved. Could you please kindly
reopen the issue?

Also, does kimera only work on D435i? IF so, I want to change my camera.

Thank you.

Regards
Abirami

MartyG-RealSense, 2021-07-04:
No problem, I have re-opened the case.  Thanks very much for the update.

I have not seen Kimera-VIO-ROS used with a D455 but as far as I am aware there is not a limitation to use only D435i.  If you find that the D455 is not suited to the pre-made D435i calibration for Kimera then you can collect calibration parameters for your camera.  This is described in point 3 of the **Setup** section of the Kimera-VOS-ROS installation documentation.

https://github.com/MIT-SPARK/Kimera-VIO-ROS/blob/master/docs/hardware_setup.md#setup

Abirami-Segar, 2021-07-11:
I upgraded from ubuntu 16 to 18 also, i use a D435i camera. Now everything
works fine.

Thank you

MartyG-RealSense, 2021-07-11:
Great to hear that you achieved a solution, @Abirami-Segar - thanks very much for the update!  I will close this case as everything is working fine for you.  Thanks again   :)

closed 2021-07-11 by MartyG-RealSense

Abirami-Segar, 2021-07-11:
Thanks for your inputs!