realsenseai/realsense-ros #1927: catkin_make error, but realsense-viewer test succesful

DanielSchimpf, 2021-06-11:
Hello,

I am getting an error when using `catkin_make` to build the realsense ros package.
I installed librealsense using these instructions https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md.
The `realsense-viewer` test works and I can access the stream of my D435. 
I am running ubuntu 16 with ros kinetic. 

Thanks for the help!

Daniel

The `catking_make` output:
```
[ 93%] Building CXX object realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/realsense_node_factory.cpp.o
[ 93%] Building CXX object realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/t265_realsense_node.cpp.o
[ 93%] Building CXX object realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o
[ 93%] Built target DeterminePathplan
[ 93%] Built target positionErrorPublisher
[ 93%] Built target AnalyzePointCloud
[ 93%] Built target grind_cell_main_generate_messages
[ 93%] Built target vision_node
[ 97%] Built target ur_robot_driver
[ 97%] Built target TransformService
[ 97%] Built target myworkcell_node
[ 98%] Built target robot_state_helper
[ 98%] Built target dashboard_client
[ 98%] Built target ur_robot_driver_node
[100%] Built target ur_robot_driver_plugin
[100%] Built target calibration_correction
[100%] Built target realsense2_camera_generate_messages_py
[100%] Built target realsense2_camera_generate_messages_eus
Scanning dependencies of target realsense2_camera_generate_messages
[100%] Built target realsense2_camera_generate_messages
/home/dan/catkin_ws/src/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp: In member function ‘void realsense2_camera::T265RealsenseNode::handleWarning()’:
/home/dan/catkin_ws/src/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp:58:5: error: ‘log_to_callback’ is not a member of ‘rs2’
     rs2::log_to_callback( rs2_log_severity::RS2_LOG_SEVERITY_WARN, [&]
     ^
/home/dan/catkin_ws/src/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp:59:41: error: ‘rs2::log_message’ has not been declared
       ( rs2_log_severity severity, rs2::log_message const & msg ) noexcept {
                                         ^
/home/dan/catkin_ws/src/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp: In lambda function:
/home/dan/catkin_ws/src/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp:60:28: error: request for member ‘raw’ in ‘msg’, which is of non-class type ‘const int’
         _T265_fault =  msg.raw();
                            ^
/home/dan/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp: In member function ‘void realsense2_camera::BaseRealSenseNode::registerDynamicOption(ros::NodeHandle&, rs2::options, std::__cxx11::string&)’:
/home/dan/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:560:60: error: ‘RS2_OPTION_SEQUENCE_ID’ was not declared in this scope
                 if (module_name == "stereo_module" && i == RS2_OPTION_SEQUENCE_ID)
                                                            ^
/home/dan/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp: In member function ‘void realsense2_camera::BaseRealSenseNode::registerHDRoptions()’:
/home/dan/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:631:51: error: ‘RS2_OPTION_SEQUENCE_SIZE’ was not declared in this scope
         unsigned int seq_size = sensor.get_option(RS2_OPTION_SEQUENCE_SIZE);
                                                   ^
/home/dan/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:634:31: error: ‘RS2_OPTION_SEQUENCE_ID’ was not declared in this scope
             sensor.set_option(RS2_OPTION_SEQUENCE_ID, seq_id);
                               ^
/home/dan/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:653:27: error: ‘RS2_OPTION_SEQUENCE_ID’ was not declared in this scope
         sensor.set_option(RS2_OPTION_SEQUENCE_ID, 0);   // Set back to default.
                           ^
/home/dan/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:654:27: error: ‘RS2_OPTION_HDR_ENABLED’ was not declared in this scope
         sensor.set_option(RS2_OPTION_HDR_ENABLED, true);
                           ^
/home/dan/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp: In member function ‘void realsense2_camera::BaseRealSenseNode::setupDevice()’:
/home/dan/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:825:25: error: ‘serializable_device’ is not a member of ‘rs2’
             if (_dev.is<rs2::serializable_device>())
                         ^
/home/dan/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:825:25: error: ‘serializable_device’ is not a member of ‘rs2’
/home/dan/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:825:51: error: no matching function for call to ‘rs2::device::is()’
             if (_dev.is<rs2::serializable_device>())
                                                   ^
In file included from /usr/local/include/librealsense2/hpp/rs_record_playback.hpp:8:0,
[... 139 lines cut ...]
                 from /opt/ros/kinetic/include/class_loader/class_loader.h:35,
                 from /opt/ros/kinetic/include/pluginlib/./class_list_macros.hpp:40,
                 from /opt/ros/kinetic/include/pluginlib/class_list_macros.h:35,
                 from /home/dan/catkin_ws/src/realsense-ros/realsense2_camera/src/../include/../include/realsense_node_factory.h:6,
                 from /home/dan/catkin_ws/src/realsense-ros/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/dan/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1:
/usr/include/c++/5/bits/shared_ptr.h:632:5: note: candidate: template<class _Tp, class ... _Args> std::shared_ptr<_Tp1> std::make_shared(_Args&& ...)
     make_shared(_Args&&... __args)
     ^
/usr/include/c++/5/bits/shared_ptr.h:632:5: note:   template argument deduction/substitution failed:
/home/dan/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1242:116: error: template argument 1 is invalid
       _filters.insert(_filters.begin(),NamedFilter("sequence_id_filter", std::make_shared<rs2::sequence_id_filter>()));
                                                                                                                    ^
realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:110: recipe for target 'realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/t265_realsense_node.cpp.o' failed
make[2]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/t265_realsense_node.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:86: recipe for target 'realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o' failed
make[2]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
CMakeFiles/Makefile2:11056: recipe for target 'realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/all' failed
make[1]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed

```

RealSenseSupport, 2021-06-15:
@DanielSchimpf What's the librealsense and ROS wrapper version you installed? Did you refer to the [Releases page](https://github.com/IntelRealSense/realsense-ros/releases) to get the match versions?

RealSenseSupport, 2021-07-05:
@DanielSchimpf Is this still issue for your? Looking forward to your reply. Thanks!

RealSenseSupport, 2021-07-13:
@DanielSchimpf Could you please update? Please note that if we don't hear from you for another 7 days, the issue will be closed. Thanks!

RealSenseSupport, 2021-07-21:
@DanielSchimpf Sorry that we don't hear from you for weeks. Will close this ticket at this point. Please create new one if you still have issues or questions. Thanks!

closed 2021-07-21 by RealSenseSupport