realsenseai/realsense-ros #1930: Librealsense2.47.0 support

weizhuowang, 2021-06-13:
Hi, when can we expect the librealsense 2.47 support? I updated the librealsense through apt-get and looks like there are some compatiability issue using realsense-ros 2.3.0 with librealsense 2.47. How do I downgrade to 2.45?

doronhi, 2021-06-14:
Hi, Please check out [this answer](https://github.com/IntelRealSense/realsense-ros/issues/1925#issuecomment-860157761).
Also, can you please point me to the compatibility issues you found? It may save some time in the updating process.

weizhuowang, 2021-06-14:
Yes! So here is the issue: 
I'm using T265 with D435 together in a project. In my old URDF definition, the joints were properly connected in the TF tree, and the STL model was moving as intended in the Rviz. However, recently after an update, I notice that the old URDF definition no longer works and the STL model stopped moving in the Rviz. Instead, a little T265 odom frame was moving by itself in the Rviz while the TF tree is seperated into two trees. I have not yet found the solution to the problem but I suspects that either librealsense or realsense-ros has changed the way it transforms between T265 to D400 frames. It would be great if you could point me a direction to solve this issue!

weizhuowang, 2021-06-14:
PS. I just downgraded all 4 librealsense library to 2.45.0 and the URDF definition is working again, so the root cause is probably in librealsense 2.47

doronhi, 2021-06-21:
librealsense2 has nothing to do with URDF files or the TF tree. I can't speculate but it seems that other changes or build failures were involved.

RealSenseSupport, 2021-07-06:
@weizhuowang Do you still have issue? We just released one new version https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.1 which can match librealsense 2.47 and 2.48. Please have a try and provide the feedback. Thanks!

RealSenseSupport, 2021-07-21:
@weizhuowang Could you please update? Please note that the ticket will be closed if we don't get feedback in another 7 days.

RealSenseSupport, 2021-07-28:
@weizhuowang Sorry that we don't get feedback from you for weeks, will close the ticket at this point. Please feel free to create another new one if you still have issue or questions. Thanks!

closed 2021-07-28 by RealSenseSupport