realsenseai/realsense-ros #1932: The proper way to get Euler angle readings on D455

mszuyx, 2021-06-16:
Hi RS community,

I wonder if there exists a method to get Euler angle estimation from D455 camera in realsense-ros (or other ros package).
I am aware of the rs-motion example:
https://github.com/IntelRealSense/librealsense/tree/master/examples/motion
This cpp example currently support D435i (there is no **d455.h** file), I ran it with D455 regardlessly, and it works to some extent. I am experiencing issue similar to:
https://github.com/IntelRealSense/librealsense/issues/4915
Sometime the imu model spins like crazy and settle down after a while. Sometime it does not spin at start up but fail to sense any rotation in the z-axis of the optical frame (pointing in the direction of camera, so it doesn't sense roll motion for some reasons, but pitch and yaw seem fine)
I suspect I would need to do a imu calibration to fix this issue, so I am going for the calibration tool:
https://github.com/IntelRealSense/librealsense/tree/development/tools/rs-imu-calibration#rs-imu-calibration-tool
I will let you know how it goes afterward, but if you have other clue, please let me know!

And again, if there exists a method to publish IMU topic directly in ROS for D455, please let me know. If not, I guess I would need to write a roscpp node to do it based on the rs-motion example. (It would be nice if this function can be included in realsense-ros tho)

mszuyx, 2021-06-16:
I completed the IMU calibration using the python tool provided by realsense, here is the outcome:
```
Writing files:
accel_y.txt
gyro_y.txt
[-0.00261551 -0.00085359 -0.00017857]
[1000 1000 1000 1000 1000 1000]
using 6000 measurements.
[[ 1.01483060e+00  3.12750691e-03 -3.47060060e-02]
 [ 2.79278780e-03  1.02460342e+00 -9.50034300e-03]
 [-5.10577905e-03  8.52512066e-06  1.01482781e+00]
 [ 3.04822787e-02  9.21907178e-02  3.49125661e-01]]
residuals: [  9.90475698  15.02521457 201.49861674]
rank: 4
singular: [441.23211728 429.12894094 422.91257365  77.29616243]
norm (raw data  ): 9.637496
norm (fixed data): 9.804580 A good calibration will be near 9.806650
Would you like to write the results to the camera? (Y/N)y
Writing calibration to device.
  Device PID:  0B5C
  Device name:  Intel RealSense D455
  Serial number:  049522250847
  Firmware version:  05.12.14.50
SUCCESS: saved calibration to camera.
Done.
```

- I did notice improved performance in rs-motion, but the random spinning issue still show up from time to time. (now it feels more like a sync issue between the gyro and accel meter) And I have no idea how to fix it.

- I noticed the measure ranges are different in each axis when using rs-motion: for roll (optical frame z-axis) it is +/-90 deg; for pitch (optical frame x-axis) it is +270/-90 deg; for yaw (optical frame y-axis) it is +/-inf. If the device is tilted beyond these ranges, the CAD model in rs-motion visualization will rotate 180 deg in that particular axis. Except for the roll, it will be put in a certain up-right pose with both roll and yaw angles changed. Is this normal/expected?

MartyG-RealSense, 2021-06-16:
Hi @mszuyx   The notes in the script for the **rs-motion** example describe how sensitivity can be reduced by increasing the **alpha** value to shift weighting towards the gyro instead of the sensitive accelerometer.

https://github.com/IntelRealSense/librealsense/blob/master/examples/motion/rs-motion.cpp#L117

In regard to the axes, the Python case about rs-motion that is linked to below may be helpful.  It describes how pitch and roll values can be obtained from the IMU but there is a limitation on **yaw**.

https://github.com/IntelRealSense/librealsense/issues/4391

mszuyx, 2021-06-16:
@MartyG-RealSense Thank you! These links are very educational! Just to be sure, there is no "official" implementation of Euler angle estimation for d400 series camera? Also, it is expected that the gyro and accel meter update rates are different for d455?
[image]

MartyG-RealSense, 2021-06-16:
The **Depth Quality Tool** incorporates a 'plane-fit' algorithm for estimating angle without requiring an IMU.

https://github.com/IntelRealSense/librealsense/issues/4440

https://github.com/IntelRealSense/librealsense/issues/6338#issuecomment-623350905

I am not aware of any differences in gyro and accel rate between D435i and D455.  Generally, the recommended frequencies are 63 or 250 for the accel stream and 200 or 400 for the gyro stream.  In other words, 63 / 200 or 250 / 400.  Intel have published a guide to IMU data in the link below.

https://www.intelrealsense.com/how-to-getting-imu-data-from-d435i-and-t265/

mszuyx, 2021-06-17:
@MartyG-RealSense I have finished the roscpp node. But then I realize something: Why the accel meter reading in the camera/imu (linear_acceleration) indicates the accel meter frame follows left hand rule in d455?

MartyG-RealSense, 2021-06-18:
The links below may be helpful references for understanding the IMU axes and their transform relationship to the other sensors.

https://github.com/IntelRealSense/librealsense/issues/7568#issuecomment-708358084

https://github.com/IntelRealSense/librealsense/issues/7865

https://github.com/IntelRealSense/realsense-ros/issues/1449

mszuyx, 2021-06-18:
Hi @MartyG-RealSense 
Thanks! I can get my own ROS node working by flipping the y-axis (not elegant but works), and I have confirmed that the imu_tool ROS package can be directly adopted to get the quaternion estimation from d455. (http://wiki.ros.org/imu_tools) And because the filter itself used quaternion representation during fusion, you have far less singularity issue compared to the rs-motion example (Euler-based). There is really no need to write an imu fusion node yourself.

MartyG-RealSense, 2021-06-18:
Thanks very much for the detailed feedback about your solution - it should be useful for RealSense ROS community members reading this in future!

I wonder if the discussion in the link below may be relevant to what you are experiencing with the IMU frame.

https://github.com/IntelRealSense/realsense-ros/issues/763

mszuyx, 2021-06-21:
@MartyG-RealSense Thanks! The mentioned discussion is quite helpful, I will check the accel frame in rviz shortly. Meanwhile, I believe all the tf frames follow right-hand-rule, so the fact that the raw accel reading from /camera/accel follows left-hand-rule is just strange. But like I said, the workaround is simple.

MartyG-RealSense, 2021-06-27:
Hi @mszuyx  Do you require further assistance with this case, please?  Thanks!

mszuyx, 2021-06-28:
@MartyG-RealSense I think I have everything I need for now! Thank you!! :  )

closed 2021-06-28 by mszuyx

MartyG-RealSense, 2021-06-29:
Great to hear, @mszuyx - thanks very much for the update!