realsenseai/realsense-ros #1933: Add L515 IMU to URDF definition

shonigmann, 2021-06-16:
As best as I can tell, the _L515.urdf.xacro file doesn't include a definition of the IMU frame. It seems like this [data is available somewhere](https://github.com/IntelRealSense/realsense-ros/issues/1427), so I'm curious why its not in the URDF and am wondering if it can be added.

Thanks!

doronhi, 2021-06-20:
It could be added, based on the [d435i.urdef example](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_description/urdf/_d435i.urdf.xacro) and [the data you linked](https://github.com/IntelRealSense/realsense-ros/issues/1427).
I'll try to get around to it but a PR will be most welcome.

shonigmann, 2021-06-20:
If the data I linked is accurate, I'm happy to add it to the URDF and submit a quick PR

doronhi, 2021-06-21:
Thanks! This issue can be closed now as future discussions regarding the model, including comparison to the [data sheet](https://www.intelrealsense.com/download/7691/) can be done inside the PR.

closed 2021-06-30 by RealSenseSupport