realsenseai/realsense-ros #1934: [ WARN] [1623997211.771164438]: Hardware Notification:USB REC overflow,1.624e+12,Error,Hardware Error

mayo233, 2021-06-18:
[ WARN] [1623997211.771164438]: Hardware Notification:USB REC overflow,1.624e+12,Error,Hardware Error
roslaunch realsense2_camera rs_camera.launchを立ち上げたところ上のようなエラーが出ました。これは何が原因でしょうか

MartyG-RealSense, 2021-06-18:
Translation from Japanese: "When I started, I got the above error.  What is the cause of this?"

Hello @mayo233  Please try a reset of your camera by placing the command **initial_reset:=true** in your roslaunch instruction.

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**

************

こんにちは@ mayo233roslaunch命令にコマンド **initial_reset:= true** を配置して、カメラのリセットを試してください。

MartyG-RealSense, 2021-06-24:
Hello, do you require further assistance with this problem please?  Thank you!

こんにちは、この問題についてさらにサポートが必要ですか？ ありがとうございました！

MartyG-RealSense, 2021-06-30:
Case closed due to no further comments received.

コメントがないため、ケースはクローズされました。

closed 2021-06-30 by MartyG-RealSense