realsenseai/realsense-ros #1935: エラーについて

mayo233, 2021-06-18:
[ WARN] [1623998395.256886791]: Given stream configuration is not supported by the device!  Stream: Depth, Stream Index: 0, Width: -1, Height: -1, FPS: -1
[ WARN] [1623998395.260388435]: Given stream configuration is not supported by the device!  Stream: Color, Stream Index: 0, Width: -1, Height: -1, FPS: -1
[ INFO] [1623998395.260723471]: num_filters: 0
[ INFO] [1623998395.260761839]: SELECTED BASE:Depth, 0
 roslaunch realsense2_camera rs_camera.launchと立ち上げると上のようなエラーが表示されてリアルセンスの接続が切れます。この場合、どうしたら良いでしょうか

MartyG-RealSense, 2021-06-18:
Translation from Japanese: "When you start up, the above error is displayed and the RealSense connection is disconnected.  What should I do in this case?"

Hello @mayo233  It is unusual that this error happened.  The launch process should have automatically used the default stream configuration of the particular RealSense camera model that you own.  Please try using the roslaunch instruction below to see if it works.

**roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30**

**************

こんにちは@ mayo233このエラーが発生したのは珍しいことです。 起動プロセスでは、所有している特定のRealSenseカメラモデルのデフォルトのストリーム構成が自動的に使用されているはずです。 以下のroslaunch命令を使用して、機能するかどうかを確認してください。

**roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30**

mayo233, 2021-06-18:
[camera/realsense2_camera_manager-1] process has died [pid 17746, exit code 127, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/suzukamayo/.ros/log/fb6f53f0-d006-11eb-9c9c-c0b6f9c7d913/camera-realsense2_camera_manager-1.log].
log file: /home/suzukamayo/.ros/log/fb6f53f0-d006-11eb-9c9c-c0b6f9c7d913/camera-realsense2_camera_manager-1*.log

mayo233, 2021-06-18:
[camera/realsense2_camera_manager-1] process has died [pid 17746, exit code 127, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/suzukamayo/.ros/log/fb6f53f0-d006-11eb-9c9c-c0b6f9c7d913/camera-realsense2_camera_manager-1.log].
log file: /home/suzukamayo/.ros/log/fb6f53f0-d006-11eb-9c9c-c0b6f9c7d913/camera-realsense2_camera_manager-1*.log
ddynamic_reconfigureをzip関数で落としてcatkin_makeをした結果、上のようなエラーが表示されるようになりました。どうしたらよいでしょうか

MartyG-RealSense, 2021-06-18:
"As a result of dropping ddynamic_reconfigure with the zip function and doing catkin_make, the above error is displayed.  What should I do?"

If you are having problems with installing ddynamic_reconfigure for Melodic, could you try to install it using the instruction below please?

**sudo apt-get update apt-get install ros-melodic-ddynamic-reconfigure**

******************

Melodicのddynamic_reconfigureのインストールで問題が発生した場合は、以下の手順を使用してインストールしてみてください。

**sudo apt-get update apt-get install ros-melodic-ddynamic-reconfigure**

MartyG-RealSense, 2021-06-25:
Case closed due to question being handled at https://github.com/IntelRealSense/realsense-ros/issues/1936

closed 2021-06-25 by MartyG-RealSense