realsenseai/realsense-ros #1936: エラーについて

mayo233, 2021-06-18:
[camera/realsense2_camera_manager-1] process has died [pid 17746, exit code 127, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/suzukamayo/.ros/log/fb6f53f0-d006-11eb-9c9c-c0b6f9c7d913/camera-realsense2_camera_manager-1.log].
log file: /home/suzukamayo/.ros/log/fb6f53f0-d006-11eb-9c9c-c0b6f9c7d913/camera-realsense2_camera_manager-1*.log
ddynamic_reconfigureをzip関数で落としてcatkin_makeをした結果、上のようなエラーが表示されるようになりました。どうしたらよいでしょうか

mayo233, 2021-06-18:
Setting Dynamic reconfig parameters.
/opt/ros/melodic/lib/nodelet/nodelet: symbol lookup error: /home/suzukamayo/ros_ws/lidar/whill_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN20ddynamic_reconfigure19DDynamicReconfigureC1ERKN3ros10NodeHandleE
と表示されます

MartyG-RealSense, 2021-06-18:
"As a result of dropping ddynamic_reconfigure with the zip function and doing catkin_make, the above error is displayed.  What should I do?"

If you are having problems with installing ddynamic_reconfigure for Melodic, could you try to install it using the instruction below please?

**sudo apt-get update apt-get install ros-melodic-ddynamic-reconfigure**

******************

Melodicのddynamic_reconfigureのインストールで問題が発生した場合は、以下の手順を使用してインストールしてみてください。

**sudo apt-get update apt-get install ros-melodic-ddynamic-reconfigure**

MartyG-RealSense, 2021-06-18:
Regarding this error:

**undefined symbol: _ZN20ddynamic_reconfigure**

It can indicate that more than one version of ddynamic_reconfigure is installed on your computer.

https://github.com/IntelRealSense/realsense-ros/issues/838

[image]

*****************

このエラーについて：

**undefined symbol: _ZN20ddynamic_reconfigure**

これは、ddynamic_reconfigureの複数のバージョンがコンピューターにインストールされていることを示している可能性があります。

MartyG-RealSense, 2021-06-24:
Hello, do you require further assistance with this problem please?  Thank you!

こんにちは、この問題についてさらにサポートが必要ですか？ ありがとうございました！

MartyG-RealSense, 2021-06-30:
Case closed due to no further comments received.

コメントがないため、ケースはクローズされました。

closed 2021-06-30 by MartyG-RealSense