realsenseai/realsense-ros #1937: how to config rs2_timestamp_domain with L515?

arohamirai, 2021-06-18:
i noticed that L515 can config domain time, and i noticed that there are maybe timestamp problems exist, like issue#1902, i am not sure whether it will be happen on me, so i did some test.

1. when i launch rs_camera.launch by  `roslaunch realsense2_camera rs_camera.launch`
wheter or not set global_time_enabled,  SDK always print ` Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)`, it seems  `global_time_enabled` is not used, and driver always use RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME, so where global_time_enabled used and what's it effect？
[image]

my luanch file :just add set global_time_enabled  to true or false
```
<launch>
   <rosparam>
      /camera/motion_module/global_time_enabled: true
      /camera/l500_depth_sensor/global_time_enabled: true
      /camera/rgb_camera/global_time_enabled: true
  </rosparam>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="-1"/>
  <arg name="depth_height"        default="-1"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="true"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="848"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="-1"/>
  <arg name="color_height"        default="-1"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="-1"/>
  <arg name="depth_fps"           default="-1"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="-1"/>
  <arg name="gyro_fps"            default="-1"/>
  <arg name="accel_fps"           default="-1"/>
  <arg name="enable_gyro"         default="false"/>
  <arg name="enable_accel"        default="false"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="false"/>
  <arg name="ordered_pc"                default="false"/>

  <arg name="enable_sync"               default="false"/>
  <arg name="align_depth"               default="false"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="unite_imu_method"          default=""/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>

  <arg name="stereo_module/exposure/1"  default="7500"/>
  <arg name="stereo_module/gain/1"      default="16"/>
  <arg name="stereo_module/exposure/2"  default="1"/>
  <arg name="stereo_module/gain/2"      default="16"/>
  
  

  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="external_manager"         value="$(arg external_manager)"/>
      <arg name="manager"                  value="$(arg manager)"/>
      <arg name="output"                   value="$(arg output)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="usb_port_id"              value="$(arg usb_port_id)"/>
      <arg name="device_type"              value="$(arg device_type)"/>
      <arg name="json_file_path"           value="$(arg json_file_path)"/>

      <arg name="enable_pointcloud"        value="$(arg enable_pointcloud)"/>
      <arg name="pointcloud_texture_stream" value="$(arg pointcloud_texture_stream)"/>
      <arg name="pointcloud_texture_index"  value="$(arg pointcloud_texture_index)"/>
      <arg name="enable_sync"              value="$(arg enable_sync)"/>
      <arg name="align_depth"              value="$(arg align_depth)"/>

      <arg name="fisheye_width"            value="$(arg fisheye_width)"/>
      <arg name="fisheye_height"           value="$(arg fisheye_height)"/>
      <arg name="enable_fisheye"           value="$(arg enable_fisheye)"/>

      <arg name="depth_width"              value="$(arg depth_width)"/>
      <arg name="depth_height"             value="$(arg depth_height)"/>
      <arg name="enable_depth"             value="$(arg enable_depth)"/>

      <arg name="confidence_width"         value="$(arg confidence_width)"/>
      <arg name="confidence_height"        value="$(arg confidence_height)"/>
      <arg name="enable_confidence"        value="$(arg enable_confidence)"/>
      <arg name="confidence_fps"           value="$(arg confidence_fps)"/>

      <arg name="color_width"              value="$(arg color_width)"/>
      <arg name="color_height"             value="$(arg color_height)"/>
      <arg name="enable_color"             value="$(arg enable_color)"/>

      <arg name="infra_width"              value="$(arg infra_width)"/>
      <arg name="infra_height"             value="$(arg infra_height)"/>
      <arg name="enable_infra"             value="$(arg enable_infra)"/>
      <arg name="enable_infra1"            value="$(arg enable_infra1)"/>
      <arg name="enable_infra2"            value="$(arg enable_infra2)"/>
      <arg name="infra_rgb"                value="$(arg infra_rgb)"/>

      <arg name="fisheye_fps"              value="$(arg fisheye_fps)"/>
      <arg name="depth_fps"                value="$(arg depth_fps)"/>
      <arg name="infra_fps"                value="$(arg infra_fps)"/>
      <arg name="color_fps"                value="$(arg color_fps)"/>
      <arg name="gyro_fps"                 value="$(arg gyro_fps)"/>
      <arg name="accel_fps"                value="$(arg accel_fps)"/>
      <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
      <arg name="enable_accel"             value="$(arg enable_accel)"/>

      <arg name="publish_tf"               value="$(arg publish_tf)"/>
      <arg name="tf_publish_rate"          value="$(arg tf_publish_rate)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="stereo_module/exposure/1" value="$(arg stereo_module/exposure/1)"/>
      <arg name="stereo_module/gain/1"     value="$(arg stereo_module/gain/1)"/>
      <arg name="stereo_module/exposure/2" value="$(arg stereo_module/exposure/2)"/>
      <arg name="stereo_module/gain/2"     value="$(arg stereo_module/gain/2)"/>

      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
      <arg name="ordered_pc"               value="$(arg ordered_pc)"/>
      
    </include>
  </group>
</launch>

```

arohamirai, 2021-06-18:
@doronhi

doronhi, 2021-06-20:
Global timestamp is available only with metadata.
The question is, why don't you have metadata? What type of machine do you have? What OS? What method did you use to install librealsense2 and realsense2_camera? Did you install the kernel patches?
Knowing that your current issue is with the metadata, you can also look for answers in the [Librealsense2 issues](https://github.com/IntelRealSense/librealsense/issues)

arohamirai, 2021-06-20:
hi doronhi
 thank you for your  reply, here is my environment:

librealsense:  2.45.0
install realsense2_camera:  yes
os system: ubuntu 16.04.1   with kernel 4.15.0-15-generic
istall the kernel patche: no, because It's not in the standard installation manual。

when i run realsense-viewer, it gives warnings indeed.
[image]

my question is:
1. what system time mean?  the document says it's the os time, but i dont know how it works, since the camera  has inner clock, is it mean the camera  synchronize with the os when first launch and write down the difference between inner clock and os clock, then ervery frame's timestamp equals inner clock  + difference? ie. the inner clock is used as the standard clock. or in the other way, erery frame is jsut get clock from os system, ie, the os system clock is used. 
2. if i don't  install the kernel patches, and just use system clock domain, is it means global_timestamp_enabled param won't affect the frame timestamp type? ie,, there are no timestamp  delay problem？

Thanks in advance。@doronhi

doronhi, 2021-06-21:
What method did you use to install realsense2_camera? kernel patches are installed when installing librealsense2-dkms and appear in Method2 - part1.
1. Without metadata, realsense2_camera uses the first method you describe - using the camera's inner clock + difference. The "System time domain" used by the realsense-viewer application is the OS time when the frame arrived.

2. Yes. If you see the message regarding the lack of metadata it means that the global_timestamp_enabled param will have no effect.

You should also note that if you used Method1 installation, i.e. install ros-$DISTRO-realsense2-camera, it will use a build of librealsense2 that does not depend on the kernel patches (i.e RSUSB backend, instead of v4l backend) and you should have metadata.

bigbellmercy, 2021-06-28:
Hi @doronhi, thanks for the explanation above. For I have the timestamp delay problem with L515 in my laptop PC as #1902, I followed your answers above. After I removed the 'Method 2' installation and did 'Method 1' installation, it still does not resolve the time delay problem of #1902.
As your explanation, after I did 'Method 2' I expected the RSUSB will be used. But, in the terminal, there is a line below as it means it uses 'V4L':
`[ INFO] [1624868624.629226274]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video2 was found.
`
Does the kernel patch remain still even though I remove it? Can I get some solution for this problem?

bigbellmercy, 2021-06-29:
Furthermore, although I succeeded to re-install `librealsense` to use `RSUSB` instead of `V4L` as the method below, `global_time_enabled` still have no effect.

---
FYI, the following method was used to install `librealsense` which uses `RSUSB` instead of `V4L`.

`https://github.com/IntelRealSense/librealsense/tree/master/scripts/libuvc_installation.sh` was executed as explained in `installation_jetson.md`. But the following line was changed:
from:
`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release`
to:
`cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release`
After that, the `physical port` is changed from `...video4linux...` to `2-1-10` in console log and `realsense-viewer`.

bigbellmercy, 2021-06-29:
Hi, the `global_time_enabled` can be enabled by the following method.

https://github.com/IntelRealSense/realsense-ros/issues/1902#issuecomment-870156354

RealSenseSupport, 2021-06-30:
@bigbellmercy thank you for the update and glad to hear that you can now enable the global time.

closed 2021-06-30 by RealSenseSupport