realsenseai/realsense-ros #1938: Nothing showing up in RVIZ with D435

KartmanCanada, 2021-06-18:
I apologize in advance as my ROS install is on an Xavier NX module but I'm writing this on a different device. This will make some log sharing quite awkward. That said, my issue really looks like the issue below and is likely a simple fix. In fact, my limited ROS experience (i.e., I'm learning!) is a major contributor.

RealSense ROS v2.3.0
LibRealSense v2.45.0
D435 firmware 5.12.14.50 (latest)
Platform Xavier NX
OS Ubuntu 18.04

When I run: roslaunch realsense2_camera rs_camera.launch filters:=pointcloud pointcloud_texture_stream:=RS2_STREAM_ANY
* I can open RVIZ and get a pointcloud2 to visualize but only with AxisColor, Intensity, or FlatColor

When I run: roslaunch realsense2_camera rs_camera.launch filters:=pointcloud pointcloud_texture_stream:=RS2_STREAM_COLOR
* I get the following: WARN: No stream match for pointcloud chosen texture process - Color
* Opening RVIZ and trying to visualize with pointcloud2 offers nothing
* no "Position Transformer" or "Color Transformer"

Please help me to understand what I'm missing! Thanks.

MartyG-RealSense, 2021-06-18:
Hi @KartmanCanada   Does the puntcloud-enabled roslaunch instruction in the comment linked to belowxeork for you please?

https://github.com/IntelRealSense/realsense-ros/issues/1886#issuecomment-847770368

**roslaunch realsense2_camera rs_camera.launch align_depth:=true color_width:=848 color_height:=480 color_fps:=30 filters:=pointcloud**

KartmanCanada, 2021-06-18:
Tired the command line above... same result. :(

Below is a log from my original command line from original issue post:
**roslaunch realsense2_camera rs_camera.launch filters:=pointcloud pointcloud_texture_stream:=RS2_STREAM_COLOR**
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Xavier:36927/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
[... 71 lines cut ...]
[ INFO] [1624048586.820656081]: RGB Camera was found.
[ INFO] [1624048586.820890131]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1624048586.821014132]: Add Filter: pointcloud
[ INFO] [1624048586.823444360]: num_filters: 1
[ INFO] [1624048586.823632617]: Setting Dynamic reconfig parameters.
[ INFO] [1624048587.443160485]: Done Setting Dynamic reconfig parameters.
[ INFO] [1624048587.445109653]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1624048587.447118885]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1624048587.447286086]: setupPublishers...
[ INFO] [1624048587.460377806]: Expected frequency for depth = 30.00000
[ INFO] [1624048587.614632760]: Expected frequency for color = 30.00000
[ INFO] [1624048587.736543457]: setupStreams...
[ INFO] [1624048587.761494248]: insert Depth to Stereo Module
[ INFO] [1624048587.761698217]: insert Color to RGB Camera
[ INFO] [1624048587.860937342]: SELECTED BASE:Depth, 0
[ INFO] [1624048587.918695561]: RealSense Node Is Up!
[ WARN] [1624048587.982309667]: 
 18/06 16:36:27,983 WARNING [546695766400] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/06 16:36:28,034 WARNING [546695766400] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/06 16:36:28,569 WARNING [546695766400] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 18/06 16:36:28,619 WARNING [546695766400] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1624048623.404228731]: publishing unordered pointcloud.
[ WARN] [1624048624.006987371]: No stream match for pointcloud chosen texture Process - Color

[image]

MartyG-RealSense, 2021-06-18:
If you use this roslaunch:

**roslaunch realsense2_camera rs_camera.launch filters:=colorizer,pointcloud align_depth:=true pointcloud_texture_stream:=RS2_STREAM_COLOR**

Are you able to change the Color Transformer from 'Intensity' to 'RGB8' 

Are you also able to change the style to **points**

KartmanCanada, 2021-06-18:
Did a cut/paste of your above command line....

1) Yes - I'm able to change style to "points"
2) No - RGB8 is not listed as a Color Transform (in fact, there are no Color or Position Transforms offered in the rviz pulldowns)

Very frustrating... if I run realsense-viewer, I can get the point cloud with RBG overlay. I also tried the realsense2_camera launch file 'rs_rgbd.launch'. This too allows visualization in rviz

MartyG-RealSense, 2021-06-18:
Thanks for your patience.  What is the result if you generate an **ordered pointcloud** by adding the command **ordered_pc:=true** into the roslaunch instruction.

KartmanCanada, 2021-06-18:
no change... except the console output for the wrapper now includes:

**[ INFO] [1624050998.557003460]: publishing ordered pointcloud.**

MartyG-RealSense, 2021-06-18:
There was once a case of the **No stream match for pointcloud chosen texture Process - Color** warning that was found to be caused by an unstable USB connection.  So please next try adding the command below to the roslaunch to perform a reset of the camera at launch to see whether that can clear a potential USB issue in the wrapper's camera connection.

**initial_reset:=true**

KartmanCanada, 2021-06-18:
Done. Still no change to rviz visualization (no Color or Position Transforms offered in the rviz pulldowns)

**roslaunch realsense2_camera rs_camera.launch filters:=colorizer,pointcloud align_depth:=true pointcloud_texture_stream:=RS2_STREAM_COLOR ordered_pc:=true initial_reset:=true**

Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Xavier:45437/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: colorizer,pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
[... 81 lines cut ...]
[ INFO] [1624052052.300003036]: Add Filter: colorizer
[ INFO] [1624052052.302639024]: Add Filter: pointcloud
[ INFO] [1624052052.304574142]: num_filters: 3
[ INFO] [1624052052.304706239]: Setting Dynamic reconfig parameters.
[ INFO] [1624052052.999000781]: Done Setting Dynamic reconfig parameters.
[ INFO] [1624052053.001395423]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1624052053.003355630]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1624052053.008761367]: setupPublishers...
[ INFO] [1624052053.021698423]: Expected frequency for depth = 30.00000
[ INFO] [1624052053.180076601]: Expected frequency for color = 30.00000
[ INFO] [1624052053.300174268]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1624052053.427357812]: setupStreams...
[ INFO] [1624052053.441535294]: insert Depth to Stereo Module
[ INFO] [1624052053.441773408]: insert Color to RGB Camera
 18/06 17:34:13,538 WARNING [546946695552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1624052053.589932469]: SELECTED BASE:Depth, 0
[ INFO] [1624052053.649936759]: RealSense Node Is Up!
[ WARN] [1624052053.659941826]: 
 18/06 17:34:13,661 WARNING [546946695552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/06 17:34:13,712 WARNING [546946695552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/06 17:34:13,937 WARNING [546946695552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/06 17:34:13,987 WARNING [546946695552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1624052093.563650196]: publishing ordered pointcloud.
[ WARN] [1624052094.111394934]: No stream match for pointcloud chosen texture Process - Color
 18/06 17:34:56,772 ERROR [546506019200] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132

MartyG-RealSense, 2021-06-18:
This is quite unusual.  Next time that you launch, instead of using RViz can you instead check whether the depth stream is publishing any data by inputting the command below into the ROS terminal.

**rostopic echo /camera/depth/image_rect_raw/header**

KartmanCanada, 2021-06-18:
**rostopic echo /camera/depth/image_rect_raw/header**
seq: 0
stamp: 
  secs: 1624055070
  nsecs: 675529003
frame_id: "camera_depth_optical_frame"
---
seq: 1
stamp: 
  secs: 1624055070
  nsecs: 675529003
frame_id: "camera_depth_optical_frame"
---
seq: 2
stamp: 
  secs: 1624055070
  nsecs: 808955431
frame_id: "camera_depth_optical_frame"
---
seq: 3
stamp: 
  secs: 1624055070
  nsecs: 808955431
frame_id: "camera_depth_optical_frame"

KartmanCanada, 2021-06-18:
**rostopic echo /camera/depth/image_rect_raw/header**

seq: 0
stamp: 
  secs: 1624055070
  nsecs: 675529003
frame_id: "camera_depth_optical_frame"

seq: 1
stamp: 
  secs: 1624055070
  nsecs: 675529003
frame_id: "camera_depth_optical_frame"

seq: 2
stamp: 
  secs: 1624055070
  nsecs: 808955431
frame_id: "camera_depth_optical_frame"

seq: 3
stamp: 
  secs: 1624055070
  nsecs: 808955431
frame_id: "camera_depth_optical_frame"

seq: 4
stamp: 
  secs: 1624055070
  nsecs: 942381382
frame_id: "camera_depth_optical_frame"

MartyG-RealSense, 2021-06-18:
That suggests to me that it is publishing the depth.  Indeed, in your original message you mentioned that "When I run **roslaunch realsense2_camera rs_camera.launch filters:=pointcloud pointcloud_texture_stream:=RS2_STREAM_ANY** I can open RVIZ and get a pointcloud2 to visualize but only with AxisColor, Intensity, or FlatColor".

I am away from computer for 7 hours now and will be happy to return to the discussion on Saturday.  I will also tag @doronhi the RealSense ROS wrapper developer into this discussion for his advice.  Thanks very much again for your patience.

KartmanCanada, 2021-06-19:
While I wait for a further response, I've been messing about more... 

Interestingly, in RVIZ, if I add an "image" display, I get no images from the /camera1/color/image/raw. But, I do get images from /camera1/depth/image_rect_raw. I should be able to display the depth and RGB images, no?

Could this be the source of the issue with the missing RGB8 color transform when trying to visualize the RealSense point cloud?

[image]

doronhi, 2021-06-20:
The issue definitely seems related to the fact that no color images reach realsense2_camera driver.
To try and find the root cause, I would try realsense-viewer. Do you get color stream there? Do you get a stream color using a different machine? 
What method did you use to install realsense2_camera and librealsense2?

KartmanCanada, 2021-06-20:
1) realsense-viewer works just fine... I can get depth and color 2D images -and- display a 3D pointcloud
2) I only have the one machine to test today but I can work on building another machine and environment. This is a Xavier NX Dev Kit running Ubuntu 18.04.
3) Installation method for install was "method1" (i.e., ROS distribution)

KartmanCanada, 2021-06-20:
Just tried some more playing about and the RS cameras seems to be a bit unreliable if I kill and restart wrapper... (i.e., missing RGB image streams and missing on init but successful on retry when starting wrapper. 

If I unplug and replug the USB connector physically, this seems to help....

After unplug and replug, I was able to get the RGB image in RVIZ but, after swapping back to RS2_STREAM_COLOR, still no RGB8 option under pointcloud visualization

doronhi, 2021-06-21:
ros-melodic-realsense2-camera depends on, and automatically installs ros-melodic-librealsense2.
realsense-viewer comes from a different installation which most likely was built with other flags. Therefore, the different behavior. Assuming it comes from an installed package, librealsense2-utils, I suggest the following:
remove the realsense2-camera packages: `sudo purge ros-melodic-realsense2-camera and ros-melodic-librealsense2`
Install realsense2_Camera using method 2. In Method2-step 1 - you can install librealsense2 the way you already did (i.e. from debians or from code)

KartmanCanada, 2021-06-21:
Removed the ROS packages as indicated.

Under Method2, Step1 - key and repositories added fine but, when I try to install the DKMS, I get the following message:
**E: Unable to locate package librealsense2-dkms**

Further up on the install page it says, DKMS install is supported by "Ubuntu LTS kernels 4.4, 4.8, 4.10, 4.13, 4.15" but my Xavier NX install from NVIDIA page is "Linux Xavier 4.9.201-tegra".

How do I get the DKMS installed as part of Method2?

MartyG-RealSense, 2021-06-21:
Hi @KartmanCanada   **Method 2** describes installing librealsense and the RealSense ROS wrapper separately - librealsense first and then the ROS wrapper secondly after librealsense has been installed.  So to install librealsense from packages on a Jetson board, an approprite set of instructions would be the ones in section **4.  Install with Debian Packages** of librealsense's official Jetson installation page.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

KartmanCanada, 2021-06-21:
Trying not to get frustrated here... I appreciate the patience.

librealsense 2.3 installed from the Jetson instructions and tested with realsense-viewer.

Following instructions for ROS_RealSense build from source, I'm getting the following error:
-- +++ processing catkin package: 'realsense2_camera'
-- ==> add_subdirectory(realsense-ros/realsense2_camera)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at /opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake:113 (message):
  Project 'cv_bridge' specifies '/usr/include/opencv' as an include dir,
  which is not found.  It does neither exist as an absolute directory nor in
  '${{prefix}}//usr/include/opencv'.  Check the issue tracker
  'https://github.com/ros-perception/vision_opencv/issues' and consider
  creating a ticket if the problem has not been reported yet.

Both ros-melodic-vision-opencv and ros-melodic-cv-bridge packages are installed. The error says it's looking for /usr/include/opencv but I only have /usr/include/opencv4????

KartmanCanada, 2021-06-21:
Following the Jetson path above seems to have left me at SDK 2.41... see message below when i try to build the ros_realsense:
CMake Warning at realsense-ros/realsense2_camera/CMakeLists.txt:44 (find_package):
  Could not find a configuration file for package "realsense2" that is
  compatible with requested version "2.45.0".

  The following configuration files were considered but not accepted:

    /usr/lib/aarch64-linux-gnu/cmake/realsense2/realsense2Config.cmake, version: 2.41.0

MartyG-RealSense, 2021-06-21:
When you run the RealSense Viewer, which version number is shown at the top of the Viewer window, please?

[image]

KartmanCanada, 2021-06-21:
[image]

KartmanCanada, 2021-06-21:
[image]

MartyG-RealSense, 2021-06-21:
The link below may be a useful reference for installing ROS on Jetson.  It was last updated in November 2020.

https://developer.ridgerun.com/wiki/index.php?title=Robot_Operating_System_ROS_on_Jetson_Platforms

KartmanCanada, 2021-06-21:
fine... except you're sending me in circles.... this is what I had 3x days
ago and will only install SDK 2.41. But the ros_realsense (latest requires
SDK 2.45...

On Jetson (Xavier NX), how do I get EITHER SDK to 2.45 -or- ros_realsense
to install on 2.41. I've spent days on this an have gotten nowhere.

Help please.

doronhi, 2021-06-22:
@MartyG-RealSense , thanks for pointing out the link error - Method 2 - step 1 should refer users with Jetson to the appropriate librealsense2 installation guide. I'll take care of that.
Also, it seems like the links you gave @KartmanCanada refer to the installation of debian ros-melodic-realsense2-camera whereas my suggestion was to avoid this installation. Hence the "moving in circles". The current issue is why @KartmanCanada got version 2.41.0 and not the latest - 2.47.0. Neglecting to `apt update` seems too simple.
I suggest trying to remove all debian installations of librealsense from your machine and try again.
I'll try to obtain a Xavier and try it out but in the meanwhile, I believe the direction to proceed is installing version 2.45 and above of librealsense2

MartyG-RealSense, 2021-06-22:
Thanks very much @doronhi

doronhi, 2021-06-22:
Regarding the kernel patches, as seen in the [release notes for librealsense2 v2.43.0](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#nvidia-jetson---user-notification) if you have Jetpack 4.4.1 you don't need kernel patches.

Block137, 2021-06-22:
Allow me to use this discussion, I was about to open same issue.

My setup: AGX Xavier + 2x D430(no RGB) camera

I can only get pointcloud to publish with 
`<arg name="pointcloud_texture_stream" value="RS2_STREAM_ANY"/>`
and it stops publishing if `enable_infra1` or `enable_infra2` is set to `true`

KartmanCanada, 2021-06-22:
#response_container_BBPPID{font-family: initial;font-size: initial;color: initial;}  So, to be clear... the current recommended path is:1) Debian install of librealsense2 (Jetson/RealSense install documents)2) Build realsense_ros from source (step 2 of "Method 2"I did a clean install of Jetson yesterday. #1 above installed libreealsense v2.41 and then the source of realsense_ros wouldn't build as it wanted 2.45.Has this been resolved? (I did notice today that the instructions are already updated to push Jetson users to the Jetson instructions. Thanks for that update)In the Jetson instructions, this mentions picking a backend under "libreealsense SDK backend". I figured this would become clearer as I worked through the install, got the cameras working, and gained experience. Right now, I find this section confusing. Just FYI.                                                                                                                                                                                                                                                                                                                          From: ***@***.***Sent: June 22, 2021 12:33 a.m.To: ***@***.***Reply to: ***@***.***Cc: ***@***.***; ***@***.***Subject: Re: [IntelRealSense/realsense-ros] Nothing showing up in RVIZ with D435 (#1938)  
@MartyG-RealSense , thanks for pointing out the link error - Method 2 - step 1 should refer users with Jetson to the appropriate librealsense2 installation guide. I'll take care of that.
Also, it seems like the links you gave @KartmanCanada refer to the installation of debian ros-melodic-realsense2-camera whereas my suggestion was to avoid this installation. Hence the "moving in circles". The current issue is why @KartmanCanada got version 2.41.0 and not the latest - 2.47.0. Neglecting to apt update seems too simple.
I suggest trying to remove all debian installations of librealsense from your machine and try again.
I'll try to obtain a Xavier and try it out but in the meanwhile, I believe the direction to proceed is installing version 2.45 and above of librealsense2

MartyG-RealSense, 2021-06-22:
Hi @KartmanCanada  **RSUSB** is a librealsense installation method that is performed over an internet connection.  It installs dependencies over the internet instead of doing so with kernel patching.  This means that when librealsense is installed with the RSUSB method, is not dependent on Linux versions or kernel versions and does not require patching.  It does have limitations compared to the patching-based installation method though, such as being unsuitable for projects that use more than one camera.  It is useful for situations where kernel patching is not possible or when a problem is being caused by a conflict with the kernel though.

In regard to the official Jetson instructions' reference to performing a binary installation from Debian packages using **DFORCE_RSUSB_BACKEND**, I am admittedly uncertain what the instructions are referring to since DFORCE_RSUSB_BACKEND is usually related to installing librealsense from **source code** rather than packages.  So I would recommend ignoring the mention of RSUSB when doing a package-based Jetson installation, as knowledge of RSUSB is not required to perform the package installation steps.

************

In regard to RSUSB installation from source code, the suggested method to install using the **libuvc_installation.h** script, it sounds similar to the **libuvc backend** installation process in the link below.  This method of installing librealsense over an internet connection was used before the RSUSB system was introduced.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

KartmanCanada, 2021-06-22:
@MartyG-RealSense thank you for the clarification... 

When will the Debian repository have something newer than 2.41 of librealsense? This issue has wandered waaaaay off track but, in the end, I hope it'll prove helpful. I looked at the /var/lib/apt/lists for "https://librealsense.intel.com/Debian/apt-repo bionic main" and, even after an "apt update" the newest version number is 2.41.

The next step of building realsense_ros is demanding 2.45 on my system.

Again, this all started because I was unable to get an RGB textured point cloud... My sincere hope is that getting librealsense v2.45 and a locally built version of realsense_ros v2.3 will fix my initial issue. I appreacate the support but this is taking more time/effort than I'd budgeted.

MartyG-RealSense, 2021-06-22:
Regarding your question of **When will the Debian repository have something newer than 2.41 of librealsense?**, I will field that to @doronhi as I am not involved in ROS development activities.  Thanks so much for your patience!

doronhi, 2021-06-23:
@RealSenseCustomerSupport , Isn't a librealsense2 debian package version 2.45.0 already available for Jetson? Any idea why @KartmanCanada keeps on receiving the old 2.41.0 version?

KartmanCanada, 2021-06-24:
@doronhi @RealSenseCustomerSupport @RealSenseSupport 

Any response on Debian 2.41 package for Jetson? I need 2.45 and I've been blocked for almost a week now.

Please advise.

doronhi, 2021-06-24:
Hi @KartmanCanada , to the best of my knowledge there was an issue updating librealsense2 debian packages for Jetson. Until further response from @RealSenseSupport, and since you are currently blocked, I can only suggest building librealsense2 from code:
- Uninstall every version you have of librealsense - `apt list --installed | grep realsense` should return nothing, perhaps maybe to `librealsense2-udev-rules` that you can keep.

- Follow that link: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-native-backend
*If you have Jetpack 4.4.1 you can skip the "Build and Patch Kernel Modules for Jetson L4T" section.
*Make sure to set the appropriate CMake flags. `-DCMAKE_BUILD_TYPE=release -DFORCE_RSUSB_BACKEND=false -DBUILD_WITH_CUDA=true`

If the viewer runs ok with all the streams you need, carry on to building realsense2_camera.

KartmanCanada, 2021-06-25:
@doronhi 

1) built librealsense 2.47 from source. Success. Tested with 'realsense-viewer'. Depth, RGB, and 3D pointcloud displayed.

2) built  realsense-ros from source. Success. BUT SAME ISSUE THAT STARTED THIS ISSUE THREAD. in RVIZ, I can subscribe to and display the depth image. I can subscribe to but get no RGB images. And, of course, the PointCloud is empty.

If change the parameter 'name="pointcloud_texture_stream"' from RS2_STREAM_COLOR to RS2_STREAM_ANY. I can view both the depth and RGB images plus the point cloud is available but only with intensity (no RGB8 overlay)

I've wasted a week on this. Can we please fix? Image and log below
[image]

SUMMARY
========

PARAMETERS
 * /camera1/realsense2_camera/accel_fps: -1
 * /camera1/realsense2_camera/accel_frame_id: camera1_accel_frame
 * /camera1/realsense2_camera/accel_optical_frame_id: camera1_accel_opt...
 * /camera1/realsense2_camera/align_depth: False
 * /camera1/realsense2_camera/aligned_depth_to_color_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/allow_no_texture_points: False
 * /camera1/realsense2_camera/base_frame_id: camera1_link
 * /camera1/realsense2_camera/calib_odom_file: 
 * /camera1/realsense2_camera/clip_distance: -2.0
 * /camera1/realsense2_camera/color_fps: -1
 * /camera1/realsense2_camera/color_frame_id: camera1_color_frame
 * /camera1/realsense2_camera/color_height: -1
 * /camera1/realsense2_camera/color_optical_frame_id: camera1_color_opt...
 * /camera1/realsense2_camera/color_width: -1
 * /camera1/realsense2_camera/confidence_fps: -1
 * /camera1/realsense2_camera/confidence_height: -1
 * /camera1/realsense2_camera/confidence_width: -1
 * /camera1/realsense2_camera/depth_fps: -1
 * /camera1/realsense2_camera/depth_frame_id: camera1_depth_frame
 * /camera1/realsense2_camera/depth_height: -1
 * /camera1/realsense2_camera/depth_optical_frame_id: camera1_depth_opt...
 * /camera1/realsense2_camera/depth_width: -1
 * /camera1/realsense2_camera/device_type: 
 * /camera1/realsense2_camera/enable_accel: False
 * /camera1/realsense2_camera/enable_color: True
 * /camera1/realsense2_camera/enable_confidence: True
 * /camera1/realsense2_camera/enable_depth: True
 * /camera1/realsense2_camera/enable_fisheye1: False
 * /camera1/realsense2_camera/enable_fisheye2: False
 * /camera1/realsense2_camera/enable_fisheye: False
 * /camera1/realsense2_camera/enable_gyro: False
 * /camera1/realsense2_camera/enable_infra1: False
 * /camera1/realsense2_camera/enable_infra2: False
 * /camera1/realsense2_camera/enable_infra: False
 * /camera1/realsense2_camera/enable_pointcloud: False
 * /camera1/realsense2_camera/enable_pose: False
 * /camera1/realsense2_camera/enable_sync: False
 * /camera1/realsense2_camera/filters: pointcloud
 * /camera1/realsense2_camera/fisheye1_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye1_optical_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye2_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye2_optical_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye_fps: -1
 * /camera1/realsense2_camera/fisheye_frame_id: camera1_fisheye_f...
 * /camera1/realsense2_camera/fisheye_height: -1
 * /camera1/realsense2_camera/fisheye_optical_frame_id: camera1_fisheye_o...
 * /camera1/realsense2_camera/fisheye_width: -1
 * /camera1/realsense2_camera/gyro_fps: -1
 * /camera1/realsense2_camera/gyro_frame_id: camera1_gyro_frame
 * /camera1/realsense2_camera/gyro_optical_frame_id: camera1_gyro_opti...
[... 221 lines cut ...]
[ INFO] [1624597268.166451703]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1624597268.168217150]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1624597268.168391934]: setupPublishers...
[ INFO] [1624597268.169755908]: Done Setting Dynamic reconfig parameters.
[ INFO] [1624597268.171482827]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1624597268.173238449]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1624597268.173851828]: setupPublishers...
[ INFO] [1624597268.180776431]: Expected frequency for depth = 30.00000
[ INFO] [1624597268.196610573]: Expected frequency for depth = 30.00000
[ INFO] [1624597268.576035319]: Expected frequency for color = 30.00000
[ INFO] [1624597268.625993242]: Expected frequency for color = 30.00000
[ INFO] [1624597268.852740048]: setupStreams...
[ INFO] [1624597268.855595323]: insert Depth to Stereo Module
[ INFO] [1624597268.855761276]: insert Color to RGB Camera
[ INFO] [1624597268.867832875]: setupStreams...
[ INFO] [1624597268.871078424]: insert Depth to Stereo Module
[ INFO] [1624597268.871240729]: insert Color to RGB Camera
[ INFO] [1624597268.927942710]: SELECTED BASE:Depth, 0
[ INFO] [1624597268.942832112]: SELECTED BASE:Depth, 0
[ WARN] [1624597268.999145964]: 
[ WARN] [1624597269.012711585]: 
[ INFO] [1624597269.036490654]: RealSense Node Is Up!
[ INFO] [1624597269.060923710]: RealSense Node Is Up!
[ INFO] [1624597332.175592722]: publishing unordered pointcloud.
[ WARN] [1624597332.356723497]: No stream match for pointcloud chosen texture Process - Color

KartmanCanada, 2021-06-25:
@doronhi @MartyG-RealSense 

https://github.com/IntelRealSense/realsense-ros/issues/1324

This issue seems to be very similar to my issue but I don't see a resolution. I would prefer not to unplug the RealSense cameras and put them back in the boxes. How do we resolve this?

MartyG-RealSense, 2021-06-25:
I have in the past suggested adding the topic **depth/color/points/PointCloud2** with the Add button in RViz.

https://github.com/IntelRealSense/realsense-ros/issues/1605#issuecomment-754523910

MartyG-RealSense, 2021-07-05:
A common factor that I identified between this case and two other cases that had no Color Transformer options available was that they were on computing boards with Arm processors (one with Raspberry Pi 3B+, and the other on Nvidia Jetson Xavier NX like in your own case).

https://github.com/IntelRealSense/realsense-ros/issues/1967#issuecomment-874084975

closed 2021-07-16 by MartyG-RealSense