realsenseai/realsense-ros #1941: intel430i

yaley302, 2021-06-21:
hello,
 current realsense2_camera do not support camera types for intel430i.How do i solve the problem?

MartyG-RealSense, 2021-06-21:
Hi @yaley302  Cases involving **D430i** are highly rare, with it only having appeared on RealSense forums a few times when the RGB sensor of a D435i model was not being detected.  

Can you confirm whether you have a D435i or a D430i (caseless Depth Module Kit) please?

yaley302, 2021-06-22:
I have a D435i like the camera.Unsupported device! Product ID:0x0B4B

doronhi, 2021-06-22:
If you have a D435i that identifies as a D430i, it is likely that the connector to the RGB unit inside is disconnected. I think you can validate by running `rs-enumerate-devices` and see that no Color streams are available. 
Consult @RealSenseSupport as to how to proceed.

MartyG-RealSense, 2021-06-22:
Thanks very much @doronhi 

@yaley302  The advice of @doronhi is correct that the RGB sensor may become undetectable to the firmware due to a detached internal cable.  

In the very small number of past cases where a D435 has been recognized as a D430, downgrading the firmware driver to an older version and then re-upgrading it fixed the issue in a couple of cases where it was not the internal RGB cable that was responsible.  If an internal cable detachment was responsible though then updating the firmware would have no effect.

closed 2021-07-04 by MartyG-RealSense

MartyG-RealSense, 2022-11-17:
The presence or absence of a color sensor could be checked using code like the highlighted Python scripting at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py#L73-L80

You could also check for a specific camera name using **RS2_INFO_CAMERA_NAME** like in the C++ script at https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412899468 that checks whether the camera name is D415 and does not allow the program to run if the attached camera is not a D415.

fangxiaomeng0830, 2023-03-31:
Is there a version of ROS wrapper support D430i?

MartyG-RealSense, 2023-03-31:
Hi @fangxiaomeng0830   No, D430i is just a way to identify a D435i camera when there is a problem with detecting its RGB sensor.  It is not a proper RealSense camera model.