realsenseai/realsense-ros #1947: Frame rate drop when recording rosbag (D455)

guglielmodev, 2021-06-23:
I'm seeing a huge frame rate drop when recording RGBD images from D455 (with `rosbag record -a`).

When I start the recording, rate drops from 30fps to less than 15fps with lag and jumps.

When I stop recording framerate goes up again.

Here the launch parameters:

    <include file="$(find realsense2_camera)/launch/rs_camera.launch">
        <arg name="align_depth" value="true"/>
        <arg name="linear_accel_cov" value="1.0"/>
        <arg name="enable_pointcloud"  value="false"/>
    </include>

MartyG-RealSense, 2021-06-23:
Hi @guglielmodev  What computer or computing device are you using, please (PC, Nvidia Jetson, Raspberry Pi, etc).  Thanks!

MartyG-RealSense, 2021-07-05:
Case closed due to no further comments received.

closed 2021-07-05 by MartyG-RealSense

guglielmodev, 2021-07-06:
Sorry, I missed the notification for your reply.

I'm using an Ubuntu PC. When I record only RGB and depth the issue disappears.

MartyG-RealSense, 2021-07-06:
No problem at all @guglielmodev - thanks very much for the update. 

As you are using **rosbag record -a** to record all topics, which other topics apart from RGB and depth do you have enabled that cause the lag that is not present when only depth and RGB are published, please?

guglielmodev, 2021-07-06:
Thank you @MartyG-RealSense 

With `rosbag record -a` the camera has lot of lagging and framerate goes below 15hz, when I stop recording it recovers. Also the exposure starts to oscillate and make the image glow in violet colors.

The recorded topics are the ones enabled by the above launch file:

```
/d455/align_to_color/parameter_descriptions
/d455/align_to_color/parameter_updates
/d455/aligned_depth_to_color/camera_info
/d455/aligned_depth_to_color/image_raw
/d455/aligned_depth_to_color/image_raw/compressed
/d455/aligned_depth_to_color/image_raw/compressed/parameter_descriptions
/d455/aligned_depth_to_color/image_raw/compressed/parameter_updates
/d455/aligned_depth_to_color/image_raw/compressedDepth
/d455/aligned_depth_to_color/image_raw/compressedDepth/parameter_descriptions
/d455/aligned_depth_to_color/image_raw/compressedDepth/parameter_updates
/d455/color/camera_info
/d455/color/image_raw
/d455/depth/camera_info
/d455/depth/image_rect_raw
/d455/extrinsics/depth_to_color
/d455/motion_module/parameter_descriptions
/d455/motion_module/parameter_updates
/d455/realsense2_camera_manager/bond
/d455/rgb_camera/auto_exposure_roi/parameter_descriptions
/d455/rgb_camera/auto_exposure_roi/parameter_updates
/d455/rgb_camera/parameter_descriptions
/d455/rgb_camera/parameter_updates
/d455/stereo_module/auto_exposure_roi/parameter_descriptions
/d455/stereo_module/auto_exposure_roi/parameter_updates
/d455/stereo_module/parameter_descriptions
/d455/stereo_module/parameter_updates
```

MartyG-RealSense, 2021-07-06:
How does it perform if you trim down the list of recorded topics, such as removing all the aligned and compressed image topics except /camera/aligned_depth_to_color/camera_info and /camera/aligned_depth_to_color/image_raw

[image]