realsenseai/realsense-ros #1949: WARNING [140607724955392] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

padhupradheep, 2021-06-24:
Hey I have the warning as shown in the title. I'm using the D435i camera, with Ubuntu 20.04 on ROS2 with kernel 5.8.

I'm not able to see or subscribe to any of the camera data?

MartyG-RealSense, 2021-06-24:
Hi @padhupradheep   Official support for Kernel 5.8 was added in librealsense 2.47.0.  There is not a ROS wrapper version currently available that matches with librealsense 2.47.0 though (a comment posted in the link below this week references this).

https://github.com/IntelRealSense/realsense-ros/issues/1925#issuecomment-864728530

In case it is not a kernel-related issue though, it is worth exploring other possibilities.  For example, a USB CAM overflow could suggest that the flow of data arriving at the USB port is too much for the port to handle.

To start with, could you try adding **initial_reset:=true** to your roslaunch instruction to reset the camera at launch and see whether this makes a positive difference please?

padhupradheep, 2021-06-24:
I already tried the initial_reset! But did not help!

MartyG-RealSense, 2021-06-24:
Would it be possible to copy and paste the text of your roslaunch log from Noetic into a comment below please?

padhupradheep, 2021-06-24:
from my terminal ? or do you need the log file ?

MartyG-RealSense, 2021-06-24:
From the terminal, please.

padhupradheep, 2021-06-24:
Alright my bad! I think there is no issue with noetic, I get the warnings but when I check the data on RViz it works fine. 

But the problem persists with ROS2!

padhupradheep, 2021-06-24:
This is for ROS2

`[INFO] [launch]: All log files can be found below /home/pradheep/.ros/log/2021-06-24-11-12-07-686409-pradheep-System-Product-Name-30377
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [30379]
[realsense2_camera_node-1] [INFO] [1624525927.845404723] [RealSenseCameraNode]: RealSense ROS v3.1.4
[realsense2_camera_node-1] [INFO] [1624525927.845543180] [RealSenseCameraNode]: Built with LibRealSense v2.45.0
[realsense2_camera_node-1] [INFO] [1624525927.845550627] [RealSenseCameraNode]: Running with LibRealSense v2.45.0
[realsense2_camera_node-1] [INFO] [1624525928.089184325] [RealSenseCameraNode]: Device with serial number 948122071645 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1624525928.089248541] [RealSenseCameraNode]: Device with physical ID 1-7-9 was found.
[realsense2_camera_node-1] [INFO] [1624525928.089265204] [RealSenseCameraNode]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1624525928.089867778] [RealSenseCameraNode]: Device with port number 1-7 was found.
[realsense2_camera_node-1] [INFO] [1624525928.089896847] [RealSenseCameraNode]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [1624525928.089917977] [RealSenseCameraNode]: Device 948122071645 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1] [INFO] [1624525928.089940495] [RealSenseCameraNode]: Resetting device...
[realsense2_camera_node-1] [INFO] [1624525934.302032444] [RealSenseCameraNode]: Device with serial number 948122071645 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1624525934.302058834] [RealSenseCameraNode]: Device with physical ID 1-7-10 was found.
[realsense2_camera_node-1] [INFO] [1624525934.302065027] [RealSenseCameraNode]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1624525934.302284277] [RealSenseCameraNode]: Device with port number 1-7 was found.
[realsense2_camera_node-1] [INFO] [1624525934.302295454] [RealSenseCameraNode]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [1624525934.302301616] [RealSenseCameraNode]: Device 948122071645 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1] [INFO] [1624525934.303442771] [RealSenseCameraNode]: getParameters...
[realsense2_camera_node-1] [INFO] [1624525934.303507065] [RealSenseCameraNode]: 707
[realsense2_camera_node-1] [INFO] [1624525934.304296466] [RealSenseCameraNode]: setupDevice...
[realsense2_camera_node-1] [INFO] [1624525934.304308607] [RealSenseCameraNode]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1624525934.304317438] [RealSenseCameraNode]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1624525934.304323925] [RealSenseCameraNode]: Device Serial No: 948122071645
[realsense2_camera_node-1] [INFO] [1624525934.304329258] [RealSenseCameraNode]: Device physical port: 1-7-10
[realsense2_camera_node-1] [INFO] [1624525934.304334278] [RealSenseCameraNode]: Device FW version: 05.12.14.50
[realsense2_camera_node-1] [INFO] [1624525934.304339375] [RealSenseCameraNode]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1624525934.304344175] [RealSenseCameraNode]: Enable PointCloud: Off
[realsense2_camera_node-1] [INFO] [1624525934.304348979] [RealSenseCameraNode]: Align Depth: Off
[realsense2_camera_node-1] [INFO] [1624525934.304353694] [RealSenseCameraNode]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1624525934.304366993] [RealSenseCameraNode]: Device Sensors: 
[realsense2_camera_node-1] [INFO] [1624525934.317403192] [RealSenseCameraNode]: Stereo Module was found.
[realsense2_camera_node-1] [INFO] [1624525934.322519795] [RealSenseCameraNode]: RGB Camera was found.
[realsense2_camera_node-1] [INFO] [1624525934.322637675] [RealSenseCameraNode]: Motion Module was found.
[realsense2_camera_node-1] [INFO] [1624525934.322664800] [RealSenseCameraNode]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1624525934.322673547] [RealSenseCameraNode]: (Infrared, 2) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1624525934.322680654] [RealSenseCameraNode]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1624525934.322687280] [RealSenseCameraNode]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1624525934.322693540] [RealSenseCameraNode]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1624525934.322700065] [RealSenseCameraNode]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1624525934.322706547] [RealSenseCameraNode]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1624525934.322715290] [RealSenseCameraNode]: num_filters: 0
[realsense2_camera_node-1] [INFO] [1624525934.322721873] [RealSenseCameraNode]: Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [WARN] [1624525934.739081487] [RealSenseCameraNode]: Failed to set parameter:stereo_module.auto_exposure_limit = 0[1, 165000]
[realsense2_camera_node-1] parameter 'stereo_module.auto_exposure_limit' could not be set: Parameter {} doesn't comply with integer range.
[... cut ...]
[realsense2_camera_node-1] [WARN] [1624525935.824447121] [RealSenseCameraNode]: Hardware Notification:Right MIPI error,1.62453e+12,Error,Hardware Error
[realsense2_camera_node-1]  24/06 11:12:15,846 WARNING [139884148791040] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[realsense2_camera_node-1]  24/06 11:12:15,896 WARNING [139884148791040] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
`

MartyG-RealSense, 2021-06-24:
Thanks for the log.  It seems that the launch is not exiting with a rash, as you have **RealSense Node Is Up!**.  Data may not be being published to topics though, as indicated by the **control_transfer returned** error.  If this error is continuously generating instead of just appearing a small number of times and then stopping, then it can suggest that there is a serious communication problem with the device.

I note that the log states that you are using ROS2 wrapper version **3.1.4**.  This is designed for librealsense SDK version 2.42.0.  Ideally, you want to match a wrapper version as closely as possible with the librealsense version listed in that wrapper version's release notes.  As you are currently using **2.45.0** according to the log, wrapper **3.2.0** would be an appropriate match.

https://github.com/IntelRealSense/realsense-ros/releases/tag/3.2.0

Could you confirm whether you are also experiencing **control_transfer returned** errors when testing with Noetic?

padhupradheep, 2021-06-24:
> Could you confirm whether you are also experiencing control_transfer returned errors when testing with Noetic? And which ROS wrapper version have you used to test with Noetic?

Yes, it happens with noetic as well. I do not know the version, I just installed as a debian package today morning.

MartyG-RealSense, 2021-06-24:
Okay, thank you.  As there are separate ROS wrapper branches for ROS1 (Noetic) and ROS2 (Foxy), I just wanted to confirm that you had installed the right wrapper type and a wrapper version that matches with librealsense 2.45.0.

**ROS1 - 2.3.0**
https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.0

**ROS2 - 3.2.0**
https://github.com/IntelRealSense/realsense-ros/releases/tag/3.2.0

MartyG-RealSense, 2021-06-30:
Hi @padhupradheep  Do you require further assistance with this case, please?  Thanks!

padhupradheep, 2021-06-30:
Yes, the issue has not been yet resolved I think. I'm still not able to get
it running with ROS2.

doronhi, 2021-06-30:
It seems from the log file that you get some "control_transfer returned error" messages at the beginning and then every now and then some more, but not a continues stream of messages, is that true?
Do you see the topics published? (`ros2 topic list`)
Are some streams being sent or non at all? `ros2 topic hz /camera/color/image_raw`? `ros2 topic hz /camera/depth/image_rect_raw`?
The QoS was changed to SYSTEM_DEFAULT in version 3.1.5 - It could be an issue watching the stream using rqt_image_view but in my experience, Rviz2 is indifferent to differences in QoS.

padhupradheep, 2021-06-30:
> It seems from the log file that you get some "control_transfer returned error" messages at the beginning and then every now and then some more, but not a continues stream of messages, is that true?

Yes, that is true! 

> Do you see the topics published? (ros2 topic list)

Yes I can see the topics! 

> Are some streams being sent or non at all? ros2 topic hz /camera/color/image_raw? ros2 topic hz /camera/depth/image_rect_raw?

Nothing is sent! 

> The QoS was changed to SYSTEM_DEFAULT in version 3.1.5 - It could be an issue watching the stream using rqt_image_view but in my experience, Rviz2 is indifferent to differences in QoS.

padhupradheep, 2021-07-06:
I'm not really sure whether the issue was resolved. For noetic it did, but for Foxy I still need to update it and check!

MartyG-RealSense, 2021-07-06:
Thanks very much for the response!

MartyG-RealSense, 2021-07-12:
Hi @padhupradheep  Do you have an update about this case that you can provide, please?  Thanks!

padhupradheep, 2021-07-13:
Hey! I need some more time to check it (2 days max!), just out of curiosity did you guys update your ROS2 packages recently?

closed 2021-07-13 by padhupradheep

MartyG-RealSense, 2021-07-13:
Thanks very much @padhupradheep for the information!   I do not have information about the update schedule of the ROS2 packages.  Please do feel free to re-open this case if you need to do so.  Good luck!

padhupradheep, 2021-07-20:
FYI, working fine in ROS-2 Foxy! Thanks @MartyG-RealSense

MartyG-RealSense, 2021-07-20:
That's great news that Foxy is working well for you - thanks very much for the update!

zhaojiaran111, 2022-06-29:
I also got this error on my ubuntu 18.04

zhaojiaran111, 2022-06-29:
29/06 15:34:16,453 WARNING [139784887027456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/06 15:34:16,504 WARNING [139784887027456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/06 15:34:16,556 WARNING [139784887027456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/06 15:34:16,609 WARNING [139784887027456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/06 15:34:16,660 WARNING [139784887027456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/06 15:34:16,711 WARNING [139784887027456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

MartyG-RealSense, 2022-06-29:
Hi @zhaojiaran111  Could you test please whether adding **initial_reset:=true** to your roslaunch instruction to reset the camera at launch makes a positive difference, please?  For example:

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**

zhaojiaran111, 2022-06-29:
still error

MartyG-RealSense, 2022-06-29:
In https://github.com/IntelRealSense/realsense-ros/issues/2386 you say that you have tried all the methods in that case.

Are you using the official 1 meter USB cable supplied with the camera or a longer USB cable of your own choice, please?

zhaojiaran111, 2022-06-29:
the official 1 meter USB

MartyG-RealSense, 2022-06-29:
Which computer are you using (Raspberry Pi, Nvidia Jetson, desktop / laptop PC, etc)

What method did you use to install librealsense and the RealSense ROS wrapper?  For example, did you install librealsense and the ROS wrapper together at the same time with the ROS wrapper's **Method 1** instructions?  Or did you build librealsense and the wrapper separately (librealsense first and the wrapper second)

zhaojiaran111, 2022-06-29:
Install according to the source code of the official website

MartyG-RealSense, 2022-06-29:
You installed librealsense from source code first with these instructions?

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

And then after that you installed the wrapper from source code with these instructions?

https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources

Is that correct?

Which computer are you using, please?