realsenseai/realsense-ros #1950: L515 control transfer error: WARNING [547054670208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

jesseArcbyt, 2021-06-24:
Hello, I am trying to use hardware syncing on two L515 cameras through ROS. Every second or so, I have one of the following errors. `WARNING [547054670208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61` or a similar error about resource being busy.  I have gotten this error just using the realsense-viewer on both a Jetson Nano running Ubuntu and a macbook pro.  I am still able to see all the ROS topics, open them in RVIZ, and see the data through the Realsense-viewer. It doesn't seem to matter what parameters I put in, I still get the errors every couple of seconds. I have tried three different cables that came with three separate L515 cameras and multiple generic cables as well. I have tried setting initial_reset:=true in the roslaunch command as well but that made no difference. P

Is this something I should worry about?  It didn't seem to be much of a problem at first but I am having trouble getting both cameras to shift into sync mode and I have a feeling its connected. 

here is my launch command
roslaunch realsense2_camera rs_multiple_devices.launch serial_no_camera1:=f0352241 serial_no_camera2:=f1062638 enable_sync:=True filters:=spatial,temporal,pointcloud_

Please advise. Thanks!

Here is the log. 

`roslaunch realsense2_camera rs_multiple_devices.launch serial_no_camera1:=f0352241 serial_no_camera2:=f1062638 enable_sync:=true initial_reset:=true filters:=spatial,temporal,pointcloud_
... logging to /home/petra2/.ros/log/984c3c5a-d2c9-11eb-b393-00044bec1e9a/roslaunch-petra2-desktop-19058.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://petra2-desktop:37975/

SUMMARY
========

PARAMETERS
 * /camera1/realsense2_camera/accel_fps: 0
 * /camera1/realsense2_camera/accel_frame_id: camera1_accel_frame
 * /camera1/realsense2_camera/accel_optical_frame_id: camera1_accel_opt...
 * /camera1/realsense2_camera/align_depth: False
 * /camera1/realsense2_camera/aligned_depth_to_color_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/allow_no_texture_points: False
 * /camera1/realsense2_camera/base_frame_id: camera1_link
 * /camera1/realsense2_camera/calib_odom_file: 
 * /camera1/realsense2_camera/clip_distance: -1.0
 * /camera1/realsense2_camera/color_fps: 30
 * /camera1/realsense2_camera/color_frame_id: camera1_color_frame
 * /camera1/realsense2_camera/color_height: 480
 * /camera1/realsense2_camera/color_optical_frame_id: camera1_color_opt...
 * /camera1/realsense2_camera/color_width: 640
 * /camera1/realsense2_camera/confidence_fps: 30
 * /camera1/realsense2_camera/confidence_height: 480
 * /camera1/realsense2_camera/confidence_width: 640
 * /camera1/realsense2_camera/depth_fps: 30
 * /camera1/realsense2_camera/depth_frame_id: camera1_depth_frame
 * /camera1/realsense2_camera/depth_height: 480
 * /camera1/realsense2_camera/depth_optical_frame_id: camera1_depth_opt...
 * /camera1/realsense2_camera/depth_width: 640
 * /camera1/realsense2_camera/device_type: 
 * /camera1/realsense2_camera/enable_accel: False
 * /camera1/realsense2_camera/enable_color: True
 * /camera1/realsense2_camera/enable_confidence: True
 * /camera1/realsense2_camera/enable_depth: True
 * /camera1/realsense2_camera/enable_fisheye1: False
 * /camera1/realsense2_camera/enable_fisheye2: False
 * /camera1/realsense2_camera/enable_fisheye: False
 * /camera1/realsense2_camera/enable_gyro: False
 * /camera1/realsense2_camera/enable_infra1: False
 * /camera1/realsense2_camera/enable_infra2: False
 * /camera1/realsense2_camera/enable_infra: False
 * /camera1/realsense2_camera/enable_pointcloud: False
 * /camera1/realsense2_camera/enable_pose: False
 * /camera1/realsense2_camera/enable_sync: False
 * /camera1/realsense2_camera/filters: 
 * /camera1/realsense2_camera/fisheye1_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye1_optical_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye2_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye2_optical_frame_id: camera1_fisheye2_...
[... 275 lines cut ...]
 24/06 11:09:03,030 WARNING [547323105664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 24/06 11:09:03,627 WARNING [547104993664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 24/06 11:09:04,293 WARNING [547323105664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 24/06 11:09:05,371 WARNING [547323105664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 24/06 11:09:06,090 WARNING [547104993664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 24/06 11:09:06,757 WARNING [547323105664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 24/06 11:09:07,354 WARNING [547104993664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 24/06 11:09:08,430 WARNING [547323105664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 24/06 11:09:09,221 WARNING [547323105664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 24/06 11:09:10,492 WARNING [547323105664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 24/06 11:09:11,022 WARNING [547104993664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 24/06 11:09:11,746 WARNING [547323105664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 24/06 11:09:12,998 WARNING [547323105664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 24/06 11:09:13,490 WARNING [547104993664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 24/06 11:09:14,258 WARNING [547323105664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 24/06 11:09:14,757 WARNING [547104993664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
^C[camera2/realsense2_camera-4] killing on exit
[camera2/realsense2_camera_manager-3] killing on exit
[camera1/realsense2_camera-2] killing on exit
[camera1/realsense2_camera_manager-1] killing on exit
 24/06 11:09:16,011 WARNING [547104993664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 24/06 11:09:16,724 WARNING [547323105664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
shutting down processing monitor...
... shutting down processing monitor complete
done`

jesseArcbyt, 2021-06-24:
Here is the other error  `WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`

MartyG-RealSense, 2021-06-26:
Hi @jesseArcbyt  If the **control_transfer returned** error is generating continuously instead of a small number of times during launch and then stopping, or at a rate of one or two per second after launch, then this can indicate that there is a serious communication problem with the device. Having said that, it is a positive sign that the topics are publishing and you are able to view their data in RViz.

As you mentioned that the issue occurs no matter what settings you use, can you confirm please that removing the spatial and temporal filters and enable_sync from the roslaunch instruction makes no difference.

Do you still experience control_transfer returned errors in the ROS wrapper if you launch a single L515 in the ROS wrapper using the roslaunch instruction below: 

**roslaunch realsense2_camera rs_camera.launch filters:=pointcloud**

A simple one-camera roslaunch like this one should help to confirm whether or not the issue is related to use of more than one camera.

jesseArcbyt, 2021-06-26:
Hello Marty,

I get the issue no matter what filters or parameters are added on whether
using one or both cameras, using both the realsense viewer and through the
ROS wrapper. I also got the same error on a MacBook Pro just running the
viewer and on two jetson nanos. If it helps I can post the log when I get
back to my computer.

Thanks,

Jesse

MartyG-RealSense, 2021-06-26:
Yes, please do use the small arrow in the bottom corner of the Viewer to expand open the debug log and post an image snapshot of the Viewer log in this discussion.  Thanks!

[image]

jesseArcbyt, 2021-06-28:
[image]
[image]
[image]

MartyG-RealSense, 2021-06-28:
As you mention hardware sync of multiple cameras at the start of this case, I should emphasize that the **enable_sync** setting in the RealSense ROS wrapper is not related to multi-camera hardware sync.  The wrapper's documentation states that "enable_sync gathers closest frames of different sensors, infra red, color and depth, to be sent with the same timetag.  This happens automatically when such filters as pointcloud are enabled".  In other words, sync between the different types of imaging sensor on a camera.

Details of how to perform **hardware sync** of multiple L515 cameras are provided in Intel's white-paper document on the subject in the link below.  

https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup

I can see from your Viewer images though that in that situation you are experiencing the **control_transfer returned** warnings when using only one camera.

Which firmware version is your camera using please?  You can find this in the Viewer by left-clicking on the **Info** option in the side-panel.

[image]

jesseArcbyt, 2021-06-28:
Hi Marty, 

Yes I have seen the paper. Is there a different method of enabling intercam sync through ROS?  I have it all setup to send triggers to each of the cameras. How do I consistently put it in intersync mode from the terminal or launch file?

The warnings do not prevent me from streaming normally through the viewer or block any topics and data from being published, but they do sometimes slow down the stream. The warnings come regardless of which method used (viewer or ROS) and are generated continuously. 

Thanks.

MartyG-RealSense, 2021-06-28:
You could try setting the value in ROS with **inter_cam_sync_mode** after launch during runtime using a dynamic_reconfigure instruction.  The links below provide examples of doing so.

https://github.com/IntelRealSense/realsense-ros/issues/984#issuecomment-707729244

https://github.com/IntelRealSense/realsense-ros/issues/1379

jesseArcbyt, 2021-06-28:
Hi Marty, 

I had previously tried that too. That worked but we still got all the warnings/errors in the console

MartyG-RealSense, 2021-06-29:
Let's try a very basic L515 custom roslaunch configuration with depth only at a low resolution to test whether the **control_transfer returned** warnings are occurring because the USB system on your computer is being over-burdened.

**roslaunch realsense2_camera rs_camera.launch depth_width:=320 depth_height:=240 depth_fps:=30**

jesseArcbyt, 2021-06-29:
Hi Marty, 

Here is the log from that command. 

` roslaunch realsense2_camera rs_camera.launch depth_width:=320 depth_height:=240 depth_fps:=30
... logging to /home/petra2/.ros/log/8d52584a-d85d-11eb-92d4-a1f6eae63aa5/roslaunch-petra2-desktop-7961.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://petra2-desktop:43687/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 240
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 320
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
[... 89 lines cut ...]
[ INFO] [1624918304.047143536]: RealSense Node Is Up!
[ WARN] [1624918304.086683432]: 
[ WARN] [1624918304.086876297]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
 28/06 17:11:44,894 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/06 17:11:46,150 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/06 17:11:47,401 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/06 17:11:48,462 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/06 17:11:48,656 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/06 17:11:49,908 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/06 17:11:51,161 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/06 17:11:52,412 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/06 17:11:53,664 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/06 17:11:54,918 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/06 17:11:56,170 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 28/06 17:11:59,926 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/06 17:12:01,179 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
^C[camera/realsense2_camera-2] killing on exit
[camera/realsense2_camera_manager-1] killing on exit
 28/06 17:12:02,431 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
shutting down processing monitor...
... shutting down processing monitor complete
done
`

MartyG-RealSense, 2021-06-29:
It does seem as though it is having trouble streaming even a basic configuration.  

You stated at the beginning of this case though that "I am still able to see all the ROS topics, open them in RVIZ, and see the data through the Realsense-viewer", so that suggests that the camera hardware is functioning.  As you are experiencing the same issue on both Jetson Nano and Macbook, this also reduces the likelihood that it is a problem with one specific computer / computing device.

You have also tried multiple USB cables, so this reduces the likelihood of it being a USB issue.  Could you try unplugging the cable that you are currently using from the side of the camera, flipping it round to the opposite side and re-inserting it into the side of the camera please to test whether it makes a difference (the cable is two-way insertion at the camera end).

jesseArcbyt, 2021-06-29:
I tried flipping the cable, I tested both directions in both of my cameras but they all had the same result. I have tested with other USB3 to USBC cables too to rule out the cable being a problem. 

I am starting to think it could be an installation issue or mismatched versions maybe that is causing this.  I am redoing the installation on another Jetson Nano and using the Jetson Hacks installation method the viewer works error free for the most part but the ROS wrapper then doesn't recognize the camera.

MartyG-RealSense, 2021-06-29:
If the librealsense installation is changed then the ROS wrapper needs to be built again afterwards.  Have you rebuilt the wrapper after installing librealsense with the JetsonHacks method?

in regard to L515 compatibility, support for the camera was introduced in ROS wrapper version **2.2.13** in March 2020.

jesseArcbyt, 2021-06-29:
Yes I did, it kept saying the device was not recognized but rs-enumerate showed it was connected. Reading through their install files, it looks like the Jetson Hacks ROS install method installs v2.2.11, while the librealsense install is v2.31

MartyG-RealSense, 2021-06-29:
It is likely that the 2.2.11 ROS wrapper does not know how to use the L515 if support for it was not introduced until 2.2.13.

**Doronhi** the RealSense ROS wrapper developer comments in the link below with the suggestion to build librealsense from source code on Jetson using Intel's official Jetson instructions and then build the RealSense ROS wrapper from source in order to have an up-to-date librealsense and RealSense ROS configuration.

https://github.com/IntelRealSense/realsense-ros/issues/1938#issuecomment-867379211

jesseArcbyt, 2021-06-29:
Yes, it had an up to date installation of librealsense on my mac but I did not have ROS installed there. Doing so gave the same error at the same frequency. 

I am trying to reinstall according to the jetson instructions on the from the comment you mentioned.

MartyG-RealSense, 2021-06-29:
Okay, thanks very much for your patience during this case.

jesseArcbyt, 2021-06-29:
No problem, thank you very much for your help

mshong0320, 2021-06-30:
I'm also having the same warning error on Ubuntu 18.04 using ROS Melodic. I've been trying to use l515 with t265 pose tracking but no success yet. I'm thinking that this issue has to do with it. I have no trouble getting d455 with t265 working together. And while using l515 I get [ WARN] [1625072340.813573352]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically. Do you think this clock warning might also interfere with getting l515 working with t265?

MartyG-RealSense, 2021-07-06:
Hi @jesseArcbyt  Do you have an update about this case that you can provide please?  Thanks!

MartyG-RealSense, 2021-07-11:
Hi @jesseArcbyt  Do you require further assistance with this case, please?  Thanks!

jesseArcbyt, 2021-07-12:
Hi Marty, sorry for the lack of response. I got it working. I reflashed on a new jetson, installed librealsense using the jetson hacks method, and downloaded the ros-package via the jetson instructions on your website. I edited the CMake file in the ros package to match that of librealsense. I had to play around with version control of opencv and reinstall ddynamic_reconfigure. From there., I played around with making clean builds with Cmake to get everything to compile. I kept getting a J4 error but eventually worked my way past it. From there everything worked for that Nano.  You can go ahead and close the issue if @mshong0320 doesn't need help. 

Thanks for all of your help!

MartyG-RealSense, 2021-07-12:
That's excellent news @jesseArcbyt  - thanks so much for the update and for sharing details of your solution with the community!

@mshong0320  I recommend creating a new case for your issue using the green **New Issue** button on the front page of the RealSense ROS forum at the link below, as I am not involved in T265 support.  Thanks!

https://github.com/IntelRealSense/realsense-ros/issues

As suggested by @jesseArcbyt, I will close this case.

closed 2021-07-12 by MartyG-RealSense

korroktheslavemaster, 2021-07-19:
@jesseArcbyt hi, could please tell us what versions of realsense-ros and libreallsense you ended up installing that worked? Did you build librealsense using the jetson hacks build scripts here https://github.com/JetsonHacksNano/installLibrealsense? Any special flags that you had to use?
Any help will be much appreciated. I seem to be having similar issues as you.

jesseArcbyt, 2021-07-19:
Hello @korroktheslavemaster, I got it to work by altering the CMake file to make the ROS wrapper version match the librealsense version. Then I had to change the CVbridge.cmake file to accept opencv4 instead of opencv. 

I recently put together a list of install instructions for the jetson NX but it should be the same. The document details the installation of ROS, librealsense, and then the Realsense-ROS wrapper. It is attached below. Please let me know if you have any questions. This should solve your problems I hope. I used the file to quickly setup other jetson NXs. 
[installs.txt](https://github.com/IntelRealSense/realsense-ros/files/6843036/installs.txt).  

Thanks, 

Jesse