realsenseai/realsense-ros #1953: Why the frequency of /camera/imu is wrong?

opendvd, 2021-06-25:
I set the accelerometer frequency to 63 and the gyroscope frequency to 200 in the launch file, but the topic: /camera/imu with  published  is still 400Hz.

  <arg name="gyro_fps"            default="400"/>
  <arg name="accel_fps"           default="250"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>
  <arg name="enable_sync"               default="true"/>
  <arg name="unite_imu_method"          default="linear_interpolation"/>

MartyG-RealSense, 2021-06-25:
Hi @opendvd  If the **imu** topic is publishing then I would guess that you are using **unite_imu_method** to combine the two separate 'gyro' and 'accel' topics into a single 'imu' topic.  Is that correct please?

If you are using unite_imu_method, are you setting this option to **linear_interpolation** or to **copy**

opendvd, 2021-06-29:
> Hi @opendvd If the **imu** topic is publishing then I would guess that you are using **unite_imu_method** to combine the two separate 'gyro' and 'accel' topics into a single 'imu' topic.
> 
> Also, you have likely enabled the publishing of the gyro and accel topics using **enable_accel** and **enable_gyro**.
> 
> Is that correct please?
> 
> If you are using unite_imu_method, are you setting this option to **linear_interpolation** or to **copy**

Thanks for your reply. What you said is right. Some key configuration is below:
```
  <arg name="gyro_fps"            default="200"/>
  <arg name="accel_fps"           default="63"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>
  <arg name="enable_sync"               default="true"/>
  <arg name="unite_imu_method"          default="linear_interpolation"/>
```
Just like the configuration:
If I set gyro_fps and accel_fps 200, 63 respectively. The topic /camera/imu will publish with 400 Hz.(This situation is wrong)
If I set gyro_fps and accel_fps 200, 250 respectively. The topic /camera/imu will publish with 1580 Hz.(This situation is wrong)
If I set gyro_fps and accel_fps 400, 250 respectively. The topic /camera/imu will publish with 400 Hz.(This situation is right)

MartyG-RealSense, 2021-06-29:
If you are only streaming IMU data then you should not need **enable_sync** to be set to true according to the comment by **Doronhi** the RealSense ROS wrapper developer in the lnk below

https://github.com/IntelRealSense/realsense-ros/issues/987#issuecomment-553244835

MartyG-RealSense, 2021-07-05:
Hi @opendvd  Do you require further assistance with this case, please?  Thanks!

opendvd, 2021-07-08:
> If you are only streaming IMU data then you should not need **enable_sync** to be set to true according to the comment by **Doronhi** the RealSense ROS wrapper developer in the lnk below
> 
> [#987 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/987#issuecomment-553244835)

Thanks for your reply!
I set the **enable_sync** to be false, but the results still same. I change the fps of gyr and acc, but the frequency of /camera/imu always 400Hz. It confused me so much.
The version of drive is : `LibRealSense v2.38.1` `RealSense ROS v2.2.17`

MartyG-RealSense, 2021-07-08:
I researched your problem further but did not find further information that suggested a cause, so I will refer the question to Doronhi.

Hi @doronhi   What might cause the **imu** topic to always have a frequency of 400 no matter what frequencies the individual gyro and accel are set to, please?

opendvd, 2021-07-09:
> I researched your problem further but did not find further information that suggested a cause, so I will refer the question to Doronhi.
> 
> Hi @doronhi What might cause the **imu** topic to always have a frequency of 400 no matter what frequencies the individual gyro and accel are set to, please?

Thank you very much! The problem is so strange.

MartyG-RealSense, 2021-07-14:
Hi @doronhi  Could you offer advice about the question from @opendvd above, please, regarding what might cause the IMU topic to always have a frequency of 400 no matter what frequencies the individual gyro and accel are set to.  Thanks!

MartyG-RealSense, 2021-07-20:
Hi @opendvd  Did you manage to find an answer yourself regarding the IMU frequency always being 400?

opendvd, 2021-07-22:
> Hi @opendvd Did you manage to find an answer yourself regarding the IMU frequency always being 400?

I'm sorry. I don't have the answer for the moment.

MartyG-RealSense, 2021-08-02:
Hi @opendvd  I have received ROS wrapper developer feedback about your question regarding the IMU frequency always being fixed at 400.  May I ask the following questions please:

-  Which ROS branch are you using (e.g Melodic)?

-  Which computer / computing device are you using (PC, Raspberry Pi, Nvidia Jetson, etc)

-  Which method did you use to install the RealSense ROS wrapper - **Method 1** (install librealsense and the wrapper together using packages) or **Method 2** (installing librealsense separately and building the wrapper from source code)?

-  Given the age of your librealsense and RealSense ROS wrapper versions, is it possible for you to update to more modern SDK and wrapper versions in your project?

-  Could you use the command **rostopic echo /camera/imu** in your ROS terminal please and post some of the messages that you receive.  This will help to diagnose whether there is a problem with the timestamps, or whether there are more wrappers in the background publishing on the same topic.

closed 2021-08-23 by MartyG-RealSense