realsenseai/realsense-ros #1954: pointcloud filter

THLL245, 2021-06-27:
D435I - Jetson Nano - ROS Melodic

When running 

roslaunch realsense2_camera rs_camera.launch

Immediately get control_transfer returned error, index 768, error: No data available, number: 61

When using RVIZ  topic /camera/depth/image_rect_raw has data and great frame rate

When I try and add the Color Image Topic /camera/color/image_raw  I get the following error

Error updating depth cloud: Depth image resolution (840x480) does not match color image resolution (1280 x 720)

new error now  control_transfer returned error, index 768, error: No data available, number: 11

Now really I would like to use this camera for its pointcloud but when doing

roslaunch reaslsense2_camera rs_camera.launch filter:=pointcloud 

The topic /camera/depth/color/points doesn't even publish anything

I am not sure if these problems are related to each other but pointcloud doesnt work

MartyG-RealSense, 2021-06-28:
Hi @THLL245   If the **control_transfer returned** warning only generates a small number of times at launch and then stops, or appears once or twice a minute after launch, then it can be safely ignored.  If it is generating continuously though then it can indicate that there is a serious communication problem with the device.  I note though that the topics are being published successfully in your particular case.

Regarding your problem with rendering the pointcloud in RViz: do you have 'Fixed Frame' set to **camera_link** like in the ROS wrapper's pointcloud example, please?

https://github.com/IntelRealSense/realsense-ros#point-cloud

[image]

THLL245, 2021-06-28:
Yes fixed frame is camera_link and control_transfer error:61 is constant about once every 3 secs but this is only when subscribed to topic /camera/depth/image_rect_raw when subscribed to /camera/color/image_raw it changes to transfer error: 11 and that is about once a sec and unable to show because of different resolutions.

MartyG-RealSense, 2021-07-05:
H @THLL245  Do you require further assistance with this case, please?  Thanks!

closed 2021-07-11 by MartyG-RealSense