realsenseai/realsense-ros #1957: Control_transfer returned error, index 768 from D435i

MiPlayer123, 2021-06-30:
Hello

I am trying to use the D435i on my Jetson Nano with ROS, but I am getting this error: `(messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61` when I run this command `roslaunch realsense2_camera rs_camera.launch`

None of the topics are being published and I cannot see anything in Rviz. If I use the RealSense viewer, everything works fine. The camera is connected to the USB3 port on the Jetson Nano. I have looked at other similar forum posts, but now of them seemed to fix the issue. Any help would be appreciated. 

Thanks

MartyG-RealSense, 2021-06-30:
Hi @MiPlayer123    If you are receiving the **control_transfer returned** warning only a small number of times during launch and then they stop, or they occur one or two times a minute after launch, then this warning can be safely ignored.  If this warning is generating continuously though then it can indicate a serious communication problem with the device.

The first test that I recommend when trying to correct this issue is to add an **initial_reset:=true** command to the roslaunch instruction to reset the camera at launch:

**roslaunch realsense2_camera rs_camera.launch initial_reset":=true**

Does using this command in your roslaunch instruction make a positive difference to the performance of your launch, please?

MiPlayer123, 2021-06-30:
@MartyG-RealSense Thanks for your reply. Resetting the camera does not seem to help. I am still getting the same error. I also noticed that very few of the topics are being published properly. I cannot see a video feed in Rviz or even the Odom topic. I am not sure where the error is coming from.

MartyG-RealSense, 2021-07-01:
Could you paste the text of the terminal log from your roslaunch into a comment please?  Thanks very much.

MiPlayer123, 2021-07-01:
Here is the log:
```
SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: 848
[... 88 lines cut ...]
[ INFO] [1625140170.585820716]: SELECTED BASE:Depth, 0
[ INFO] [1625140170.630518109]: RealSense Node Is Up!
[ WARN] [1625140170.664565422]: 
 01/07 07:49:30,665 WARNING [546299687296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 07:49:30,715 WARNING [546299687296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 07:49:30,766 WARNING [546299687296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 07:49:30,913 WARNING [546299687296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 07:49:30,964 WARNING [546299687296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 07:49:31,014 WARNING [546299687296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 07:49:31,166 WARNING [546299687296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 07:49:31,317 WARNING [546299687296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 07:49:31,772 WARNING [546299687296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 07:49:39,337 WARNING [546299687296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 01/07 07:52:02,949 WARNING [546299687296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/07 07:52:07,002 WARNING [546299687296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/07 07:52:13,058 WARNING [546299687296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```

MartyG-RealSense, 2021-07-01:
Thank you very much.  The versions of the RealSense ROS wrapper and librealsense SDK are correctly matched.

I note your comment that "very few of the topics are being published".  A number of the topics are set to False by default in the **rs_camera** launch file and therefore have to be enabled in order to be published.  The disabled by default topics that are supported on D435i include: infra1 infra2, gyro and accel.  You can enable these topics by including the commands below in your roslaunch instruction:

enable_infra1:=true
enable_infra2:=true
enable_gyro:=true
enable_accel:=true

MiPlayer123, 2021-07-01:
Thanks, I tried that, but I noticed that I am still running into issues. I can see now that the topics are being published, but I cannot see anything when I add these topic to Rviz. 
I ran `roslaunch realsense2_camera rs_camera.launch enable_infra1:=true enable_infra2:=true enable_gyro:=true enable_accel:=true`  and also ran `roslaunch realsense2_camera rs_camera.launch initial_reset:=true enable_infra1:=true enable_infra2:=true enable_gyro:=true enable_accel:=true`

Side question, is there a example launch file that has all these topics enabled and the parameters set properly?
Anyway, here is the output log:

```
SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: True
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
[... 105 lines cut ...]
[ INFO] [1625168045.287151054]: insert Infrared to Stereo Module
[ INFO] [1625168045.287222049]: insert Gyro to Motion Module
[ INFO] [1625168045.287317941]: insert Accel to Motion Module
 01/07 15:34:05,300 WARNING [547657048448] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ WARN] [1625168045.334371881]: 
[ WARN] [1625168045.336991032]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
 01/07 15:34:05,459 WARNING [547356643712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1625168045.509717627]: SELECTED BASE:Depth, 0
[ INFO] [1625168045.567584134]: RealSense Node Is Up!
 01/07 15:34:05,583 WARNING [547356643712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 15:34:05,633 WARNING [547356643712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 15:34:05,684 WARNING [547356643712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ WARN] [1625168045.701075439]: Hardware Notification:Motion Module failure,1.62517e+12,Error,Hardware Error
[... cut ...]
 01/07 15:34:27,094 ERROR [546400366976] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
 01/07 15:34:27,126 ERROR [546400366976] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
 01/07 15:34:27,161 ERROR [546400366976] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
 01/07 15:34:27,193 ERROR [546400366976] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
```

MartyG-RealSense, 2021-07-02:
Are the **control_transfer returned** messages being generated continuously or do they stop after the ones shown in the log above, please?

Which topics are you adding to RViz?

**rs_aligned_depth.launch** has depth, color, infra and guro / accel stream types all enabled as True.

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_aligned_depth.launch

MiPlayer123, 2021-07-02:
The control_transfer returned messages stop momentarily after what you see in the output log, but it keeps coming slowly after that. For some reason, I get this error `Hardware Notification:Motion Module failure,1.62524e+12,Error,Hardware Error` even when I reset the camera. 
When I ran `rs_aligned_depth.launch`, i get this error repeatedly:
```
WARNING [547046277504] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/07 10:57:11,429 ERROR [546039644544] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
 02/07 10:57:11,732 ERROR [546039644544] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
```
with the last error showing up repeatedly.

Here is an image from Rviz.
[image]
The Images show up properly, but I am not sure I I should be seeing points from the DepthCloud or not on the grid. The IMUs are not working though. Here are all the topics I see:
[image]
I am not seeing an Odom topic, am I supposed to?

Thank you for your help so far.

MartyG-RealSense, 2021-07-02:
If the **control_transfer returned** warning appears only once or twice per minute after launch then it is safe to ignore it.

**Out of frame resources** could suggest that the USB port is being over-burdened by the rate at which incoming data is arriving and jamming up the 'frame queue', preventing new frames that are arriving from being processed.

You could try a simple custom launch where only depth (no color) will be published at a 640x480 resolution and 30 FPS to test how your computer copes with incoming data.

**roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30**

As far as I am aware, **odom** is a topic that is produced by the **RealSense T265 Tracking Camera**, not the D435i.

MiPlayer123, 2021-07-02:
The command seems to work fine. I still get the control_transfer returned warning, but I get no other errors. However, when  I run
 with the gyro and accel enabled, i get this error: `Hardware Notification:Motion Module failure,1.62524e+12,Error,Hardware Error` I am not sure how to fix it. 

Thanks for the resource. I looked into it, but i get a **failed to meet update rate** error. 

As for RViz, I am not sure what you mean.

MartyG-RealSense, 2021-07-02:
Do you still experience the motion module failure if you include the command below in your roslaunch instruction:

**unite_imu_method:=linear_interpolation**

Regarding RViz, I mean going to the **Fixed Frame** option of the RViz side-panel and changing its setting from the **map** that it is currently using so that it says **camera_link**, like the image below.

[image]

MiPlayer123, 2021-07-02:
I still face the motion module failure error, but the IMU now shows up on Rviz. I am also able to see the DepthCloud now. 
[image]

I am wondering if you know how to fix **Failed to meet update rate!** when running `roslaunch realsense2_camera opensource_tracking.launch` Everything works, but it is really slow, so I am guessing its because of this error.

MartyG-RealSense, 2021-07-03:
Do you see an improvement if you set unite_imu_method to 'copy' instead of 'linear_interpolation' by adding the command below to the roslaunch instruction to override the launch file's default linear_interpolation setting?  

**unite_imu_method:=copy**

MiPlayer123, 2021-07-04:
No, its the same as before. I still get the error, but I see something like the previous image in RViz. Do you think that the camera's IMU might be affected hardware wise? This is a brand new camera though.

MartyG-RealSense, 2021-07-04:
There have been past reports of the **Failed to meet update rate!** message happening when using the **opensource_tracking.launch** launch file, so I would not strongly suspect that there is a hardware problem in the camera.  The camera certainly can publish data, though at a slow rate in this particular case.

In the RealSense Viewer, the **Motion Module Failure** error is more of an annoyance than a serious problem, as the accel and gyro streams usually activate normally the next time that they are enabled after the error has occurred.  It is more complicated in RealSense ROS though, as you cannot just press a button to re-start the IMU streams after the failure message has occurred.

Intel provide a Python-based tool to calibrate the IMU, though I would not expect IMU calibration to be a cause of this particular issue.  Indeed, a Nano user who had the Motion Module Failure message in a past RealSense ROS case performed the IMU calibration but it did not resolve the problem.  Their next step was going to be to reduce the IMU frequency to see if it helped, but they did not report the outcome of that test.

https://github.com/IntelRealSense/realsense-ros/issues/1234#issuecomment-643591578

MiPlayer123, 2021-07-05:
> I would not strongly suspect that there is a hardware problem in the camera. The camera certainly can publish data, though at a slow rate in this particular case.

Oh ok.

> Intel provide a Python-based tool to calibrate the IMU

Do you happen to know which directory the `rs-imu-calibration.py` is because I am not able to locate it.

> Is the camera moving when the message is occurring or is the camera stationary?

The camera is stationary. I am not sure if the slowdown/ freeze up is caused due to the Jetson Nano not having enough processing power or not. The CPU is up near 100% when I launch it and that could potentially cause the **Failed to meet update rate!** error.

MartyG-RealSense, 2021-07-06:
**rs-imu-calibration** can be found in the librealsense source code folder **librealsense/tools/rs-imu-calibration**.  It can be launched using the command **python rs-imu-calibration.py**

https://github.com/IntelRealSense/librealsense/tree/master/tools/rs-imu-calibration

Did you build the RealSense ROS wrapper from packages using the **Method 1** installation that installs the librealsense SDK and the wrapper together, please?  If you did then you may not have support installed in the SDK for CUDA graphics acceleration.  You can install CUDA support for your Nano by building the SDK and ROS wrapper separately, as described in the link below.

https://github.com/IntelRealSense/realsense-ros/issues/1911#issuecomment-860245792

MiPlayer123, 2021-07-11:
I tried installing **librealsense** because I have **librealsense2** and I cant find the /tools/ direcotory, but now I get this error:
[image]

While doing something, i broke the realsense-ros package somehow and I get this error:
```
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "ddynamic_reconfigure" with any of the following names:

    ddynamic_reconfigureConfig.cmake
    ddynamic_reconfigure-config.cmake
```
I tried installing Cuda, but I am running into issues. I will try that again later.

Looks like I  might have messed something up, which is causing me to run into these issues. Not sure what happened though.

MartyG-RealSense, 2021-07-12:
The multiple udev rules warning typically occurs if librealsense has been installed with both the package and source-code methods on the same computer.  Deleting the rule in the **etc/udev** version of the folder can correct the problem.  

The **lib/udev** rule should not be deleted, as doing so can break linkages to the librealsense packages.

MartyG-RealSense, 2021-07-18:
Hi @MiPlayer123  Do you require further assistance with this case, please?  Thanks!

MiPlayer123, 2021-07-22:
Sorry, I haven't posted anything lately, I got busy. I am getting the normal Jetson Nano (4GB) and I am going to test on that (with CUDA enabled). I will update here if it works or not.

Thank you for your help so far @MartyG-RealSense. It has been really helpful.

MartyG-RealSense, 2021-07-22:
You are very welcome @MiPlayer123 - please do update here when you have test results with Jetson Nano.  Good luck!

MartyG-RealSense, 2021-07-28:
Hi @MiPlayer123  Do you have an update about this case that you can provide, please?  Thanks!

MiPlayer123, 2021-07-28:
Hi @MartyG-RealSense I just go the new jetson and i will update here with any issues in the next couple days

MartyG-RealSense, 2021-07-29:
Thanks very much @MiPlayer123  :)

MiPlayer123, 2021-08-14:
Hi @MartyG-RealSense I was busy before and was not able to test it. I was able to get it to work and I can see the data in rviz. I still get the control transfer error, but its rare and does not affect performance. I have also gotten a T265 tracking camera for reliable VSLAM. I am wondering how I can run both the tracking and depth camera ros nodes. 

I was not able to install with cuda as the documentation was unclear to me. I get a `failed to meet update rate` when running [https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i](https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i), but this is not a major issue due to the tracking camera.

MartyG-RealSense, 2021-08-14:
Thanks @MiPlayer123 for the update.  There is little documentation available about the **Failed to meet update rate! ** error. When ROS does generate this error though, it sometimes provides this advice: "Try decreasing the rate, limiting sensor output frequency, or limiting the number of sensors".

I am not involved in support of the T265 Tracking Camera model and so would recommend creating a new issue on this forum with 'T265' in the message title if you have T265 questions.  You could though try launching a T265 and D435i together with **rs_d400_and_t265.launch**

MartyG-RealSense, 2021-08-20:
Case closed due to D435i performance being declared satisfactory and advice provided to create a T265 question as a separate case.

closed 2021-08-20 by MartyG-RealSense