realsenseai/realsense-ros #1958: RealSense Depth Camera 453i NOT found : RS2_USB_STATUS_BUSY

AlaricLcrs, 2021-06-30:
Hello everyone ! I'd like to access the video stream of my Intel Realsense 435i camera but when I run : **roslaunch realsense2_camera rs_camera.launch**, I get the following error message : 

 30/06 15:41:30,478 ERROR [546668802432] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 30/06 15:41:30,484 WARNING [546769703296] (rs.cpp:304) null pointer passed for argument "device"
[ WARN] [1625038890.484317600]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1625038890.484488384]: The requested device with  is NOT found. Will Try again.

Which is quite weird because it worked on Monday but then when I got back to work yesterday it had stopped working so the configuration hasn't changed. 
And yes I do have the **99-realsense-libusb.rules** in **/etc/udev/rules.d/**

I'm connected to my NVidia board via SSH so I don't know if that changes anything. 

When i run **dmesg**, I get : 

 30/06 15:43:06,086 ERROR [547721556288] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
Could not create device - failed to set power state . Check SDK logs for details

And I have put the **dmesg** response as an attachment an the full message that I get when I run **roslaunch realsense2_camera rs_camera.launch**

I'm on Ubuntu 18.04.

If you need any more info to help me I'll be happy to deliver ! 

Hoping to get an answer !

[dmesg.txt](https://github.com/IntelRealSense/realsense-ros/files/6739151/dmesg.txt)
[roslaunch.txt](https://github.com/IntelRealSense/realsense-ros/files/6739152/roslaunch.txt)

MartyG-RealSense, 2021-06-30:
Hi @AlaricLcrs  If your ROS setup worked on the Monday and you did not make any changes to it then it is reasonable to conclude that it was probably still set up correctly on the Tuesday when it no longer worked.  The roslaunch process does seem to be working correctly up until the point where it tries to detect the camera and fails to do so.

Is the camera detectable if you try to access it directly with a command such as **rs-enumerate-devices** on the computer that it is attached to instead of connecting via SSH?

AlaricLcrs, 2021-07-01:
Just tried and I get exactly the same mistakes if I connect directly to the board

MartyG-RealSense, 2021-07-01:
Are there any signs that the USB cable has been changed for another one by someone whilst you were out of the workplace?  It has been known to happen.

AlaricLcrs, 2021-07-01:
No it's impossible, the board is inside an A1 Robot (https://www.unitree.com/products/a1/) so we don't have access to the USB cable. Furthermore, I'm an intern for one of my teacher and the robot is left inside a locked suitcase so it really is impossible. 
I should however mention that I work on a virtual machine, so if the problem comes from anywhere I think it would be from my configuration because it was after turning it off on Monday and turning it on on Tuesday that I started getting the mistake. So of course I run "source ~/catkin-ws/devel/setup.bash" but are there any other commands that I should be running before the roslaunch command ?

MartyG-RealSense, 2021-07-01:
If you do need to use a virtual machine then Intel recommends doing so with **VMWare Workstation Player**, though the VM system that you were using was working until Tuesday.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#linux-ubuntu-installation

Which VM tool are you using please?

MartyG-RealSense, 2021-07-01:
Ok, thanks very much.  So when you connected the camera directly instead of using ssh and tried rs-enumerate-devices to read the camera's information, was the camera undetected and displayed a message saying **No device detected. Is it plugged in?** or did it provide an information readout?

AlaricLcrs, 2021-07-01:
I have the exact same messages as yesterday, i.e. : 

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [9164]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 931c66ca-d976-11eb-a1e6-48b02d2f6cad
process[rosout-1]: started with pid [9179]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [9182]
process[camera/realsense2_camera-3]: started with pid [9186]
[ INFO] [1625038889.738289312]: Initializing nodelet with 6 worker threads.
[ INFO] [1625038890.397387648]: RealSense ROS v2.2.15
[ INFO] [1625038890.397540800]: Built with LibRealSense v2.38.1
[ INFO] [1625038890.397642560]: Running with LibRealSense v2.38.1
[ INFO] [1625038890.449893088]:  
 30/06 15:41:30,478 ERROR [546668802432] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 30/06 15:41:30,484 WARNING [546769703296] (rs.cpp:304) null pointer passed for argument "device"

MartyG-RealSense, 2021-07-01:
**rs-enumerate-devices** is not a RealSense ROS tool, it is related to the **librealsense** SDK.  It can be used to check if a camera can be detected by the SDK.  As it is not a ROS tool, it can be useful for checking whether a problem is isolated to RealSense ROS or whether the camera has become undetectable in general.

Are there any other RealSense cameras attached to the robot?

AlaricLcrs, 2021-07-01:
No there isn't ! Here's what the **lsusb** command returns : 

[image]

MartyG-RealSense, 2021-07-01:
Thanks very much.  Based on these results and the circumstances in which the problem arose, I will seek advice from Intel about your case.  In the meantime, I would suggest putting the problem aside for now if possible to do so until I have advice to pass on to you.

AlaricLcrs, 2021-07-01:
Okay thank you very much !

MartyG-RealSense, 2021-07-01:
Hi @AlaricLcrs  Intel advised me that the first line of your lsusb log shows that the D435i has been enumerated (detected) and that it is therefore likely a software related issue.  I will continue investigation of the case with you and focus on the software related aspects of your system.

AlaricLcrs, 2021-07-02:
Okay, let's do that !

MartyG-RealSense, 2021-07-02:
I went through the case again from the beginning.  I note that there is not a mention of having tried resetting the camera at launch with **initial_reset:=true**.  For example:

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**

If you have not already tried a launch reset, could you attempt it please to see whether it makes a difference.

AlaricLcrs, 2021-07-05:
Hello ! I just tried it and get the exact same error messages ! I tried both as regular user and as super user

AlaricLcrs, 2021-07-05:
Also I have tried to open the cameras in realsense-viewer but get this error message : 
**UNKNOWN in rs2_create_device (info_list:0x559bc8f670, index:0): failed to set power state**

AlaricLcrs, 2021-07-07:
I think I'm using the first link you sent. It's not the exact same configuration but it's quite close I believe. 
I also tried to remake the librealsense SDK and got this error message : 

[image]

It's strange to get this error message since I managed to build the SDK a first time. I think I'm going to try uninstalling everything to reinstall it afterwards.

MartyG-RealSense, 2021-07-07:
I note that after the CMake instruction it says "Failed to identify internet connection".  The RSUSB backend installation method requires an internet connection, as it installs dependencies over the internet instead of through kernel patching.  

Can you confirm that your computer was able to **successfully** access the internet at the time of this installation?  For example, even if the computer had access to the internet, was Ubuntu successfully logged in to the net?

AlaricLcrs, 2021-07-08:
I'm not sure what you mean. How can Ubuntu not be connected to the Internet if the computer is ? I can go to a web browser and I can ping google via terminal so yes I'm connected to the Internet.

MartyG-RealSense, 2021-07-08:
I have experienced cases where an installation that relied on an internet connection would fail due to being unable to download packages if Ubuntu's network proxy settings were not configured.  That is why I wanted to check this point with you.  Thanks for the confirmation that you can access the internet on your machine.

AlaricLcrs, 2021-07-08:
Oh yes I can download packages on the terminal without any problem

AlaricLcrs, 2021-07-08:
Reinstalling the **librealsense** didn't change anything, still getting the same error messages.

AlaricLcrs, 2021-07-08:
Hello, so my teacher gave me an external RealSense 435i which, when plugged in to the robot, works. I can access the stream on realsense-viewer and I can successfully execute commands like **rs-enumerate-devices**. So it would mean that the configuration of the SDK isn't the problem here but the internal camera is. But what could do that ? Because if it didn't detect the camera at all I'd get a message like "No device detected. Is it plugged in?", wouldn't I ?

MartyG-RealSense, 2021-07-08:
**Failed to set power state** suggests that the camera failed to be **accessed** for some reason when trying to start it.    I would expect "No device detected. Is it plugged in?" if the camera was undetectable.

It looks as though the D435i that is part of the **A1 Robot** is a normal D435i model that is strapped into the front of the robot rather than a specially made version of the camera created by the manufacturer.  

[image]

If the camera is supplied as part of the robot product though then the manufacturer Unitree Robotics would be the best place to make an enquiry about it, as they would be responsible for providing technical support for that camera (as the robot is a non-Intel product).

AlaricLcrs, 2021-07-08:
Okay thanks I'll ask them about it !

AlaricLcrs, 2021-07-12:
I can access the camera now, had to follow this : https://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line
Closing this issue !

closed 2021-07-12 by AlaricLcrs

MartyG-RealSense, 2021-07-12:
Thanks very much @AlaricLcrs for the update and for sharing your solution with the RealSense ROS community!

LeedsRamseyPeng, 2023-08-10:
> I can access the camera now, had to follow this : https://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line Closing this issue !
Hey, I met the same problem as you.  May I ask which command have you tried to solve this issue?