realsenseai/realsense-ros #1959: Increase the fps of the rgb and depth stream

Bannigo, 2021-06-30:
Hello everyone,
                         I am facing an issue regarding the manipulation of the fps parameter while launching the camera node. I am using ROS kinetic on Ubuntu 16.04 and intel realsense d435i. I am working with the launch file rs_rgbd.launch. The current fps is 15 Hz. I want to increase it to 30Hz or even better to 45. I found a [solution](https://github.com/IntelRealSense/realsense-ros/issues/152) to change the mode of the camera to manual. But I am not able to find the mode parameter. I also tried another way to directly change depth_fps & rgb_fps to 35. None of the ways works, can anyone please suggest the perfect way to change the fps? I have directly installed the library using sudo-apt get install. So the only manipulation I can do is while launching the node in the terminal.

Best Regards,
Yash

MartyG-RealSense, 2021-06-30:
Hi @Bannigo  30 FPS is supported by RealSense cameras but not 35 or 45 FPS.  After 30, the next highest supported FPS is 60, though this is not available at 1280x720 resolution (where the maximum is 30).  Details of supported resolutions and FPS speeds for depth, RGB and infrared can be found on page 63 onwards of the current edition of the data sheet document for the 400 Series cameras.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheetCoul

Could you try the rs_rgbd roslaunch instruction below please to see whether it provides you with 30 FPS:

**roslaunch realsense2_camera rs_rgbd.launch depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30**

The above roslaunch instruction, which uses 640x480 resolution at 30 FPS, provides three factors for each stream type that is being custom-defined: height, width and FPS.

If you wish to try 60 FPS then change the '30' in the depth_fps and color_fps instructions to '60'.

Bannigo, 2021-06-30:
Hi @MartyG-RealSense, thank you for your reponse. This solution worked. Thanks!!

MartyG-RealSense, 2021-06-30:
Great news @Bannigo - thanks for the confirmation!

closed 2021-06-30 by Bannigo