realsenseai/realsense-ros #1960: Hardware Notification:Motion Module failure,1.62506e+12,Error,Hardware Error

zheng-xiao, 2021-06-30:
Hi
I have a problem about the hardware.
There is the issue:

roslaunch realsense2_camera rs_camera.launch initial_reset:=true
... logging to /home/zxn/.ros/log/ae0c2040-d978-11eb-8185-08beac02483f/roslaunch-zxn-pc-9410.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://zxn-pc:36211/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
[... 100 lines cut ...]
[ INFO] [1625060306.432907181]: Start publisher IMU
[ INFO] [1625060306.433749831]: setupStreams...
[ INFO] [1625060306.465491959]: insert Depth to Stereo Module
[ INFO] [1625060306.465517612]: insert Color to RGB Camera
[ INFO] [1625060306.465527300]: insert Infrared to Stereo Module
[ INFO] [1625060306.465536836]: insert Infrared to Stereo Module
[ INFO] [1625060306.465546501]: insert Gyro to Motion Module
[ INFO] [1625060306.465554789]: insert Accel to Motion Module
[ WARN] [1625060306.478714861]: 
[ WARN] [1625060306.478757761]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
 30/06 21:38:26,476 WARNING [140208466298624] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 30/06 21:38:26,509 WARNING [140207890343680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 30/06 21:38:26,559 WARNING [140207890343680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 30/06 21:38:26,609 WARNING [140207890343680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1625060306.660707606]: SELECTED BASE:Depth, 0
[ INFO] [1625060306.692552579]: RealSense Node Is Up!
 30/06 21:38:26,737 WARNING [140207890343680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 30/06 21:38:26,787 WARNING [140207890343680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 30/06 21:38:26,939 WARNING [140207890343680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 30/06 21:38:26,989 WARNING [140207890343680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 30/06 21:38:27,049 WARNING [140207890343680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 30/06 21:38:27,101 WARNING [140207890343680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 30/06 21:38:27,151 WARNING [140207890343680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 30/06 21:38:27,202 WARNING [140207890343680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ WARN] [1625060307.355472326]: Hardware Notification:Motion Module failure,1.62506e+12,Error,Hardware Error

MartyG-RealSense, 2021-06-30:
Hi @zheng-xiao  It looks as though you are enabling and publishing every stream type available (depth, color, infrared 1 & 2, gyro and accel).  This could place a burden on the USB system of the computer / computing device being used with the camera.

Which computer / computing device are you using, please (PC, Nvidia Jetson, Raspberry Pi, etc)

MartyG-RealSense, 2021-07-07:
Hi @zheng-xiao  Do you require further assistance with this case, please?  Thanks!

zheng-xiao, 2021-07-07:
Yes，
I don't solve it.
And I find another issue. When I record a rosbag, the topic called /camera/infra1/image_rect_raw will lose frame. If I play the rosbag, the image is not smooth.
If you can solve my issue, I will be verygrateful.
Thanks.

MartyG-RealSense, 2021-07-07:
Whic topics are you recording to a bag please?  Are you recording particular topics, or recording all currently available topics using **rosbag record -a** or **rosbag record --all** ?

zheng-xiao, 2021-07-07:
Hi ，
Thank you for answering my question. I use three topics include /camera/infra1/image_rect_raw, /camera/infra2/image_rect_raw,/camera/imu. And I record the rosbag using rosbag record -a.

MartyG-RealSense, 2021-07-07:
Could you try limiting the rosbag record command to these three specific topics instead of using the '--all' command please to see whether it improves performance?

**rosbag record /camera/infra1/image_rect_raw /camera/infra2/image_rect_raw /camera/imu**

zheng-xiao, 2021-07-08:
Hi,
Thanks. I solve it.

MartyG-RealSense, 2021-07-08:
Great news @zheng-xiao that you found a solution - thanks for the update!

closed 2021-07-14 by MartyG-RealSense