realsenseai/realsense-ros #1962: /camera/color/image_raw topic not publishing messages when pointcloud is enabled

Nick-Kou, 2021-07-01:
For the current project I am working on I am using a Raspberry Pi 3B+ with the Ros Noetic Image taken from: https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/

Details are: 

- Operating System: Ubuntu 20.04.2 LTS
-  Kernel: Linux 5.4.0-1039-raspi
-  Architecture: arm64

After successfully the SDK:
```
cd ~/librealsense
mkdir  build  && cd build
cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DFORCE_LIBUVC=true
make -j1
sudo make install
```
And then: `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`

I managed to operate the Intel Realsense D435i camera.

When I do a simple launch:
`roslaunch realsense2_camera rs_camera.launch `
The /camera/color/image_raw topic seems to be publishing fine at the desired FPS shown in rviz.

However, when I to run with point cloud enabled for example:
`roslaunch realsense2_camera rs_camera.launch filters:=pointcloud depth_width:=424 color_width:=424 depth_height:=240 color_height:=240 depth_fps:=15 color_fps:=15 pointcloud_texture_stream:=RS2_STREAM_ANY`

The pointcloud2 messages are being published, however, the /camera/color/image_raw topic doesn't seem to be publishing anymore.

I was wondering why the topics (point cloud and camera image) can't be published together.

[image]

roslaunch realsense2_camera rs_camera.launch filters:=pointcloud depth_width:=424 color_width:=424 depth_height:=240 color_height:=240 depth_fps:=15 color_fps:=15 pointcloud_texture_stream:=RS2_STREAM_ANY
... logging to /home/ubuntu/.ros/log/de7d655e-da4a-11eb-9870-b92bd22f468a/roslaunch-ubuntu-3539.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.2.144:41675/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 240
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 424
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 15
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 240
[... 121 lines cut ...]
 01/07 09:01:42,230 WARNING [281472158134640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 09:01:42,280 WARNING [281472158134640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 09:01:42,331 WARNING [281472158134640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 09:01:42,381 WARNING [281472158134640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 09:01:42,432 WARNING [281472158134640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 09:01:42,483 WARNING [281472158134640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 09:01:42,533 WARNING [281472158134640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 09:01:42,585 WARNING [281472158134640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 09:01:42,636 WARNING [281472158134640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 09:01:42,686 WARNING [281472158134640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 09:01:42,737 WARNING [281472158134640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 09:01:42,788 WARNING [281472158134640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 09:01:42,838 WARNING [281472158134640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 09:01:42,889 WARNING [281472158134640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 09:01:42,940 WARNING [281472158134640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 09:01:42,990 WARNING [281472158134640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 09:01:43,041 WARNING [281472158134640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 09:01:43,092 WARNING [281472158134640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 09:01:43,142 WARNING [281472158134640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/07 09:01:43,193 WARNING [281472158134640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
^C[camera/realsense2_camera-2] killing on exit
[camera/realsense2_camera_manager-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Nick-Kou, 2021-07-01:
Also on this specific image that I have on the SD card `realsense-viewer` doesn't open anything, however we can confirm the camera works through ROS wrapper when running:
`roslaunch realsense2_camera rs_camera.launch`
 
And the point cloud messages are being published when running:
`
roslaunch realsense2_camera rs_camera.launch filters:=pointcloud depth_width:=424 color_width:=424 depth_height:=240 color_height:=240 depth_fps:=15 color_fps:=15 pointcloud_texture_stream:=RS2_STREAM_ANY`

But when running: `
roslaunch realsense2_camera rs_camera.launch filters:=pointcloud depth_width:=424 color_width:=424 depth_height:=240 color_height:=240 depth_fps:=15 color_fps:=15 pointcloud_texture_stream:=RS2_STREAM_ANY`

No camera images are being published at from the /camera/color/image_raw topic.

MartyG-RealSense, 2021-07-01:
Hi @Nick-Kou  Could you try using **RS2_STREAM_COLOR** instead of RS2_STREAM_ANY please and check whether doing so makes a positive difference?

Nick-Kou, 2021-07-02:
https://github.com/IntelRealSense/realsense-ros/issues/1938#issue-925127468. With RS2_STREAM_COLOR the same issue 1938 persists.

MartyG-RealSense, 2021-07-03:
The RealSense ROS wrapper's documentation states that using the **pointcloud** filter should cause a topic called **/camera/depth/color/points** to be added.  Are you able to obtain any data from that topic?

Nick-Kou, 2021-07-04:
After running: `roslaunch realsense2_camera rs_camera.launch camera:=camera1 filters:=pointcloud depth_width:=424 color_width:=424 depth_height:=240 color_height:=240 depth_fps:=30 color_fps:=30 pointcloud_texture_stream:=RS2_STREAM_COLOR`

A [ WARN] [1625357506.310201328]: No stream match for pointcloud chosen texture Process - Color was shown in the output summary as shown below towards the end of the log:

ubuntu@ubuntu:~$ roslaunch realsense2_camera rs_camera.launch camera:=camera1 filters:=pointcloud depth_width:=424 color_width:=424 depth_height:=240 color_height:=240 depth_fps:=30 color_fps:=30 pointcloud_texture_stream:=RS2_STREAM_COLOR
... logging to /home/ubuntu/.ros/log/a6c9be4c-dc5b-11eb-bf11-afe8a803aefa/roslaunch-ubuntu-2578.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.2.144:36579/

SUMMARY
========

PARAMETERS
 * /camera1/realsense2_camera/accel_fps: -1
 * /camera1/realsense2_camera/accel_frame_id: camera1_accel_frame
 * /camera1/realsense2_camera/accel_optical_frame_id: camera1_accel_opt...
 * /camera1/realsense2_camera/align_depth: False
 * /camera1/realsense2_camera/aligned_depth_to_color_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/allow_no_texture_points: False
 * /camera1/realsense2_camera/base_frame_id: camera1_link
 * /camera1/realsense2_camera/calib_odom_file: 
 * /camera1/realsense2_camera/clip_distance: -2.0
 * /camera1/realsense2_camera/color_fps: 30
 * /camera1/realsense2_camera/color_frame_id: camera1_color_frame
 * /camera1/realsense2_camera/color_height: 240
 * /camera1/realsense2_camera/color_optical_frame_id: camera1_color_opt...
 * /camera1/realsense2_camera/color_width: 424
 * /camera1/realsense2_camera/confidence_fps: -1
 * /camera1/realsense2_camera/confidence_height: -1
 * /camera1/realsense2_camera/confidence_width: -1
 * /camera1/realsense2_camera/depth_fps: 30
 * /camera1/realsense2_camera/depth_frame_id: camera1_depth_frame
 * /camera1/realsense2_camera/depth_height: 240
 * /camera1/realsense2_camera/depth_optical_frame_id: camera1_depth_opt...
 * /camera1/realsense2_camera/depth_width: 424
 * /camera1/realsense2_camera/device_type: 
 * /camera1/realsense2_camera/enable_accel: False
 * /camera1/realsense2_camera/enable_color: True
 * /camera1/realsense2_camera/enable_confidence: True
 * /camera1/realsense2_camera/enable_depth: True
 * /camera1/realsense2_camera/enable_fisheye1: False
 * /camera1/realsense2_camera/enable_fisheye2: False
 * /camera1/realsense2_camera/enable_fisheye: False
 * /camera1/realsense2_camera/enable_gyro: False
 * /camera1/realsense2_camera/enable_infra1: False
 * /camera1/realsense2_camera/enable_infra2: False
 * /camera1/realsense2_camera/enable_infra: False
 * /camera1/realsense2_camera/enable_pointcloud: False
 * /camera1/realsense2_camera/enable_pose: False
 * /camera1/realsense2_camera/enable_sync: False
 * /camera1/realsense2_camera/filters: pointcloud
 * /camera1/realsense2_camera/fisheye1_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye1_optical_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye2_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye2_optical_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye_fps: -1
 * /camera1/realsense2_camera/fisheye_frame_id: camera1_fisheye_f...
 * /camera1/realsense2_camera/fisheye_height: -1
 * /camera1/realsense2_camera/fisheye_optical_frame_id: camera1_fisheye_o...
 * /camera1/realsense2_camera/fisheye_width: -1
 * /camera1/realsense2_camera/gyro_fps: -1
 * /camera1/realsense2_camera/gyro_frame_id: camera1_gyro_frame
[... 108 lines cut ...]
 04/07 00:11:34,204 WARNING [281472640098672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/07 00:11:34,255 WARNING [281472640098672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/07 00:11:34,406 WARNING [281472640098672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/07 00:11:34,456 WARNING [281472640098672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1625357505.758857334]: publishing unordered pointcloud.
[ WARN] [1625357506.310201328]: No stream match for pointcloud chosen texture Process - Color
 04/07 00:12:14,156 WARNING [281472640098672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 04/07 00:12:14,206 WARNING [281472640098672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 04/07 00:12:15,258 WARNING [281472640098672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
rostopic echo /camera1/depth/color/points shown in the image below shows no PointCloud2 messages are being  published.
[image]
 The reason it's /camera1 is becasue I changed the namespace when I called the:

 `roslaunch realsense2_camera rs_camera.launch camera:=camera1 filters:=pointcloud depth_width:=424 color_width:=424 depth_height:=240 color_height:=240 depth_fps:=30 color_fps:=30 pointcloud_texture_stream:=RS2_STREAM_COLOR`

in the camera:=camera1 parameter which also shouldn't have any effect.

MartyG-RealSense, 2021-07-04:
So in summary, when you launch with **roslaunch realsense2_camera rs_camera.launch** then the **/camera/color/image_raw** topic is published correctly.

But the **/camera/color/image_raw** topic is not published when you perform a pointcloud-equipped launch with the roslaunch instruction below:

**roslaunch realsense2_camera rs_camera.launch filters:=pointcloud depth_width:=424 color_width:=424 depth_height:=240 color_height:=240 depth_fps:=15 color_fps:=15 pointcloud_texture_stream:=RS2_STREAM_ANY**

However, **pointcloud2** messages are publishing correctly.

Going through the launch log, I note that the launch decided that setting depth to 424x240 at 30 FPS was invalid and instead set it to 640x480 at 15 FPS (whilst color remained at the 424x240 set by the custom-resolution roslaunch instruction).  So you now had a mismatch of both resolution and FPS between depth and color.

[image]

This was likely caused because the camera is on a USB 2 connection due to the use of a Raspberry Pi 3B+ (Pi models before Pi 4 only had USB 2 instead of USB 3).

[image]

RealSense cameras on a USB 2 connection have a limited range of supported resolution and FPS modes compared to the full range of capabilities on a USB 3 connection.  This is described in Intel's document about USB 2 operation at the link below.

https://www.intelrealsense.com/usb2-support-for-intel-realsense-technology/

The list of depth and color resolution and FPS modes currently supported on USB 2 with a D435i at the time of writing this is shown in the image below, which illustrates that whilst 424x270 color is supported, 424x270 depth is not.  I would therefore suggest trying a roslaunch with both color and depth set to 640x480 at 30 FPS.

[image]

MartyG-RealSense, 2021-07-05:
A common factor that I identified between this case and two other cases that had no Color Transformer options available was that they were on computing boards with Arm processors (your one on Pi 3B+, and the other two on Nvidia Jetson NX).

https://github.com/IntelRealSense/realsense-ros/issues/1967#issuecomment-874084975

chrisdalke, 2021-07-08:
Posted this on #1967 as well since these seem to be related:

I've encountered this same issue with a D415 on a RPI 4; I'm able to view the depth image but the RGB image and color point clouds do not get published. Inspecting the /camera/color/image_raw and /camera/depth/color/points topics shows nothing being published on either topic.
This issue showed up recently, so I reverted to a manually-built earlier version of librealsense and realsense-ros which fixed the issue. Maybe there's a regression or incompatibility with the recent libraries on ARM?

**I reverted to v2.41.0 of librealsense, and version 2.2.21 of realsense-ros and rebuilt from git on the Raspberry Pi. This combination does not exhibit this issue.**

Nick-Kou, 2021-07-08:
@MartyG-RealSense Thanks for the response. However, I will only be in the lab on Saturday again to try this.
@chrisdalke Thanks, will definitely give this a try. Does it also take you several hours to cmake build the librealsense library on the Raspberry Pi?

chrisdalke, 2021-07-08:
@Nick-Kou Yeah, it certainly takes a while, about ~1hr for me on a RPi 4B, but that seems typical for a large project. You could probably reduce the compile time by not compiling any of the examples/tools.

Make sure you're using a decent SD card, good power supply, sufficient cooling, just to eliminate any potential throttling. I just left the compilation running in the background and came back later :)

MartyG-RealSense, 2021-07-09:
Thanks again for your advice @chrisdalke   :)

@Nick-Kou  I look forward to your  update about your lab results after Saturday.  Good luck!

Nick-Kou, 2021-07-10:
@MartyG-RealSense 

After setting 640x480 and 30fps for each I obtain the following summary:

roslaunch realsense2_camera rs_camera.launch camera:=camera1 filters:=pointcloud depth_width:=640 color_width:=640 depth_height:=480 color_height:=480 depth_fps:=30 color_fps:=30 pointcloud_texture_stream:=RS2_STREAM_ANY
... logging to /home/ubuntu/.ros/log/e1423ed6-e1ca-11eb-969a-d38632247d1a/roslaunch-ubuntu-2239.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.113:45327/

SUMMARY
========

PARAMETERS
 * /camera1/realsense2_camera/accel_fps: -1
 * /camera1/realsense2_camera/accel_frame_id: camera1_accel_frame
 * /camera1/realsense2_camera/accel_optical_frame_id: camera1_accel_opt...
 * /camera1/realsense2_camera/align_depth: False
 * /camera1/realsense2_camera/aligned_depth_to_color_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/allow_no_texture_points: False
 * /camera1/realsense2_camera/base_frame_id: camera1_link
 * /camera1/realsense2_camera/calib_odom_file: 
 * /camera1/realsense2_camera/clip_distance: -2.0
 * /camera1/realsense2_camera/color_fps: 30
 * /camera1/realsense2_camera/color_frame_id: camera1_color_frame
 * /camera1/realsense2_camera/color_height: 480
 * /camera1/realsense2_camera/color_optical_frame_id: camera1_color_opt...
 * /camera1/realsense2_camera/color_width: 640
 * /camera1/realsense2_camera/confidence_fps: -1
 * /camera1/realsense2_camera/confidence_height: -1
 * /camera1/realsense2_camera/confidence_width: -1
 * /camera1/realsense2_camera/depth_fps: 30
 * /camera1/realsense2_camera/depth_frame_id: camera1_depth_frame
 * /camera1/realsense2_camera/depth_height: 480
 * /camera1/realsense2_camera/depth_optical_frame_id: camera1_depth_opt...
 * /camera1/realsense2_camera/depth_width: 640
 * /camera1/realsense2_camera/device_type: 
 * /camera1/realsense2_camera/enable_accel: False
 * /camera1/realsense2_camera/enable_color: True
 * /camera1/realsense2_camera/enable_confidence: True
 * /camera1/realsense2_camera/enable_depth: True
 * /camera1/realsense2_camera/enable_fisheye1: False
 * /camera1/realsense2_camera/enable_fisheye2: False
 * /camera1/realsense2_camera/enable_fisheye: False
 * /camera1/realsense2_camera/enable_gyro: False
 * /camera1/realsense2_camera/enable_infra1: False
 * /camera1/realsense2_camera/enable_infra2: False
 * /camera1/realsense2_camera/enable_infra: False
 * /camera1/realsense2_camera/enable_pointcloud: False
 * /camera1/realsense2_camera/enable_pose: False
 * /camera1/realsense2_camera/enable_sync: False
 * /camera1/realsense2_camera/filters: pointcloud
 * /camera1/realsense2_camera/fisheye1_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye1_optical_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye2_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye2_optical_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye_fps: -1
 * /camera1/realsense2_camera/fisheye_frame_id: camera1_fisheye_f...
 * /camera1/realsense2_camera/fisheye_height: -1
 * /camera1/realsense2_camera/fisheye_optical_frame_id: camera1_fisheye_o...
 * /camera1/realsense2_camera/fisheye_width: -1
 * /camera1/realsense2_camera/gyro_fps: -1
 * /camera1/realsense2_camera/gyro_frame_id: camera1_gyro_frame
[... 80 lines cut ...]
[ INFO] [1625954845.876345404]: setupPublishers...
[ INFO] [1625954845.886944465]: Expected frequency for depth = 30.00000
[ INFO] [1625954845.916337534]: Expected frequency for color = 30.00000
[ INFO] [1625954845.925305919]: setupStreams...
[ INFO] [1625954845.938526386]: insert Depth to Stereo Module
[ INFO] [1625954845.938916594]: insert Color to RGB Camera
 10/07 22:07:26,049 WARNING [281472493678976] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1625954846.100850012]: SELECTED BASE:Depth, 0
[ INFO] [1625954846.154896569]: RealSense Node Is Up!
[ WARN] [1625954846.195683543]: 
 10/07 22:07:26,198 WARNING [281472493678976] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/07 22:07:26,248 WARNING [281472493678976] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/07 22:07:26,300 WARNING [281472493678976] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/07 22:07:26,455 WARNING [281472493678976] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/07 22:07:26,505 WARNING [281472493678976] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/07 22:07:26,557 WARNING [281472493678976] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 10/07 22:07:26,658 WARNING [281472493678976] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/07 22:07:26,708 WARNING [281472493678976] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1625954893.215535785]: publishing unordered pointcloud.
 10/07 22:09:08,570 ERROR [281472241996160] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132

[image]

Point Cloud Message appears in rviz however image_raw topic is still not received, however, the image_rect_raw topic is received.

Nick-Kou, 2021-07-10:
After trying the same except the RS2_STREAM_COLOR, the pointcloud message status is fine however is not displaying anything in rviz as mentioned before. Also the color/image_raw topic is still not being received as shown in the left panel of rviz.
roslaunch realsense2_camera rs_camera.launch camera:=camera1 filters:=pointcloud depth_width:=640 color_width:=640 depth_height:=480 color_height:=480 depth_fps:=30 color_fps:=30 pointcloud_texture_stream:=RS2_STREAM_COLOR
... logging to /home/ubuntu/.ros/log/e1423ed6-e1ca-11eb-969a-d38632247d1a/roslaunch-ubuntu-2446.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.113:37785/

SUMMARY
========

PARAMETERS
 * /camera1/realsense2_camera/accel_fps: -1
 * /camera1/realsense2_camera/accel_frame_id: camera1_accel_frame
 * /camera1/realsense2_camera/accel_optical_frame_id: camera1_accel_opt...
 * /camera1/realsense2_camera/align_depth: False
 * /camera1/realsense2_camera/aligned_depth_to_color_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/allow_no_texture_points: False
 * /camera1/realsense2_camera/base_frame_id: camera1_link
 * /camera1/realsense2_camera/calib_odom_file: 
 * /camera1/realsense2_camera/clip_distance: -2.0
 * /camera1/realsense2_camera/color_fps: 30
 * /camera1/realsense2_camera/color_frame_id: camera1_color_frame
 * /camera1/realsense2_camera/color_height: 480
 * /camera1/realsense2_camera/color_optical_frame_id: camera1_color_opt...
 * /camera1/realsense2_camera/color_width: 640
 * /camera1/realsense2_camera/confidence_fps: -1
 * /camera1/realsense2_camera/confidence_height: -1
 * /camera1/realsense2_camera/confidence_width: -1
 * /camera1/realsense2_camera/depth_fps: 30
 * /camera1/realsense2_camera/depth_frame_id: camera1_depth_frame
 * /camera1/realsense2_camera/depth_height: 480
 * /camera1/realsense2_camera/depth_optical_frame_id: camera1_depth_opt...
 * /camera1/realsense2_camera/depth_width: 640
 * /camera1/realsense2_camera/device_type: 
 * /camera1/realsense2_camera/enable_accel: False
 * /camera1/realsense2_camera/enable_color: True
 * /camera1/realsense2_camera/enable_confidence: True
 * /camera1/realsense2_camera/enable_depth: True
 * /camera1/realsense2_camera/enable_fisheye1: False
 * /camera1/realsense2_camera/enable_fisheye2: False
 * /camera1/realsense2_camera/enable_fisheye: False
 * /camera1/realsense2_camera/enable_gyro: False
 * /camera1/realsense2_camera/enable_infra1: False
 * /camera1/realsense2_camera/enable_infra2: False
 * /camera1/realsense2_camera/enable_infra: False
 * /camera1/realsense2_camera/enable_pointcloud: False
 * /camera1/realsense2_camera/enable_pose: False
 * /camera1/realsense2_camera/enable_sync: False
 * /camera1/realsense2_camera/filters: pointcloud
 * /camera1/realsense2_camera/fisheye1_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye1_optical_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye2_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye2_optical_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye_fps: -1
 * /camera1/realsense2_camera/fisheye_frame_id: camera1_fisheye_f...
 * /camera1/realsense2_camera/fisheye_height: -1
 * /camera1/realsense2_camera/fisheye_optical_frame_id: camera1_fisheye_o...
 * /camera1/realsense2_camera/fisheye_width: -1
 * /camera1/realsense2_camera/gyro_fps: -1
 * /camera1/realsense2_camera/gyro_frame_id: camera1_gyro_frame
 * /camera1/realsense2_camera/gyro_optical_frame_id: camera1_gyro_opti...
 * /camera1/realsense2_camera/imu_optical_frame_id: camera1_imu_optic...
 * /camera1/realsense2_camera/infra1_frame_id: camera1_infra1_frame
[... 76 lines cut ...]
[ INFO] [1625956242.911393203]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1625956242.911625338]: setupPublishers...
[ INFO] [1625956242.919915755]: Expected frequency for depth = 30.00000
[ INFO] [1625956242.935631277]: Expected frequency for color = 30.00000
[ INFO] [1625956242.945658934]: setupStreams...
[ INFO] [1625956242.958820758]: insert Depth to Stereo Module
[ INFO] [1625956242.959193674]: insert Color to RGB Camera
 10/07 22:30:43,068 WARNING [281472596595072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1625956243.120106600]: SELECTED BASE:Depth, 0
[ INFO] [1625956243.173605197]: RealSense Node Is Up!
[ WARN] [1625956243.215787179]: 
 10/07 22:30:43,217 WARNING [281472596595072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/07 22:30:43,268 WARNING [281472596595072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/07 22:30:43,319 WARNING [281472596595072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 10/07 22:30:43,625 WARNING [281472596595072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/07 22:30:43,676 WARNING [281472596595072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ WARN] [1625956244.147019317]: No stream match for pointcloud chosen texture Process - Color
 10/07 22:30:53,319 ERROR [281472074240384] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132

[image]

Nick-Kou, 2021-07-10:
I guess I should give @chrisdalke's suggestion a shot as it seemed to have worked for him. 

Thanks for all the suggestions @MartyG-RealSense @chrisdalke

MartyG-RealSense, 2021-07-11:
You are very welcome @Nick-Kou - good luck, and please do let us know how it turns out!

lupusorina, 2021-07-15:
Hey. I have a similar issue. I have a Raspi4, running Ubuntu 20.04, kernel 5.4 connected to a D435 via USB3. ROS distribution: noetic

I'm running the following command
`roslaunch realsense2_camera rs_camera.launch camera:=camera1 filters:=pointcloud depth_width:=640 color_width:=640 depth_height:=480 color_height:=480 depth_fps:=30 color_fps:=30 pointcloud_texture_stream:=RS2_STREAM_ANY`

When doing so, I am able to stream in RVIZ the pointclouds, but they don't have color. In addition, streaming /camera1/color/image_raw in RVIZ shows the image for 2-3 seconds, then it freezes, and the output of the command line prints:
` 15/07 16:35:12,398 ERROR [281472745329008] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132`
Although the camera1/color/image_raw freezes, the point cloud is still being published. 

[image]

I have tried to downgrade to 15 fps for the color, but nothing changed. 

Also, sometimes, this printout appears, although I am not sure the issues are related:
` 15/07 16:49:02,888 ERROR [281471713517936] (ds5-options.cpp:88) Asic Temperature value is not valid!`

I then tried to run the command with pointcloud_texture_stream:=RS2_STREAM_COLOR:

`roslaunch realsense2_camera rs_camera.launch camera:=camera1 filters:=pointcloud depth_width:=640 color_width:=640 depth_height:=480 color_height:=480 depth_fps:=30 color_fps:=30 pointcloud_texture_stream:=RS2_STREAM_COLOR`

Then, the pointclouds are not published at all and the following warning appears:
`[ INFO] [1626367933.517223488]: publishing unordered pointcloud.
[ WARN] [1626367933.948946733]: No stream match for pointcloud chosen texture Process - Color`

I guess I could downgrade to an earlier version @chrisdalke suggested, but it will be great if this issue can be fixed for the latest updates and versions.

MartyG-RealSense, 2021-07-16:
Hi @lupusorina   Under the **PointCloud2** options in RViz, do you have any **Color Transformer** options available or are they absent like they were for @Nick-Kou please?

https://github.com/IntelRealSense/realsense-ros/issues/1967#issuecomment-873375954

If the temperature can be read normally most of the time and the **Asic Temperature value is not valid!** warning only appears sometimes then it is likely just a timing hiccup where the camera was not ready to respond at that particular instance when the temperature was being polled.  You can monitor Asic, Projector and Motion Module temperatures using the **/diagnostics** topic of the RealSense ROS wrapper.

MartyG-RealSense, 2021-07-22:
Hi @Nick-Kou and @lupusorina  Do either of you require further assistance with this case, please?  Thanks!

Nick-Kou, 2021-07-23:
Hi @MartyG-RealSense I didn't really get a chance to try the older libraries, but I assume they should work. Thanks for all the assistance.

lupusorina, 2021-07-23:
@MartyG-RealSense I do not have any Color Transformer, as well.

I tried the same setup using a Jetson Nano. It seems that with the RS2_STREAM_COLOR, the pointclouds are not published, but with RS2_STREAM_ANY, they are black and white, as they were on the Raspberry Pi. So same thing happens. The librealsense is OK because I tried the pointcloud example, which works.

Thank you!

MartyG-RealSense, 2021-07-23:
Hi @lupusorina  In another case on Jetson where there were no Color Transformer options available, **Doronhi** the RealSense ROs wrapper developer commented "The issue definitely seems related to the fact that no color images reach realsense2_camera driver".

https://github.com/IntelRealSense/realsense-ros/issues/1967#issuecomment-874084975

In a comment directly below the linked-to one above, a RealSense ROS user who was a Raspberry Pi user (which has an Arm processor like Jetson boards do) said that they had the same problem and solved it by manually building the earlier **2.41.0** version of librealsense and the matching **2.2.21** version of the RealSense ROS wrapper.

https://github.com/IntelRealSense/realsense-ros/issues/1967#issuecomment-876615270

lupusorina, 2021-07-23:
@MartyG-RealSense the latest driver librealsense definitely outputs colored pointclouds. I tried to run on the Jetson nano

`./pointcloud/rs-pointcloud`

See the output below:

[image]

So I guess the issue is on the ros driver.

MartyG-RealSense, 2021-07-29:
Hi @lupusorina  Do you still require assistance with the problem above, please?  Thanks!

lupusorina, 2021-07-29:
@MartyG-RealSense thank you for your help. I think the problem was not solved yet. My workaround was to use a Jetson Nano and try to use the black and white pointclouds, but it will be great if the RGB would work. "building the earlier 2.41.0 version of librealsense and the matching 2.2.21 version of the RealSense ROS wrapper" didn't really work for me, but I can try more.

MartyG-RealSense, 2021-07-29:
Thanks very much @lupusorina for the update - please do post the results of further tests here.   Good luck!

MartyG-RealSense, 2021-08-15:
Case closed due to no further comments received.

closed 2021-08-15 by MartyG-RealSense

kaushal07wick, 2023-01-19:
I have a similar problem, but in sitl.

/fastplanner_ws$ rostopic info camera/depth/image_raw

Type: sensor_msgs/Image

Publishers: None

Subscribers: 
 * /rviz (http://leonidas:42003/)

no data in rviz, how to resolve it.