realsenseai/realsense-ros #1963: Confusion about camera mounts (3d printed parts, stl files)

Combinacijus, 2021-07-01:
There are two 3d printable mounts: [mount1](https://github.com/IntelRealSense/realsense-ros/blob/occupancy-mapping/realsense2_camera/meshes/bracket_t265nd435_external.stl) and [mount2](https://github.com/IntelRealSense/realsense-ros/blob/occupancy-mapping/realsense2_camera/meshes/mount_t265_d435.stl)
And I'm not quite sure which one should I use or how it should be mounted to the robot or cameras to the mount. Could you provide example photos of how it might look? This would help decide which one to choose before printing.

In this [URDF file](https://github.com/IntelRealSense/realsense-ros/blob/occupancy-mapping/realsense2_camera/urdf/mount_t265_d435.urdf.xacro) a relative pose between cameras is defined but it's not clear is it true for both mounts or if not for which one?

This info could be added to README too

RealSenseCustomerSupport, 2021-07-07:
----------------------------------------------

I guess this page might be helpful to you. Please check it out.
https://dev.intelrealsense.com/docs/depth-and-tracking-cameras-alignment

closed 2021-07-09 by Combinacijus

Combinacijus, 2021-07-10:
Accidentally found image for other mount too <https://www.intelrealsense.com/depth-and-tracking-combined-get-started/>