realsenseai/realsense-ros #1965: rosbag record buffer exceeded

aditdoshi333, 2021-07-02:
Hello,

I am trying to record realsense L515 data into a .bag file. The folllowing is my launch command:

    ```
<include file="$(find realsense2_camera)/launch/rs_camera.launch">
        <arg name="align_depth" value="true"/>
        <arg name="unite_imu_method" value="linear_interpolation"/>
        <arg name="enable_gyro" value="true"/>
        <arg name="enable_accel" value="true"/>
        <arg name="enable_infra" value="true"/>
        <arg name="filters" value="pointcloud"/>
    </include>
```
 The following are the topics I am recording in a bag file:
   
/camera/accel/imu_info 
/camera/align_to_color/parameter_descriptions 
/camera/align_to_color/parameter_updates 
/camera/aligned_depth_to_color/camera_info 
/camera/aligned_depth_to_color/image_raw 
/camera/color/camera_info 
/camera/color/image_raw /camera/color/image_raw/compressed 
/camera/depth/camera_info 
/camera/depth/color/points /camera/depth/image_rect_raw 
/camera/extrinsics/depth_to_color 
/camera/gyro/imu_info /camera/imu 
/camera/infra/camera_info 
/camera/infra/image_raw 
/camera/l500_depth_sensor/parameter_descriptions 
/camera/l500_depth_sensor/parameter_updates 
/camera/motion_module/parameter_descriptions 
/camera/motion_module/parameter_updates 
/camera/pointcloud/parameter_descriptions 
/camera/pointcloud/parameter_updates 
/camera/realsense2_camera_manager/bond 
/camera/rgb_camera/parameter_descriptions 
/camera/rgb_camera/parameter_updates 
/diagnostics 
/rosout 
/rosout_agg 
/tf 
/tf_static

After few seconds I am getting the following error:

> [ WARN] [1625201375.003035664]: rosbag record buffer exceeded.  Dropping oldest queued message.

I am using hardware with 2TB SSD and 16 GB ram. Is there any workaround possible for the problem? If yes please let me know and also I am not sure am I really losing data due to this error msg? I am not really what's the effect of this error on the collected bag file.

Thank you

MartyG-RealSense, 2021-07-02:
Hi @aditdoshi333  There are cases of this error that dates back to Microsoft Kinect cameras in 2013-14, indicating that it is a ROS issue and not a problem with the RealSense ROS wrapper.  

It is important to ensure that the drive path that is being recorded to has sufficient available space to record the file.   It sounds as though this is not the cause of your problem though.  And even though rosbag recordings can become huge in size over time, a file should not be very large after only a few seconds.

You could instead try using the '-b' condition of rosbag record to increase the buffer size.

[image]

aditdoshi333, 2021-07-02:
Thanks for the reply @MartyG-RealSense . I am trying to record a large bag file (more than 100 GB). Is it not advisable?

MartyG-RealSense, 2021-07-02:
You should be able to record a bag of that size and you have a suitably large capacity drive to do so.   Some people though prefer to avoid doing so because it carries the risk of losing the entire recording if something goes wrong during recording.  rosbag record allows recordings to be automatically split into multiple files, closing the current recording and starting the next one when a defined type of limit is reached (maximum recording duration, file size, etc).

[image]

closed 2021-07-15 by MartyG-RealSense