realsenseai/realsense-ros #1966: Unknown Filter. Using multiple filters like pointcloud, decimation etc.

Combinacijus, 2021-07-02:
I already found the solution but writing it for documentation purposes

So the problem was if I tried using multiple filters in [rs_camera.launch](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_camera.launch) file the second filter was not recognized:
```
<arg name="filters"                   default="pointcloud, decimation"/>
```
Error:
```
[ERROR] [1625234460.526932960]: Unknown Filter:  decimation
```

SOLUTION is to list filters separated ONLY by commas no spaces allowed:
```
<arg name="filters"                   default="pointcloud,decimation"/>
```

At least for me in programming it's usual to put spaces after commas so that code looks tidier it took some time to find the problem.
I wonder is this intended behavior? Is it some xml list data type that has to be defined without spaces or this is problem with realsense string parsing code?

In [README](https://github.com/IntelRealSense/realsense-ros/blob/development/README.md#launch-parameters) it's told that parameters has to be separated by commas but it's not explicitly said that it has to have no spaces.
```
filters: any of the following options, separated by commas:
```

MartyG-RealSense, 2021-07-02:
Hi @Combinacijus  Thanks very much for sharing your feedback about defining filters!

@doronhi  Could you provide @Combinacijus with advice please about the reason for not having spaces after commas when defining a list of filters?  Thanks!

MartyG-RealSense, 2021-07-09:
Hi @doronhi   Could you provide @Combinacijus with advice please about the reason for not having spaces after commas when defining a list of filters? Thanks!

Combinacijus, 2021-07-20:
Took a look at it myself and it seems it was a problem with how strings were split and iterated. [Made a pull request](https://github.com/IntelRealSense/realsense-ros/pull/1988) which removes spaces before checking which filter is selected. Not sure if it's right and correct way to do it but checked with "pointcloud, decimation" and it worked

MartyG-RealSense, 2021-07-20:
Excellent, @Combinacijus - thanks for the PR!  I have added the **Enhancement** tag to this case to signify that there is a PR associated with it.  This case shold be kept open whilst that PR is active.  Thanks again   :)

MartyG-RealSense, 2023-07-18:
Case closed due to PR being merged into the RealSense ROS1 wrapper on July 25, 2021.

closed 2023-07-18 by MartyG-RealSense