realsenseai/realsense-ros #1967: No pointcloud is published and RGB image is stuck.

vincent51689453, 2021-07-03:
Currently, I am using NVIDIA Jetson Xavier NX with JetPack 4.4. I can use realsense-viewer to capture color and depth image of D435i. However, when I want to publish realsense pointcloud and image in ROS Melodic. No point cloud can be displayed and the color image is stuck after several receives,
[image]
[image]
[image]

MartyG-RealSense, 2021-07-03:
Hi @vincent51689453  Do you receive a point cloud if the method outlined in the pointcloud example of the ROS wrapper documentation is used, please (launching with **roslaunch realsense2_camera rs_camera.launch filters:=pointcloud** and setting Fixed Frame in RViz to **camera_link**)

https://github.com/IntelRealSense/realsense-ros#point-cloud

vincent51689453, 2021-07-03:
Yes. This is the way for me to publish the point cloud. However, nothing is displayed in RVIZ. Meanwhile, the RGB image stopped displaying after a while.

MartyG-RealSense, 2021-07-03:
Could you try the following steps please:

1.  In the **PointCloud2** options of RViz, change 'Color Transformer' from "Intensity" to "RGB8"

https://github.com/IntelRealSense/realsense-ros/issues/975#issuecomment-552199690

2.  Set the style to **points**.

https://github.com/IntelRealSense/realsense-ros/issues/1327#issuecomment-672731318

vincent51689453, 2021-07-05:
When i am trying to change the color transformer. There is no option can be chosen.
[image]

MartyG-RealSense, 2021-07-05:
There are a couple of other ROS cases open at the time of writing this that have the problem of no selectable color transformer options.

https://github.com/IntelRealSense/realsense-ros/issues/1962#issuecomment-873165085
https://github.com/IntelRealSense/realsense-ros/issues/1938#issuecomment-864271885

A common factor between these two cases is that they are on devices with Arm processors (Raspberry Pi 3B and Jetson Xavier NX).  You too are using a Jetson Xavier NX board.

@doronhi  the RealSense ROS wrapper developer advised that the lack of color images being received is responsible.

https://github.com/IntelRealSense/realsense-ros/issues/1938#issuecomment-864501410

Hi @doronhi  Do you have any advice about these cases of Color Transformer options not being available, with a common factor being boards with an Arm processor, please?

chrisdalke, 2021-07-08:
@MartyG-RealSense @vincent51689453 

I've encountered this same issue with a D415 on a RPI 4; I'm able to view the depth image but the RGB image and color point clouds do not get published. Inspecting the `/camera/color/image_raw` and `/camera/depth/color/points` topics shows nothing being published on either topic.
This issue showed up recently, so I reverted to a manually-built earlier version of librealsense and realsense-ros which fixed the issue. Maybe there's a regression or incompatibility with the recent libraries on ARM?

**I reverted to v2.41.0 of librealsense, and version 2.2.21 of realsense-ros and rebuilt from git on the Raspberry Pi. This combination does not exhibit this issue.**

MartyG-RealSense, 2021-07-09:
Thanks very much @chrisdalke for the feedback to @vincent51689453 about your own installation experience!

MartyG-RealSense, 2021-07-15:
Hi @vincent51689453  Do you require further assistance with this case, please?  Thanks!

vincent51689453, 2021-07-15:
Thankyou for answering my question.

MartyG-RealSense, 2022-01-03:
This issue has been reopened in order for Intel to evaluate reported FPS lag and missing color issues when the pointcloud filter is enabled on Nvidia Jetson boards.  This case should be kept open whilst that review process is active.

PaddyCube, 2022-02-01:
> I just wanted to share, for anyone else who has this problem, that the version that worked for me was v2.43.0 of librealsense built with CUDA and version 2.2.23 of the ROS wrapper.
> 
> It may work with more recent versions before 2.48.0 but I haven't tried them.
> 
> I am using a Jetson Xavier NX with jetpack 4.6

How did you install this version? Did you build im from source or did you use a deb package? I'm struggling with the same issue on Jetson Nano with JetPack 4.6 and librealsense 2.50.0 as well as latest realsense-ros release

MartyG-RealSense, 2022-02-01:
Hi @PaddyCube   I am not @JeremieBourque1 but you could build version 2.43.0 from source code by obtaining the source as a zip file from the 'Assets' file list at the bottom of the page linked to below.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.43.0

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 provides instructions for an **RSUSB backend** build on Jetson from source code.  This build method is not dependent on Linux versions or kernel versions and does not require patching.

PaddyCube, 2022-02-04:
Hi @MartyG-RealSense ,
I was able to install everything from source. Thank you for your support When I run realsense-viewer now, I can see Jetson uses GPU, which wasn't the fact before. However, pointcloud in ROS is still missing. Do I need a special firmware version to be operational with V2.43.0? Right now I'm running with 05.13.00.50

MartyG-RealSense, 2022-02-04:
Firmware 5.13.0.50 should be able to work with SDK 2.43.0.  Are you also using ROS wrapper **2.2.23** with it, as that is the wrapper version that should be matched with SDK 2.43.0.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.23

PaddyCube, 2022-02-04:
Hi,
I use ROS wrapper 2.2.23 and it didn't work. However, did a test with downgrading to firmware 5.12.12.100 (March 2021, Release 5.5) and now I get a pointcloud in RVIZ.

Here is my working setup for reference:
- Jetson Nano B01
- ROS Melodic
- librealsense 2.43.0 build from source with CUDA support according to this guide https://github.com/IntelRealSense/realsense-ros/issues/1967#issuecomment-1027249574
- ROS Wrapper 2.2.23 build from source
- D435 camera with firmware 5.12.12.100 (March 2021, Release 5.5)

Thank you @MartyG-RealSense for your quick and really helpful support

MartyG-RealSense, 2022-02-04:
You are very welcome, @PaddyCube - it's great to hear that you found a solution!  Thanks very much for sharing the details of it with Jetson users in the RealSense ROS community  :)

MartyG-RealSense, 2022-03-14:
@Nir-Az  Re-opening this case as it has an open JIRA ticket DSO-17995 linked to it.   Thanks!

Nir-Az, 2022-03-14:
> @Nir-Az Re-opening this case as it has an open JIRA ticket DSO-17995 linked to it. Thanks!

Thanks, clicked close by mistake probably 👍

xkaraman, 2022-05-19:
Facing this issue on Pi 4 with librealsense v2.50.0 and ros wrapper v2.3.2.

Is the downgrading the only solution for now?

MartyG-RealSense, 2022-05-19:
As far as I am aware, the issue primarily affects Jetson boards specifically.  However, there was a Pi user with similar symptoms (but not necessarily exactly the same cause) earlier in this discussion at https://github.com/IntelRealSense/realsense-ros/issues/1967#issuecomment-876615270 who resolved their particular problem with a downgrade to librealsense 2.41.0 and ROS wrapper 2.2.21.

Typically, a downgrade to 2.43.0 and 2.2.23 is the solution that works with Jetson.  However, a Jetson user reported yesterday at https://github.com/IntelRealSense/librealsense/issues/10526 that librealsense 2.49.0 and ROS wrapper 2.3.1 was a combination that worked for them personally.

If you do not wish to downgrade as far as SDK 2.41.0 or 2.43.0, it may be worth first trying just building wrapper 2.3.1 for your current SDK 2.50.0 version to see whether it improves your ROS performance on Pi.

xkaraman, 2022-05-19:
Ok i wlll give this a try! We don't know exactly where the problem lies i guess, whether in SDK or ROS-Wrapper? The problem is no color image is published but depth is when `pointcloud` filter is enabled!

MartyG-RealSense, 2022-05-19:
My personal opinion would be that it would be more likely to be an SDK related issue, though I am not involved in development activities.

xkaraman, 2022-05-19:
OK any updated walk through (most (dependencies, patches, lib are outdated out there) on how to build SKD on either Ubuntu 20 or Raspbian in case ros wrapper downgrade does not work?

MartyG-RealSense, 2022-05-19:
I would recommend using Ubuntu and using the simple installation procedure in the link below to build the SDK on the Arm-based Raspberry Pi.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

Then build the ROS wrapper from source code after SDK installation has been successfully completed.

Samyak-ja-in, 2022-07-07:
I am using intel realsense d435i and rpi4 I was not getting pointcloud and it solved by setting these argument values in launch file
```
      <arg name="depth_width"       value="640"/>
      <arg name="depth_height"      value="480"/>
      <arg name="depth_fps"         value="6"/>
      <arg name="color_width"       value="640"/>
      <arg name="color_height"      value="480"/>
      <arg name="color_fps"         value="6"/>
      <arg name="enable_depth"      value="true"/>
      <arg name="enable_color"      value="true"/>
      <arg name="enable_infra1"     value="false"/>
      <arg name="enable_infra2"     value="false"/>
      <arg name="enable_fisheye"    value="false"/>
      <arg name="enable_gyro"       value="false"/>
      <arg name="enable_accel"      value="false"/>
      <arg name="enable_pointcloud" value="true"/>
      <arg name="enable_sync"       value="true"/>
```

oysteinskotheim, 2022-10-26:
I have been having the same problem. I tried to compile the RealSense SDK version 2.43.0 and ROS drivers 2.2.23.

Now I get the following error if I try to enable the RGB image (same error both in RealSense viewer and using the ROS launch files):

```
26/10 14:54:33,521 ERROR [547776048928] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: d rs2_deproject_pixel_to_point(float*, const rs2_intrinsics*, const float*, float)
```

MartyG-RealSense, 2022-10-26:
Hi @oysteinskotheim  A RealSense user who had a Jetson TX2 and that same error message resolved the problem in their particular case by re-flashing their JetPack and rebuilding librealsense - as described at https://github.com/IntelRealSense/librealsense/issues/8546#issuecomment-800822718

MartyG-RealSense, 2022-11-04:
Hi @oysteinskotheim  Was the advice in the comment above helpful to you, please?  Thanks!

Hypothesis-Z, 2023-02-21:
Was there a commit that fixed this issue on Jetson? A camera could cause 25% cpu usage for my situation. It seems they are originate from the same bug.

MartyG-RealSense, 2023-02-21:
Hi @Hypothesis-Z  The internal Intel bug report for the Jetson-specific pointcloud issue remains open but there is not progress to report.  Also, the issue would not receive a fix on the ROS1 wrapper (Kinetic, Melodic, Noetic) as development on that wrapper has ceased and it is now called **ros-legacy**.

25% CPU usage does not sound excessive though in terms of correctly working pointcloud generation on Jetson though when CUDA support is enabled.

RezaOptimotive, 2023-02-21:
What are the latest versions for the wrapper and librealsense that work on jetson devices?

Also, would following this issue be the way to get notified once the bug fix is released?

MartyG-RealSense, 2023-02-22:
Hi @RezaOptimotive  This issue would be updated once a fix was released, yes.  

The latest versions of librealsense and the ROS wrapper at the time of writing this that can be matched together are:

ROS1
librealsense 2.50.0 or 2.51.1 and wrapper 2.3.2

ROS2
librealsense 2.51.1 and wrapper 4.51.1

RezaOptimotive, 2023-02-22:
Hi @MartyG-RealSense,

Thanks for getting back! I assume librealsense 2.51.1 and wrapper 4.51.1 for ROS2 don't actually publish point clouds correctly.

Based on your comment [here](https://github.com/IntelRealSense/realsense-ros/issues/2575#issuecomment-1346319645) you mentioned that librealsense 2.48.0 and ROS2 wrapper 3.2.2 works for jetsons. Is that the latest versions that work? Also would you know if those versions work for jetson nanos as well?

MartyG-RealSense, 2023-02-22:
Using that old ROS2 wrapper 3.2.2 and 2.48.0 was the only known solution for ROS2 in the past, yes.  Yes, any ROS wrapper version is compatible with Jetson.

I do not have a specific reference confirming that this combination works on Nano but cannot foresee a reason why it would not if it worked on another Jetson board model.

MartyG-RealSense, 2023-03-14:
Hi @AnOrdinaryUsser  I would not recommend installing SDK 2.41.0 and wrapper 2.2.21 due to the significant age of these versions.  As you are using ROS1 Melodic, an easier workaround than installing an old librealsense version may be to install the latest available SDK and ROS1 wrapper and instead of using the launch file rs_camera.launch, use the **rs_rgbd.launch** file instead.  On ROS1 this has been shown to sucessfully publish a pointcloud without the problems associated with pointcloud generation in an rs_camera launch on Jetson.

To perform an rs_rgbd launch in ROS1 Melodic, support for an RGBD launch must first be installed using the instruction below in the Ubuntu terminal.

`sudo apt-get install ros-melodic-rgbd-launch`

For users of Kinetic and Noetic, the installation command for those ROS versions can be found at https://github.com/IntelRealSense/realsense-ros/issues/2092#issuecomment-929915262

Once support for an RGBD launch is installed, use the roslaunch command below in the ROS terminal.

**roslaunch realsense2_camera rs_rgbd.launch**

rs_rgbd.launch should by default generate a point cloud that is published at **depth_image_proc** - you do not need to enable the pointcloud filter like you do in an rs_camera launch.

AnOrdinaryUsser, 2023-03-14:
> Hi @AnOrdinaryUsser I would not recommend installing SDK 2.41.0 and wrapper 2.2.21 due to the significant age of these versions. As you are using ROS1 Melodic, an easier workaround than installing an old librealsense version may be to install the latest available SDK and ROS1 wrapper match-up and instead of using the launch file rs_camera.launch, use the **rs_rgbd.launch** file instead. On ROS1 this has been shown to successfully publish a pointcloud without the problems associated with pointcloud generation in an rs_camera launch on Jetson.
> 
> To perform an rs_rgbd launch in ROS1 Melodic, support for an RGBD launch must first be installed using the instruction below in the Ubuntu terminal.
> 
> `sudo apt-get install ros-melodic-rgbd-launch`
> 
> For users of Kinetic and Noetic, the installation command for those ROS versions can be found at [#2092 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/2092#issuecomment-929915262)
> 
> Once support for an RGBD launch is installed, use the roslaunch command below in the ROS terminal.
> 
> **roslaunch realsense2_camera rs_rgbd.launch**
> 
> rs_rgbd.launch should by default generate a point cloud that is published at **depth_image_proc** - you do not need to enable the pointcloud filter like you do in an rs_camera launch.

Thanks for all. Now i have a point cloud working on my RVIZ.

MartyG-RealSense, 2023-03-14:
That's great to hear, @AnOrdinaryUsser   :)

MartyG-RealSense, 2023-03-27:
This issue is now being closed for the following reasons.

1.  The RealSense ROS1 wrapper no longer receives updates and has been renamed ros1-legacy.

2.  A correctly working pointcloud can be generated in ROS1 using the rs_rgbd.launch launchfile instead of rs_camera.launch.

closed 2023-03-27 by MartyG-RealSense