realsenseai/realsense-ros #1968: The installed version is different per PCs when installing `ros-melodic-realsense2-camera` in Ubuntu

bigbellmercy, 2021-07-05:
Hi~, I report a problem that the installed versions are different each other for two Ubuntu PC environments when `ros-melodic-realsense2-camera` is installed by `apt-get` as described in https://github.com/IntelRealSense/realsense-ros#method-1-the-ros-distribution.

When it is installed in LattePanda Alpha onboard PC, v2.3.0 is installed. But if it is installed in a general Samsung laptop PC, v2.2.23 is installed. I'd like to install the more recent v2.3.0 on the laptop PC but I could not.

Why are the installed versions different?

### Environments
Laptop: x86 64bit, Ubuntu 18.04, ROS melodic
LattePanda: x86 64bit, Ubuntu 18.04, ROS melodic

MartyG-RealSense, 2021-07-05:
Hi @bigbellmercy   Is there the possibility that there may be a previous installation of the RealSense ROS wrapper on the Samsung laptop PC that is causing a conflict?  You could check for this possibility by performing a removal of the RealSense ROS Melodic package installation before initiating a fresh installation.

** sudo apt-get remove ros-melodic-realsense2-camera
 sudo apt-get install ros-melodic-realsense2-camera**

bigbellmercy, 2021-07-07:
Thanks for reply @MartyG-RealSense 
As you told, I removed the previous software by the command below as written in https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages:
> Remove all RealSense™ SDK-related packages with:
> dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge

So, I think this problem has another cause.

MartyG-RealSense, 2021-07-07:
When wrapper version **2.2.23** is installed on your laptop, which librealsense SDK version is installed?   The wrapper's releases page suggests that it would be librealsense **2.43.0**.

closed 2021-07-19 by MartyG-RealSense