realsenseai/realsense-ros #1969: ORB-SLAM3 Mono_Inertial mode does not work properly with RealSense D455 for image and IMU data acquisition

liangchen97, 2021-07-05:
Mono_Inertial mode does not work properly with RealSense D455 for image and IMU data acquisition

I made sure that I had ORB-SLAM3 installed correctly and that the Euroc dataset was running properly.Now I need to use RealSense D455 to run ORB-SLAM3's MONO mode and MONO_INERTIAL mode
We calibrate the D455 camera parameters, modify the topic in the code, and edit the YAML file

%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
Camera.type: "PinHole"

# Camera calibration and distortion parameters (OpenCV) 
Camera.fx: 638.1018216271331
Camera.fy: 638.3300728549654
Camera.cx: 629.4728680757129
Camera.cy: 375.5517788016263

Camera.k1: -0.05529716280253562
Camera.k2: 0.04869113994652115
Camera.p1: -0.0012081319997317495
Camera.p2: 0.000948771411746312

Camera.width: 1280
Camera.height: 720 

# Camera frames per second 
Camera.fps: 20.0

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 1000

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.2

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#---------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize: 2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -1.8
Viewer.ViewpointF: 500

I can run Mono mode as normal.So I also calibrated the IMU foreign parameters using kalibr and edited the YAML file as follows:

%YAML:1.0

[... 180 lines cut ...]
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 222
mnInitialFrameId = 199
67 Frames set to lost
First KF:35; Map init KF:24
New Map created with 77 points
Point distribution in KeyFrame: left-> 77 --- right-> 0
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 250
mnInitialFrameId = 245
69 Frames set to lost

It seems that ORB-SLAM3 has been unable to detect feature points and occasionally initializes soon after detecting and calculating the path, thus losing the result
[image]

MartyG-RealSense, 2021-07-05:
Hi @liangchen97  There are very few references for using ORB-SLAM3 with a D455 at the time of writing this.  There was a case in the link below where a RealSense D455 user extracted feature points using **Monocular + IMU** data.

https://github.com/IntelRealSense/librealsense/issues/8899

liangchen97, 2021-07-05:
I have seen this, my external parameters are calibrated with Kalibr, can be sure that there is no problem, I do not know if the author of this problem successfully run Mono + IMU

MartyG-RealSense, 2021-07-05:
As mentioned, there are a limited number of other references about using ORB-SLAM3 with RealSense.   I translated an article on the subject into English (though it involves D435i rather than D455) and have attached the PDF to this message.

[ORB-SLAM3_RealSense.pdf](https://github.com/IntelRealSense/realsense-ros/files/6765571/ORB-SLAM3_RealSense.pdf)

An ORB-SLAM3 RealSense discussion about Mono-Inertial (again using D435i) is also at the link below.

https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/67

Hopefully one of these resources will provide soe useful insights about obtaining feature points.

MartyG-RealSense, 2021-07-12:
Hi @liangchen97  Do you require further assistance with this case, please?  Thanks!

liangchen97, 2021-07-12:
No thanks. I will update if there is new progress

MartyG-RealSense, 2021-07-12:
Okay, thanks very much.  I will close the case but you are welcome to lave further comments or start a new issue when you make progress.  Good luck!

closed 2021-07-12 by MartyG-RealSense