realsenseai/realsense-ros #1970: ROS2 Foxy, librealsens 2.48.0, ros-realsense v3.2.2: No images in rviz2, error "Message Filter dropping message: frame 'camera_color_optical_frame'

admantium-sg, 2021-07-05:
Hello,

I have two computers: Rapberry Pi4 and a Linux Workstation, both with Ubuntu20.04 and ROS2 Foxy.

The D435 is connected to the Raspberry Pi4.

Raspberry Pi4 and workstation have IPs in the local network, I can ping between them, and I can see all published ROS topics:

```
ros2 topic list
/camera/color/camera_info
/camera/color/image_raw
/camera/depth/camera_info
/camera/depth/color/points
/camera/depth/image_rect_raw
/camera/extrinsics/depth_to_color
/parameter_events
/rosout
/tf
/tf_static
``` 

But when I open rviz2 on my workstation, I can not see images from topic /camera/color/image_raw rendered in rviz2, error message in rviz is:

`[INFO] [1625329665.271601068] [rviz]: Message Filter dropping message: frame 'camera_color_optical_frame' at time 1625329664,860 for reason 'Unknown'`

Tried two different installations: Provide Ubuntu packages, and also from scratch compilation/installation with librealsense v2.48.0  and realsense-ros v3.2.2.

The ROS command to start the d435 node is

`ros2 launch realsense2_camera demo_pointcloud_launch2.py initial_reset:=true `

Sample Output

```
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2021-07-05-20-44-36-049525-ubuntu-2859
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [2861]
[realsense2_camera_node-1] [INFO] [1625510676.899324523] [RealSenseCameraNode]: RealSense ROS v3.2.2
[realsense2_camera_node-1] [INFO] [1625510676.899491226] [RealSenseCameraNode]: Built with LibRealSense v2.48.0
[realsense2_camera_node-1] [INFO] [1625510676.899535430] [RealSenseCameraNode]: Running with LibRealSense v2.48.0
[realsense2_camera_node-1] [INFO] [1625510676.991105406] [RealSenseCameraNode]: Device with serial number 018322070534 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1625510676.991237406] [RealSenseCameraNode]: Device with physical ID /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1625510676.991279869] [RealSenseCameraNode]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1625510676.992265518] [RealSenseCameraNode]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1625510676.992373647] [RealSenseCameraNode]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1625510676.992427962] [RealSenseCameraNode]: Resetting device...
[realsense2_camera_node-1] [INFO] [1625510683.152204050] [RealSenseCameraNode]: Device with serial number 018322070534 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1625510683.152322198] [RealSenseCameraNode]: Device with physical ID /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1625510683.152362031] [RealSenseCameraNode]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1625510683.153211958] [RealSenseCameraNode]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1625510683.153339661] [RealSenseCameraNode]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1625510683.165928906] [RealSenseCameraNode]: getParameters...
[realsense2_camera_node-1] [INFO] [1625510683.170501112] [RealSenseCameraNode]: setupDevice...
[realsense2_camera_node-1] [INFO] [1625510683.170579001] [RealSenseCameraNode]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1625510683.170640038] [RealSenseCameraNode]: Device Name: Intel RealSense D435
[realsense2_camera_node-1] [INFO] [1625510683.170683723] [RealSenseCameraNode]: Device Serial No: 018322070534
[realsense2_camera_node-1] [INFO] [1625510683.170722556] [RealSenseCameraNode]: Device physical port: /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1625510683.170760408] [RealSenseCameraNode]: Device FW version: 05.12.13.50
[realsense2_camera_node-1] [INFO] [1625510683.170796778] [RealSenseCameraNode]: Device Product ID: 0x0B07
[realsense2_camera_node-1] [INFO] [1625510683.170830204] [RealSenseCameraNode]: Enable PointCloud: On
[realsense2_camera_node-1] [INFO] [1625510683.170863760] [RealSenseCameraNode]: Align Depth: Off
[realsense2_camera_node-1] [INFO] [1625510683.170895112] [RealSenseCameraNode]: Sync Mode: On
[realsense2_camera_node-1] [INFO] [1625510683.171143260] [RealSenseCameraNode]: Device Sensors: 
[realsense2_camera_node-1] [INFO] [1625510683.204649864] [RealSenseCameraNode]: Stereo Module was found.
[realsense2_camera_node-1]  05/07 20:44:43,206 WARNING [281472993135008] (backend-v4l2.cpp:1358) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[realsense2_camera_node-1] [INFO] [1625510683.266804126] [RealSenseCameraNode]: RGB Camera was found.
[realsense2_camera_node-1] [INFO] [1625510683.266940330] [RealSenseCameraNode]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1625510683.266989349] [RealSenseCameraNode]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1625510683.267027219] [RealSenseCameraNode]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1625510683.267062608] [RealSenseCameraNode]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1625510683.267098108] [RealSenseCameraNode]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1625510683.267135367] [RealSenseCameraNode]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1625510683.267176052] [RealSenseCameraNode]: Add Filter: pointcloud
[realsense2_camera_node-1] [INFO] [1625510683.280456353] [RealSenseCameraNode]: num_filters: 1
[realsense2_camera_node-1] [INFO] [1625510683.280564631] [RealSenseCameraNode]: Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [WARN] [1625510683.316406329] [RealSenseCameraNode]: Could not set param: stereo_module.auto_exposure_limit with 0Range: [1, 165000]: parameter 'stereo_module.auto_exposure_limit' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [WARN] [1625510683.318068663] [RealSenseCameraNode]: Could not set param: stereo_module.auto_gain_limit with 0Range: [16, 248]: parameter 'stereo_module.auto_gain_limit' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [WARN] [1625510683.333642464] [RealSenseCameraNode]: Could not set param: rgb_camera.power_line_frequency with 3Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1625510683.335995761] [RealSenseCameraNode]: Done Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [INFO] [1625510683.338396651] [RealSenseCameraNode]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[realsense2_camera_node-1] [INFO] [1625510683.342440412] [RealSenseCameraNode]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[realsense2_camera_node-1] [INFO] [1625510683.342566153] [RealSenseCameraNode]: setupPublishers...
[realsense2_camera_node-1] [INFO] [1625510683.352397971] [RealSenseCameraNode]: setupStreams...
[realsense2_camera_node-1] [INFO] [1625510683.470379141] [RealSenseCameraNode]: SELECTED BASE:Depth, 0
[realsense2_camera_node-1] [INFO] [1625510683.478110292] [RealSenseCameraNode]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1625510683.541182240] [RealSenseCameraNode]: 
```

Sample Image Data (when running from my workstation)

`ros2 topic echo /camera/color/image_raw`

```
---
header:
  stamp:
    sec: 1625510352
    nanosec: 537663488
  frame_id: camera_color_optical_frame
height: 480
width: 640
encoding: rgb8
is_bigendian: 0
step: 1920
data:
- 122
- 129
- 129
- 128
- ...
```

Any help is greatly appreciated - want to start building my robots vision sensor with the D435 :)

MartyG-RealSense, 2021-07-06:
Hi @admantium-sg   Do the **Message Filter dropping message** information messages continue generating after launch or do they stop at launch after appearing a small number of times, please?

In a past RealSense ROS discussion in which this message appeared (which also involved **camera_color_optical_frame**), it was corrected after using **initial_reset:=true** but you have already tried that method.

https://github.com/IntelRealSense/realsense-ros/issues/1547

I note that you state that you are using a launch file called **demo_pointcloud_launch2.py**, whereas the ROS2 launch file provided with the ROS2 wrapper is **demo_pointcloud_launch.py**.  If you are using 'launch2' instead of 'launch', are you using your own custom launch file?  If you are using a custom launch, do you have the same problem when launching with demo_pointcloud_launch.py

admantium-sg, 2021-07-06:
Hello Marty,

thx for trying to help.

the messages are printed continuously as long as rviz is open.

My launch file makes a minor modification only: disabled the rviz node because the Raspi4 runs heedless, without desktop environment.

MartyG-RealSense, 2021-07-06:
There are very few references about this error in relation to RealSense.  Further research did locate a PDF paper though that was published a week ago (June 30 2021) and encountered this error with camera_color_optical_frame in ROS2 Foxy.  It reported on page 127 of the document (page 123 if using the numbers on the bottom of the page) that it occurred for them if camera_color_optical_frame was not present.

https://easy-manipulation-deployment-docs.readthedocs.io/_/downloads/en/latest/pdf/

Serafadam, 2021-07-09:
Hi, I've been having similar problems on Foxy installed by docker on Jetson Xavier NX, and by going through the issues, I guess this may be connected to #1962 and #1967. (Though when I do `ros2 topic echo` on pointcloud, it receives some messages (though it seems very laggy). Rviz2 run on another PC doesn't seem to pick anything up.
I'll try rebuilding the librealsense (as in https://github.com/IntelRealSense/realsense-ros/issues/1962#issuecomment-876615709) and see what will happen.

MartyG-RealSense, 2021-07-09:
Thanks very much @Serafadam - please do report the results of your librealsense rebuild.  Good luck!

admantium-sg, 2021-07-11:
Hello @Serafadam,

thank you for commenting here. 

At the moment, I can still not run D435 on my Pi4 and stream the data to my workstation.

Yesterday, I installed ROS1 on my Pi4 and Workstation. Only when the roscore runs on the Pi4, I could start the librealsense-wrapper in ROS. However, no pointcloud or image data were streamed (means `rostopic echo /camera/color/image_raw` is empty when run from the workstation).

Then, reading this issue #1962 , the author says 

> After trying the same except the RS2_STREAM_COLOR, the pointcloud message status is fine however is not displaying anything in rviz as mentioned before. Also the color/image_raw topic is still not being received as shown in the left panel of rviz.

So, in essence I'm having the same issue with ROS1 and ROS2 by installing the most recent version of librealsense and realsense-ros. Which brings me to #1967:

> I reverted to v2.41.0 of librealsense, and version 2.2.21 of realsense-ros and rebuilt from git on the Raspberry Pi. This combination does not exhibit this issue.

I will try this solution

@MartyG-RealSense , in the meantime, I want to exclude any other error sources. Can you help me with these questions? 
- Can I run realsense-ros on a headless Pi4 with ARM architecture? (e.g do you know a ROS project in which this works?)
- Does the Raspberry Pi installation also require to install `protobuf` and the python bindings as outlined in [Raspberry Pi4 with D435](https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435)?

MartyG-RealSense, 2021-07-11:
Pi has been used headless by other RealSense users, such as the examples in the links below.

https://github.com/IntelRealSense/librealsense/issues/8007#issuecomment-745555920
https://github.com/IntelRealSense/librealsense/issues/4272#issuecomment-554877281

If you are using Raspbian then you should need to install Protobuf as per the Raspberry Pi guides by Acrobotic and Intel.  Some RealSense users have found that installation works best with a more modern Protobuf version such as **3.14** rather than the older versions suggested in the guides though.

MartyG-RealSense, 2021-07-16:
Hi @admantium-sg  Do you require further assistance with this case, please?  Thanks!

admantium-sg, 2021-07-19:
Hello Marty,

sorry for the late reply, I could continue with my project only this weekend.

I continued with librealsense v2.0.48. After changing the configuration of my ROS installation - running `roscore` directly on the Pi -  I could finally stream some data from my Raspberry Pi 4 to my Linux workstation. 

When using this command

```bash
roslaunch realsense2_camera rs_camera.launch camera:=camera1 depth_width:=640 color_width:=640 depth_height:=480 color_height:=480 depth_fps:=30 color_fps:=30 initial_reset:=true -v
```

I do only receive `/camera1/depth/image_rect_raw` and `/camera1/colora/image_raw` topics 

When using 

```bash
roslaunch realsense2_camera rs_camera.launch camera:=camera1 depth_width:=640 color_width:=640 depth_height:=480 color_height:=480 depth_fps:=30 color_fps:=30 pointcloud_texture_stream:=RS2_STREAM_ANY enable_sync:=true align_depth:=true initial_reset:=true filters:=pointcloud -v
```

When publishing point cloud data to `/camera/depth/color/points`, then nothing is published  in `/camera1/colora/image_raw`. Thats the very same issue as in #1962.

I will continue to work on gettng `/camera/depth/color/points` and `/camera1/colora/image_raw` working. 

As for this issue, you can close it since the original error does not occur anymore.

MartyG-RealSense, 2021-07-20:
Thanks very much @admantium-sg for the detailed feedback.  Please do update here about the results of your tests wth the color topics and ROS2 when you have them.  Good luck!

MartyG-RealSense, 2021-08-01:
Hi @admantium-sg   Do you have an update about this case that you can provide, please?  Thanks!

admantium-sg, 2021-08-01:
Update while working with ROS1, see #2014

MartyG-RealSense, 2021-08-07:
Thanks @admantium-sg for your earlier update about ROS1.  Do you have any more questions about the ROS2 Foxy case here, please?  Thanks!

closed 2021-08-18 by MartyG-RealSense