realsenseai/realsense-ros #1971: RealSense D455 Depth Image issue

swiz23, 2021-07-08:
Hi, I'm working wiith a RealSense D455 camera on Ubuntu 18.04.5 with ROS Melodic. I'm using tag 2.2.20 with librealsense 2.40.

When subscribing to the /camera/depth/image_rect_raw topic, I get depth images similar to the one below.

[image]

[image]

[image]

The issue is that there appears to be a lot of white dots specifically on a narrow vertical strip along the left-hand side of the image. No matter where I point the camera, those dots (more or less of them) are there. Why is that happening? To clarify, I'm not pointing the camera at anything too close which might cause issues.

MartyG-RealSense, 2021-07-08:
Hi @swiz23   As a starting point in investigating your case, could you test whether the dot pattern from the IR emitter is leaking nto the depth image please?

A way to do this would be:

1.  Open the dynamic_reconfigure interface by inputting the line below into the ROS terminal after the roslaunch completes: 

**rosrun rqt_reconfigure rqt_reconfigure**

dynamic_reconfigure is available in the ROS wrapper (Kinetic, Melodic, Noetic) and enables the changing of camera values during runtime.

2.  Select the **stereo_module** category in the side-panel of the dynamic-reconfigure interface.

[image]

3.  In the stereo_module options, find the **laser_power** option and reduce its value.  

Does the visibility of the dots at the side of the image reduce as the laser_power value is reduced?

swiz23, 2021-07-08:
Tried that, and it actually gets worse as you go down - granted I don't have the best lighting in my apartment though so that's kind of expected.

MartyG-RealSense, 2021-07-08:
It is expected that depth image quality decreases as Laser Power is reduced.  The purpose of the test was to determine whether the white dots became less visible as the Laser Power value was decreased.   Do the dots noticably change as Laser Power reduces, please?

swiz23, 2021-07-08:
No, they do not become less visible as I decrease the laser power.

Something else to note is that the left-hand side of the depth image appears jagged while on the right side - it's a nice vertical 'line'. I've included an image below without normalizing - just for contrast purposes.

[image]

MartyG-RealSense, 2021-07-08:
It is normal for the far-left edge of the image to be jagged.  This area is called the **Invalid Depth Band**.  Pages 68-69 of the current edition of the data sheet document for the 400 Series cameras have information about this phenomenon.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

[image]

swiz23, 2021-07-08:
So I don't have to worry about the white dots on the left-hand side since it's inside the invalid depth band region?

MartyG-RealSense, 2021-07-08:
Whilst it may look a little untidy when there is visible noise inside that far-left border area beyond the jagged edge, it is a region of no depth and so shouldn't have an impact on the depth results within the main image.

closed 2021-07-11 by swiz23