realsenseai/realsense-ros #1975: Question about the /camera/depth/image_rect_raw topic

swiz23, 2021-07-11:
Is this topic a rectified depth image or is it a raw depth image?

MartyG-RealSense, 2021-07-11:
Hi @swiz23  Most of the stream formats of the RealSense 400 Series cameras other than **Y16** are **rectified**, whilst Y16 is unrectified as it is used for camera calibration.

swiz23, 2021-07-11:
perfect. so since the depth stream is in Z16 format, then it would be safe to assume that the depth image on this topic is rectified?

MartyG-RealSense, 2021-07-12:
Z16 depth is rectified, yes. The depth on 400 Series cameras is generated using the left and right infrared frames.

https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#d400-series

MartyG-RealSense, 2026-10-23:
Case closed due to no further comments received after a significant time period.

closed 2026-10-23 by MartyG-RealSense