realsenseai/realsense-ros #1976: 2D to 3D

Lai-dongdong, 2021-07-12:
Hi, Where is the 3D plane position of the imaging surface of the L515 2D RGB image corresponding to the point cloud? What is the rotation, translation, and scaling matrix of the 2D RGB image to the corresponding point cloud plane?

RealSenseSupport, 2021-07-13:
One way to find the extrisic calibration data (ie, depth to color) for the L515 is using the [rs-enumerate-devices -c](https://github.com/IntelRealSense/librealsense/tree/master/tools/enumerate-devices#command-line-parameters) tool from the RealSense SDK. Please take a look. Thank you.

Lai-dongdong, 2021-07-14:
> One way to find the extrisic calibration data (ie, depth to color) for the L515 is using the [rs-enumerate-devices -c](https://github.com/IntelRealSense/librealsense/tree/master/tools/enumerate-devices#command-line-parameters) tool from the RealSense SDK. Please take a look. Thank you.

Thank you. I only saved ply, color, infrared and depth files when I used the L515. Can I still read the external camera parameters from these data?
My purpose is to generate a new marked point cloud by marking the RGB image, depth data and camera internal and external parameters.

RealSenseSupport, 2021-07-14:
The camera intrinsic and extrinsic values can be obtained via the APIs:
https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_frame.hpp#L238
https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_frame.hpp#L148
Below is link to a very good documentation about the 2D-3D projection in the SDK and should provide further help in this case:
https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0