realsenseai/realsense-ros #1977: Jetson Xavier NX does not work properly with D435i

krishnaadithya, 2021-07-12:
1. Realsense-Ros control transfer error, while displaying pointcloud, No Data is recieved but in realsense-viewer point cloud is visible.
2. RGB and depth image are not received at the same time in ROS.
3. Depth Image has a latency of 1-2s.
4. E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/bionic/InRelease  403  Forbidden [IP: 52.218.28.194 80]
>>Error while trying to install from source

MartyG-RealSense, 2021-07-12:
Hi @krishnaadithya 

1.  If the **control_transfer returned** warning occurs only a small number of times at launch and then stops, or at a rate of once or twice per minute after launch, then it can be safely ignored.  If it is generating continuously though then this can indicate that there is a serious communication problem with the device.

2 and 3.  Are you using depth-color alignment and the depth and color are publishing at different rates (for example, 15 FPS for depth and 30 FPS for color)?

4.  The method that you are using for the installation may have outdated links.  Intel switched from http to **https** links in March 2021.  For installing librealsense on Jetson, the official instruction page is at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

If the methods on the page do not work for you then the link below has a user-contributed guide for installing from source code with CMake on Jetson boards that has been shown to work on Jetson NX:

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

krishnaadithya, 2021-07-12:
Hi @MartyG-RealSense 
Thank you for the quick reply, 
1. control_transfer returned warning is continue does not stop, so we are still unable to get the pointcloud data from D435i.
2 and 3 both are being published at the same rate (30fps) but it is still has 1-2 sec latency. 
4 [solved]. thank you for the instruction IntelRealSense/librealsense#6964 (comment) this worked.

MartyG-RealSense, 2021-07-12:
Is there any improvement in the generation of warnings or the latency if you reset the camera at launch by adding **initial_reset:=true** to the roslaunch instruction?  For example:

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**

krishnaadithya, 2021-07-13:
Thank you very much 
roslaunch realsense2_camera rs_camera.launch initial_reset:=true
This fixed the latency issue

But we are still unable to get the poincloud values 
This is the error we get when we try to publish pointcloud:

 [WARN] [1626155118.718739495]: Param '/camera/stereo_module/auto_exposure_limit' has value 0 that is not in range [1, 165000]. Removing this parameter from dynamic reconfigure options.
[ WARN] [1626155118.736110124]: Param '/camera/stereo_module/auto_gain_limit' has value 0 that is not in range [16, 248]. Removing this parameter from dynamic reconfigure options.
[ INFO] [1626155119.203026815]: Done Setting Dynamic reconfig parameters.
[ INFO] [1626155119.206952014]: depth stream is enabled - width: 1280, height: 720, fps: 30, Format: Z16
[ INFO] [1626155119.210432508]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1626155119.218518877]: setupPublishers...
[ INFO] [1626155119.231521361]: Expected frequency for depth = 30.00000
[ INFO] [1626155119.541558733]: Expected frequency for color = 30.00000
[ INFO] [1626155119.760037177]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1626155120.007814619]: setupStreams...
[ INFO] [1626155120.014108437]: insert Depth to Stereo Module
[ INFO] [1626155120.014311349]: insert Color to RGB Camera
 13/07 11:15:20,168 WARNING [546534584704] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ WARN] [1626155120.207720604]: Hardware Notification:Motion Module failure,1.62616e+12,Error,Hardware Error
[ INFO] [1626155120.220554064]: SELECTED BASE:Depth, 0
[ WARN] [1626155120.325325364]: 
 13/07 11:15:20,327 WARNING [546534584704] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1626155120.340048911]: RealSense Node Is Up!
 13/07 11:15:20,378 WARNING [546534584704] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 13/07 11:15:20,430 WARNING [546534584704] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
[ WARN] [1626155123.208195976]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
[ WARN] [1626155125.209994369]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
[ WARN] [1626155127.212839919]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
[ WARN] [1626155129.218962202]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
[ WARN] [1626155131.222001416]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
[ WARN] [1626155133.225294728]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
 13/07 11:15:33,604 WARNING [546534584704] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 13/07 11:15:34,655 WARNING [546534584704] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ WARN] [1626155135.229167867]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
[ WARN] [1626155137.231385400]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
[ WARN] [1626155139.233218307]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
[ WARN] [1626155141.236036259]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
[ WARN] [1626155143.238036688]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
 13/07 11:15:44,715 WARNING [546534584704] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ WARN] [1626155146.239955987]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
[ WARN] [1626155148.241605369]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
 13/07 11:15:49,769 WARNING [546534584704] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ WARN] [1626155150.244849206]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
[ WARN] [1626155152.247797314]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
 13/07 11:15:52,822 WARNING [546534584704] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ WARN] [1626155154.252017924]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
[ WARN] [1626155156.253341100]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
[ WARN] [1626155158.257206991]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
[ WARN] [1626155160.260367040]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
[ WARN] [1626155162.261675355]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
[ WARN] [1626155164.263363305]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
[ WARN] [1626155166.267276433]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
 13/07 11:16:07,887 WARNING [546534584704] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
[ WARN] [1626155290.374354663]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
[ WARN] [1626155292.375874993]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
[ WARN] [1626155294.377231727]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
[ WARN] [1626155296.379331620]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error
^[[ WARN] [1626155298.380683884]: Hardware Notification:USB CAM overflow,1.62616e+12,Error,Hardware Error

MartyG-RealSense, 2021-07-13:
The **USB CAM overflow** error suggests to me that the USB port may be over-burdened by the volume of incoming data from the camera, which would be a lot of bandwidth when both depth and color are using 1280x720 resolution at 30 FPS.

Are you using the official 1 meter USB cable supplied with the camera, or a longer cable of your own choice?

Please try the custom roslaunch instruction in the link below, which should publish depth and color at a low resolution of 640x480 and also enable aligned_depth and the pointcloud filter.  If the error no longer occurs then this would add weight to the possibility of it being a problem related to excessive incoming data due to the resolution used.

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true align_depth:=true filters:=pointcloud depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30**

krishnaadithya, 2021-07-15:
Hey @MartyG-RealSense 

We are using the USB supplied with the camera 

We tried this 
**roslaunch realsense2_camera rs_camera.launch initial_reset:=true align_depth:=true filters:=pointcloud depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30**

The same issue persists, we did not get the pointcloud and this time there is a latency as well.

MartyG-RealSense, 2021-07-15:
The latency was reported earlier in this discussion as corrected if **roslaunch realsense2_camera rs_camera.launch initial_reset:=true** was used.  This suggests that adding **align_depth:=true filters:=pointcloud** (either one or both of these instructions) might add latency.

You continue to experience the **USB CAM overflow** warning no matter what roslaunch instruction that you use though, right?

Given the **control_transfer returned** errors continue to generate repeatedly after launch when **RealSense Node Is Up!**, the USB CAM Overflow messages may be resulting from a communication problem with the device, which can be a cause of continuously generating 'control_transfer returned' warnings as mentioned at the start of this case.  Such a communication problem may be preventing the point cloud from being published in RViz.

Could you provide an image of your RViz window, please?  Thanks!

MartyG-RealSense, 2021-07-21:
Hi @krishnaadithya  Do you require further assistance with this case, please?  Thanks!

krishnaadithya, 2021-07-22:
Yes, will send you the detail apologies for the delay.

krishnaadithya, 2021-07-22:
@MartyG-RealSense 
I was testing the realsense again, now I have the latency even with initial reset. 
But on realsense-viewer things work perfectly fine. 

You continue to experience the USB CAM overflow warning no matter what roslaunch instruction that you use though, right?
Yes

Image of RVIz window, do you mean screenshot?

krishnaadithya, 2021-07-22:
@MartyG-RealSense 
for some reason now I am not even getting the camera feed.

And about the communication problem, why does it work perfectly fine with realsense-viewer and not with ROS?

MartyG-RealSense, 2021-07-22:
Yes, please provide a screenshot of your Rviz window if possible.

The RealSense Viewer is built directly upon the librealsense SDK, whilst the RealSense ROS wrapper acts as a compatibility layer for ROS with librealsense in the background.  The ROS wrapper also performs some operations differently to librealsense in order to align with ROS standards.

krishnaadithya, 2021-07-23:
[image]

Log File
[roslaunch-accio-x1-13816.log](https://github.com/IntelRealSense/realsense-ros/files/6867311/roslaunch-accio-x1-13816.log)

krishnaadithya, 2021-07-23:
@MartyG-RealSense 
Camera is not giving any output, it did come for sometime and then it just stopped.
Only Depth comes properly but that too there is a latency

MartyG-RealSense, 2021-07-23:
I went through past cases involving the **68, error: Resource temporarily unavailable, number: 11** variety of the control_transfer returned message and found a Jetson case involving USB communication problems where a RealSense user offered very detailed advice based on their experiences.

https://github.com/IntelRealSense/realsense-ros/issues/1164#issuecomment-620903027

krishnaadithya, 2021-07-24:
@MartyG-RealSense  I am bit confused, 
From the comment there I learnt two things:
1. Power Limitation 
2. Wait for realsense to fix it (again that comment was April 2020) 

For 1, the we are getting all outputs on realsense-viewer and interestingly we got all the output on Nvidia Isaac with minimum latency.
2, Its been a year, will they fix it?

MartyG-RealSense, 2021-07-24:
In general, successful operation with the RealSense Viewer does not always translate to successful operation in the RealSense ROS wrapper.  The Viewer runs directly in librealsense, whilst the ROS wrapper provides a compatibility layer between librealsense and ROS that means that it sometimes has to implement functions differently in order to be aligned with ROS standards.

In regard to point 2, I am not involved in development activities and so do not have information to provide on that subject.

krishnaadithya, 2021-07-24:
So this problem can't be solved?

MartyG-RealSense, 2021-07-24:
@krishnaadithya  I just mean that I cannot comment on development-related activities as I am a support engineer and do not work on wrapper development.

krishnaadithya, 2021-07-24:
Not Development 
But is there any other way to fix these problems?

MartyG-RealSense, 2021-07-24:
If you could define in a detailed description what you believe the problem is in your particular case then I can refer that to a developer to seek their input.

krishnaadithya, 2021-07-24:
Connecting Realsense D435i to Xavier nx ROS
Jetpack: 4.5
ROS Melodic
Issue:
1. ROS does not receive pointcloud data from the camera
2. Camera output does not come 
3. Depth there is a latency. 

Log file can be found in this link:
https://github.com/IntelRealSense/realsense-ros/issues/1977#issuecomment-885464445

MartyG-RealSense, 2021-07-24:
@krishnaadithya  Thanks very much for the description of your Jetson Xavier NX problem.

Hi @doronhi  Do you have advice that you can offer @krishnaadithya about the Jetson problem described above, please?  Thanks!

krishnaadithya, 2021-07-28:
@MartyG-RealSense @doronhi 

I am bit confused, when I run rs_camera.launch there are a lot of issues:
1.ROS does not receive pointcloud data from the camera
2.Camera output does not come
3.Depth there is a latency.

But when  I run rs_rgbd.launch, all of them work but there is still latency/ frame drop issue.

We are trying to use the realsense for our product, can you please help me out in setting this to work in real time.

MartyG-RealSense, 2021-07-28:
@krishnaadithya  Given that rs_rgbd.launch works, do you see any difference with rs_camera.launch if you set **ordered_pc:=true** to produce an **ordered point cloud** like that of rs_rgbd instead of the default unordered cloud?  For example:

roslaunch realsense2_camera rs_camera.launch filters:=colorizer,pointcloud align_depth:=true pointcloud_texture_stream:=RS2_STREAM_COLOR initial_reset:=true ordered_pc:=true

krishnaadithya, 2021-07-29:
rs_rgbd.launch works, but even that has a pretty bad latency when it comes to aligned depth, aligned camera, pointcloud and left and right camera.

I will check the command that you suggested.

krishnaadithya, 2021-07-29:
Hi tried the command, I did not get any output

[image]

MartyG-RealSense, 2021-07-29:
At this point it may be worth doing a wipe of the whole installation and install eveything fresh ... Ubuntu, ROS, RealSense ROS wrapper ... and see whether that fixes the problem due to eliminating a hidden problem somewhere in the system.

krishnaadithya, 2021-07-30:
@MartyG-RealSense 
Ya I felt the same, we had a another jetson that we had to do the same installation on. 
On that we did everything from scratch.
For another dependency they updated the python3 from 3.6 to 3.7 
after that 
https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049
This installation doesn't seem to work.
I changed python3.6 to python3.7
and the entire process goes fine but pyrealsense2 folder is not there in /usr/local/lib/python3.7/

MartyG-RealSense, 2021-07-30:
You can confirm what the Python path is on your particular computer using the command **which python3**

I believe that on Jetson and Raspberry Pi, the pyrealsense2 files are built in **librealsense/build/wrappers/python**

[image]

closed 2021-08-11 by MartyG-RealSense