realsenseai/realsense-ros #1978: Realsense D435, Pointcloud2 is not available in Realsense viewer

vanerkz, 2021-07-12:
[image]

[image]

Hi, I have the problem shown above. I have tried different methods to solve the problem above but yet I still have the same problem. Both depth and RGB is working as shown above. 

anyone encounter this problem before ? Thank you !

MartyG-RealSense, 2021-07-12:
Hi @vanerkz  The 3D image in your Viewer does not look like a typical **textured pointcloud** where the RGB image is mapped to the depth points.   It resembles a mis-colored depth only cloud.

Could you try disabling the two **GLSL** settings in the Viewer's settings interface if they are enabled please, using the instructions in the link below to do so, and see whether it makes a positive difference to the point cloud in 3D mode? 

https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023

vanerkz, 2021-07-12:
Hi, I have disable both GLSL setting. However, I still have the same problem.

MartyG-RealSense, 2021-07-12:
Does the point cloud update correctly if you go to the **Texture** option at the top of the Viewer window and toggle between the depth and color settings?

[image]

Also, may I confirm please that you are not using the RealSense ROS wrapper at the same time that you are using the Viewer?

vanerkz, 2021-07-12:
[image]
[image]

I have tried to toggle between the texture option. It does change between the options. 
I did not use ROS wrapper at the same time. However, I also faced the same problem when I uses ROS wrapper. It does not show any pointcloud2 in rivz.

MartyG-RealSense, 2021-07-12:
If you are pointing the camera at the curtains during your toggle test, it still does not look as though it is displaying the RGB detail correctly when selecting the Color option.

in regard to ROS, are you using **roslaunch realsense2_camera rs_camera.launch filters:=pointcloud** as your roslaunch instruction like in the ROS wrapper's pointcloud example in the link below?

roslaunch realsense2_camera rs_camera.launch filters:=pointcloud

vanerkz, 2021-07-12:
The below are the output
SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: -1
 * /camera/realsense2_camera/infra_height: -1
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: -1
 * /camera/realsense2_camera/initial_reset: False
[... 52 lines cut ...]
[ INFO] [1626084024.037485244]: Device Product ID: 0x0B07
[ INFO] [1626084024.037512563]: Enable PointCloud: On
[ INFO] [1626084024.037538494]: Align Depth: Off
[ INFO] [1626084024.037563344]: Sync Mode: On
[ INFO] [1626084024.037628013]: Device Sensors: 
[ INFO] [1626084024.043376995]: Stereo Module was found.
[ INFO] [1626084024.052497892]: RGB Camera was found.
[ INFO] [1626084024.052583144]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1626084024.052735998]: Add Filter: pointcloud
[ INFO] [1626084024.053370608]: num_filters: 1
[ INFO] [1626084024.053414316]: Setting Dynamic reconfig parameters.
[ WARN] [1626084024.143063700]: Param '/camera/stereo_module/auto_exposure_limit' has value 0 that is not in range [1, 165000]. Removing this parameter from dynamic reconfigure options.
[ WARN] [1626084024.145404539]: Param '/camera/stereo_module/auto_gain_limit' has value 0 that is not in range [16, 248]. Removing this parameter from dynamic reconfigure options.
[ INFO] [1626084024.181720215]: Done Setting Dynamic reconfig parameters.
[ INFO] [1626084024.182881406]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1626084024.184090015]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1626084024.184194439]: setupPublishers...
[ INFO] [1626084024.190743844]: Expected frequency for depth = 30.00000
[ INFO] [1626084024.228344343]: Expected frequency for color = 30.00000
[ INFO] [1626084024.248694676]: setupStreams...
[ INFO] [1626084024.251835471]: insert Depth to Stereo Module
[ INFO] [1626084024.251915671]: insert Color to RGB Camera
[ INFO] [1626084024.272747010]: SELECTED BASE:Depth, 0
[ INFO] [1626084024.288344672]: RealSense Node Is Up!
[ WARN] [1626084024.394158319]:

vanerkz, 2021-07-12:
If I run realsense2_camera rs_d435_camera_with_model.launch. I cant see any pointcloud2 in rivz. 

robot@robot:~$ roslaunch realsense2_camera rs_d435_camera_with_model.launch 
... logging to /home/robot/.ros/log/34080a00-e2f8-11eb-8bbd-0024d7098ccc/roslaunch-robot-4223.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/robot/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://localhost:46353/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 240
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 424
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
[... 73 lines cut ...]
[ INFO] [1626084124.376400862]: Sync Mode: On
[ INFO] [1626084124.376469632]: Device Sensors: 
[ INFO] [1626084124.381110919]: Stereo Module was found.
[ INFO] [1626084124.388773905]: RGB Camera was found.
[ INFO] [1626084124.388871894]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1626084124.388905624]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1626084124.389063735]: Add Filter: pointcloud
[ INFO] [1626084124.389738825]: num_filters: 1
[ INFO] [1626084124.389786066]: Setting Dynamic reconfig parameters.
[ WARN] [1626084124.458867475]: Param '/camera/stereo_module/auto_exposure_limit' has value 0 that is not in range [1, 165000]. Removing this parameter from dynamic reconfigure options.
[ WARN] [1626084124.460777485]: Param '/camera/stereo_module/auto_gain_limit' has value 0 that is not in range [16, 248]. Removing this parameter from dynamic reconfigure options.
[ INFO] [1626084124.480285374]: Done Setting Dynamic reconfig parameters.
[ INFO] [1626084124.480695156]: depth stream is enabled - width: 424, height: 240, fps: 30, Format: Z16
[ INFO] [1626084124.481141335]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1626084124.481184816]: setupPublishers...
[ INFO] [1626084124.484863116]: Expected frequency for depth = 30.00000
[ INFO] [1626084124.512936888]: Expected frequency for color = 30.00000
[ INFO] [1626084124.531511673]: setupStreams...
[ INFO] [1626084124.535434556]: insert Depth to Stereo Module
[ INFO] [1626084124.535516328]: insert Color to RGB Camera
[ INFO] [1626084124.553308427]: SELECTED BASE:Depth, 0
[ INFO] [1626084124.563054403]: RealSense Node Is Up!
[ERROR] [1626084124.582088042]: Ignoring transform for child_frame_id "camera_color_frame" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 -0.500000 0.000000 0.000000)
[ WARN] [1626084124.670303541]: 
[ INFO] [1626084125.199959831]: publishing unordered pointcloud.

MartyG-RealSense, 2021-07-12:
The launch file **rs_d435_camera_with_model.launch** has not been updated for years.  If you are aiming to generate a point cloud in the RealSense ROS wrapper with a 400 Series camera (like your D435) then I would recommend using the **rs_camera** or **rs_rgbd** launch files that have been updated much more recently.

Could you post an image of your RViz screen please after having launched with **roslaunch realsense2_camera rs_camera.launch filters:=pointcloud**

closed 2021-07-24 by MartyG-RealSense