realsenseai/realsense-ros #1979: Exception was thrown during user callback

vislero, 2021-07-12:
Hello.

We are testing d-435i with autonomous drone.

In 'void BaseRealSenseNode::frame_callback' function, We added below line to check Laser status in EMITTER ON/OFF mode.

'int emitter_state = (int)frame.get_frame_metadata(RS2_FRAME_METADATA_FRAME_EMITTER_MODE)';

However, I got the runtime error 'ERROR [140615779022592] (source.cpp:130) Exception was thrown during user callback' .

Is there a way to check the laser status?  

thank you.

MartyG-RealSense, 2021-07-12:
Hi @vislero   In the link below, **Doronhi** the RealSense ROS wrapper suggests to a RealSense ROS user who was using emitter on-off about a method to log the emitter status.

https://github.com/IntelRealSense/realsense-ros/issues/1379#issuecomment-698104504

***************

There is currently no way to log emitter status using ROS.  You could do the following.  Add the following line inside the [publishFrame() function:](url)

```
auto projector_state = f.get_frame_metadata(RS2_FRAME_METADATA_FRAME_LASER_POWER_MODE);
ROS_INFO_STREAM("projector_state : " << projector_state);
```

vislero, 2021-07-12:
Thank you for your kind reply.

We followed your instruction as figure below.

[image]

And we got the same error.

By the way, we are testing on the configuration below.

librealsense : v.2.48.0
realsense-ros : v.2.3.0
device firmware : 05.12.14.50

And when we tested the older versions of librealsense (v.2.45.0) and realsense-ros, this error did not occur.

MartyG-RealSense, 2021-07-12:
ROS wrapper 2.3.0 is designed for librealsense 2.45.0.  A wrapper should ideally be matched as closely as possible with the librealsense version listed on a particular wrapper version's release notes.  The ideal wrapper match for librealsense SDK 2.48.0 when using ROS1 is wrapper **2.3.1**. 

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.1

If you wish to stay with 2.45.0 for now, wrapper 2.3.0 is the ideal match for it.

vislero, 2021-07-13:
Thank you.

We updated the version of realsense-ros from 2.3.0 to 2.3.1.
Now the function is works !

However the message "An error has occurred during frame callback" is occur from below code.
https://github.com/IntelRealSense/realsense-ros/blob/09a263a311542b6521f0389490b6df4484b18191/realsense2_camera/src/base_realsense_node.cpp#L1760-L1763

is this normal?

MartyG-RealSense, 2021-07-13:
At the beginning of this case you had added a line to the frame_callback function that was generating an error.  Did you remove that original line after adding the new code to the publishFrame() function please?

https://github.com/IntelRealSense/realsense-ros/issues/1979#issue-941819970

vislero, 2021-07-14:
yes of course. 
the f.get_frame_metadata is only in the publishFrame function.

MartyG-RealSense, 2021-07-14:
If you are using Ubuntu, which **kernel** version are you using, please?

vislero, 2021-07-15:
I am using Ubuntu 18.04 (5.4.0-77-generic)

MartyG-RealSense, 2021-07-15:
Thank you.  Is there any further information that accompanies the **An error has occurred during frame callback** message?  Here are the two most common examples of such messages from roslaunch logs in past cases.

******************

An error has occurred during frame callback: **Error occurred during execution of the processing block!**
An error has occurred during frame callback: **Time is out of dual 32-bit range**

*******************

Are you experiencing either of these messages, please?

vislero, 2021-07-15:
yes. I only get the error message "An error has occurred during frame callback:" like the picture below.

[image]

And, after updating realsense ros version 2.3.0 to 2.3.1, the image is published, but the error message is remained.
When i checked this problem in Jetson-nx-xavier using the same version of SDK and ROS wrapper, the result was the same(same error message).

MartyG-RealSense, 2021-07-15:
As the emitter status polling is working and the checkerboard image is apparently working too, I suspect that these messages are simply harmless timing hiccups where the sensor was not ready at the particular instant that it was polled.  For example, these kind of messages are sometimes seen in RealSense ROS when polling the camera temperature, resulting in an 'Asic Temperature value is not valid!' message, followed by a normal temperature reading.

vislero, 2021-07-16:
Ok, I see.
Thank you so far.

closed 2021-07-16 by vislero