realsenseai/realsense-ros #1982: How to get camera's internal temperature using this package?

JoshuaSiy, 2021-07-15:
Hello,
I would like to inquire if it is possible to obtain the camera's ASIC temperature using this package? I asked because I am unsure where to find the source code where pipelines and configurations are included.  I would also like to ask :: Given that this package is actively running, can we make another program(either in Python or C++)  to obtain the camera's projector/asic temperature?

MartyG-RealSense, 2021-07-15:
Hi @JoshuaSiy   ASIC, Projector and Motion Module temperatures are published by the RealSense ROS wrapper's **/diagnostics** topic.

JoshuaSiy, 2021-07-15:
Oh, we just have to wait a few seconds for it to show up. Thank you very much.

closed 2021-07-15 by JoshuaSiy