realsenseai/realsense-ros #1984: Slam with l515 and t265?

aditdoshi333, 2021-07-16:
Hello,

I am trying to use l515 with rtabmap_ros for the slam. I am happy with the map generation part but the odometry is not up to the mark. I am thinking to add t265 for better tracking but a bit confused as l515 is also with IMU. Will t265 improves the performance?

RealSenseSupport, 2021-08-23:
@aditdoshi333 Do you still need support on this? Looking forward to your update. Thanks!

RealSenseSupport, 2021-09-17:
@aditdoshi333 Could you please update? Please note that the ticket will be closed if we still don't get update from your side in another 7 days. Thanks!

RealSenseSupport, 2021-09-24:
@aditdoshi333 Sorry that we didn't get update from you for weeks and this ticket will be closed at this point. Please feel free to create new ticket if you still have questions or issues. Thanks!

closed 2021-09-24 by RealSenseSupport