realsenseai/realsense-ros #1985: D435i - /dev/video is showing only 3 ports in Ubuntu 18.04

Gowresh-MARC, 2021-07-19:
When I connect D435i to a system running Ubuntu 18.04, it shows only 3 video ports in /dev/video. But in 20.04 its showing all 5 ports.  I need the 4the port to directly access the rgb stream.

MartyG-RealSense, 2021-07-19:
Hi @Gowresh-MARC   A RealSense camera should produce /dev/video 0 to 5.  Are there any error messages in your roslaunch log in 18.04 indicating a problem, such as **Cannot open '/dev/video4  Last Error: No such file or directory** or does the launch seem to complete successfully but the full range of dev/video channels are not available?

Gowresh-MARC, 2021-07-19:
The following was displayed when i plugged in the D435i:-

[ 613.515838] usb 1-4.3: new high-speed USB device number 10 using tegra-xusb
[ 613.536831] usb 1-4.3: New USB device found, idVendor=8086, idProduct=0b3a
[ 613.536841] usb 1-4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 613.536845] usb 1-4.3: Product: Intel(R) RealSense™ Depth Camera 435i
[ 613.536848] usb 1-4.3: Manufacturer: Intel(R) RealSense™ Depth Camera 435i
[ 613.538345] uvcvideo: Found UVC 1.50 device Intel(R) RealSense™ Depth Camera 435i (8086:0b3a)
[ 613.540210] uvcvideo: Unable to create debugfs 1-10 directory.
[ 613.540409] uvcvideo 1-4.3:1.0: Entity type for entity Intel(R) RealSense™ Depth Ca was not initialized!
[ 613.540615] uvcvideo 1-4.3:1.0: Entity type for entity Processing 2 was not initialized!
[ 613.540751] uvcvideo 1-4.3:1.0: Entity type for entity Camera 1 was not initialized!
[ 613.542145] input: Intel(R) RealSense™ Depth Ca as /devices/3610000.xhci/usb1/1-4/1-4.3/1-4.3:1.0/input/input11
[ 613.543147] uvcvideo: Found UVC 1.50 device Intel(R) RealSense™ Depth Camera 435i (8086:0b3a)
[ 613.544361] uvcvideo: Unable to create debugfs 1-10 directory.
[ 613.544518] uvcvideo 1-4.3:1.3: Entity type for entity Processing 7 was not initialized!
[ 613.544691] uvcvideo 1-4.3:1.3: Entity type for entity Extension 8 was not initialized!
[ 613.544824] uvcvideo 1-4.3:1.3: Entity type for entity Camera 6 was not initialized!
[ 613.607869] hid-sensor-hub 0003:8086:0B3A.0006: No report with id 0xffffffff found
[ 613.609461] hid-sensor-hub 0003:8086:0B3A.0006: No report with id 0xffffffff found
[ 613.610458] hid-sensor-hub 0003:8086:0B3A.0006: No report with id 0xffffffff found
[ 613.611373] hid-sensor-hub 0003:8086:0B3A.0006: No report with id 0xffffffff found
[ 614.087855] hid-sensor-hub 0003:8086:0B3A.0006: No report with id 0xffffffff found

Gowresh-MARC, 2021-07-19:
I actually want to directly access /video3

MartyG-RealSense, 2021-07-19:
It looks as though the log was generated using the **dmesg** command.  May I confirm whether you are using librealsense or using ROS (as this message is posted on the RealSense ROS forum).  Thanks!

Gowresh-MARC, 2021-07-19:
I am using librealsense

MartyG-RealSense, 2021-07-19:
Okay, it's fine to discuss librealsense on this forum - librealsense has its own forum at https://github.com/IntelRealSense/librealsense/issues but we can continue here.  Thanks for the confirmation.

What method did you use to install librealsense on your 18.04 computer please - Debian packages or source code?

Gowresh-MARC, 2021-07-19:
Thank you, very much! 
I used Debian packages, i.e I installed the librealsense packages by adding the repo and using apt-get install

MartyG-RealSense, 2021-07-19:
I ask about the build method because I recall a case where /dev/video channels were unavailable after installing from packages.  A reported solution for this was to use a different kernel version.

https://github.com/IntelRealSense/librealsense/issues/7287#issuecomment-708479006

Which kernel versions are you currently using on 18.04 and 20.04?

Gowresh-MARC, 2021-07-19:
18.04 - 4.15.0-147-generic
20.04 - 5.8.0-59-generic

MartyG-RealSense, 2021-07-19:
Which /dev/video ports do you have available?

Gowresh-MARC, 2021-07-19:
/dev/video0 - 2 is only available

MartyG-RealSense, 2021-07-19:
Apparently, a D435i should only produce /dev/video0 to 2, with RGB being mapped onto /video2

https://github.com/IntelRealSense/librealsense/issues/3416#issuecomment-471720605

Gowresh-MARC, 2021-07-19:
But when I use the D435i with the Ubuntu 20,04,  I get 6 ports from /dev/video0 - 5. Also I am not able to access the rgb by /video2 through usb-cam ros package

MartyG-RealSense, 2021-07-19:
And the D435i RGB appears on usb-cam when using 20.04?

Gowresh-MARC, 2021-07-19:
Yes. On /video4

Gowresh-MARC, 2021-07-19:
On ubuntu 20.04 rgb appears on /video4 when launching with usb-cam

MartyG-RealSense, 2021-07-19:
Let's test whether the **udev** device handling rules need reloading.  Please enter the commands below into your Ubuntu terminal.

**sudo udevadm control --reload-rules
sudo udevadm trigger**

Gowresh-MARC, 2021-07-19:
Okay. I did that and now Im getting rgb from /video2. Although only 3 ports are visible. But still Im able to stream RGB from usb-cam. Thanks for the support

MartyG-RealSense, 2021-07-19:
You're very welcome - that's great news!

MartyG-RealSense, 2021-07-25:
Hi @Gowresh-MARC  Do you require further assistance with this case, please?  Thanks!

Gowresh-MARC, 2021-07-25:
@MartyG-RealSense Nope. My issue is solved. Thank you for your support. Will close this issue.

closed 2021-07-25 by Gowresh-MARC