realsenseai/realsense-ros #1987: Add rostopic about realsense in my code

hyebin-rdv, 2021-07-20:
If you do 'roslaunch realsense2_camera rs_camera.launch ' and type 'rostopic list' in, you get this result.
> /camera/color/camera_info
/camera/color/image_raw
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_infra1
/camera/extrinsics/depth_to_infra2
/camera/infra1/camera_info
/camera/infra1/image_rect_raw
/camera/infra2/camera_info
/camera/infra2/image_rect_raw
/camera/gyro/imu_info
/camera/gyro/sample
/camera/accel/imu_info
/camera/accel/sample
/diagnostics

I'm using a realense camera in my own code. And here's the result of my rosbag list.

> /rdvcam0/camera/rgb/image_mask
/rdvcam0/camera/rgb/image_raw
/rdvcam0/rdv_rcnn_topic/pc_all
/rdvcam0/rdv_rcnn_topic/plane_points
/rdvcam0/rdv_rcnn_topic/plane_points_color_mask
/rdvcam0/rdv_rcnn_topic/plane_points_color_mask_filter
/rosout
/rosout_agg

That's the 'image_raw' coming into the camera. I want to save the video, not the image. What shoud I do?

MartyG-RealSense, 2021-07-21:
Hi @hyebin-rdv  Let's handle this question at your existing case at https://github.com/IntelRealSense/librealsense/issues/9423 so that we are not talking in two different locations.  Thanks!

MartyG-RealSense, 2021-07-27:
Case closed due to being handled at https://github.com/IntelRealSense/librealsense/issues/9423

closed 2021-07-27 by MartyG-RealSense