realsenseai/realsense-ros #1990: Launch only 1 ROS node.

kevinlinxc, 2021-07-21:
Using rs_camera.launch, many different topics are started, each presumably taking some compute power to run. I'm wondering if I can modify the launch files to only launch the one node I want. Apologies if this has been asked before.

MartyG-RealSense, 2021-07-21:
Hi @kevinlinxc  I would recommend **unsubscribing** from the topics that you do not want published.  **Doronhi** the RealSense ROS wrapper developer recommends this approach to saving CPU usage too in the link below.

https://github.com/IntelRealSense/realsense-ros/issues/1195

kevinlinxc, 2021-07-25:
Hi @MartyG-RealSense , can you clarify: would this just be any listeners (subscribers) unsubscribing, or is there a way to unsubsribe from publisher side so that the topics are not published at all in `rostopic list`. Is there an example that I could use that uses a launch file?

MartyG-RealSense, 2021-07-25:
My understanding of Doronhi's advice in https://github.com/IntelRealSense/realsense-ros/issues/1195#issuecomment-647939763 is that unsubscribing after launch is considered to be preferable in terms of ensuring stability instead of trying to cut off topics from being published in the first place.

.In regard to the mention in the above linked-to discussion of enable / disable of sensors being unsupported: that is out of date now in ROS2 at least, as a sensor toggle feature was added in ROS2 wrapper **3.1.6**.

https://github.com/IntelRealSense/realsense-ros/releases/tag/3.1.6

kevinlinxc, 2021-07-28:
Unfortunately, I'm still using Ros melodic, C++.  Can you point me to an example of how to unsubscribe after launch?

MartyG-RealSense, 2021-07-29:
I hope that the information resources listed in the link below will be helpful to you in setting up unsubscribing.

https://github.com/IntelRealSense/realsense-ros/issues/1909#issuecomment-852236311

MartyG-RealSense, 2021-08-04:
Hi @kevinlinxc  Do you require further assistance with this case, please?  Thanks!

kevinlinxc, 2021-08-04:
I don't think the C++ node tutorial explains how to make the nodes disappear, as I want. I think maybe the nodes that aren't being subscribed to have minimal impact on cpu and networking anyway

MartyG-RealSense, 2021-08-05:
Thanks for the update.  So do you believe that it may be unnecessary to reduce the number of topics because doing so does not have a noticable impact on processing in your particular project, please?

kevinlinxc, 2021-08-05:
Yes

MartyG-RealSense, 2021-08-05:
Okay, thanks very much @kevinlinxc for the confirmation.  If you feel that you no longer need to continue with this case and are satisifed then you are welcome to close it using the **Close issue** button underneath the comment writing box.  Thanks!

closed 2021-08-11 by kevinlinxc