realsenseai/realsense-ros #1991: D435i not found by this package (No RealSense devices were found! Terminating RealSense Node...), rs-enumerate-devices or realsense-viewer

kevinlinxc, 2021-07-21:
I have purchased a brand new D435i, unboxed today. Preivously, I have used this node on Ubuntu 18.04 LTS to run the D415, where I could see the various video feeds using rqt_image_view. Plugging in this new D435i and running the node yields 
```
 No RealSense devices were found! Terminating RealSense Node...
```
I know that the device is detectable to some degree because `lsusb` shows an Intel device and disappears when I unplug it.
`rs-enumerate-devices` yields `No device detected. Is it plugged in?`

Plugging this the camera into Windows lets the camera set up properly, and running the Window camera app shows the video feed as expected.

Running `realsense-viewer` doesn't detect the camera, and add source does not show the camera as an option. 

What are the troubleshooting steps here? I've looked at 4 or 5 threads with this problem and tried copying some of the solutions but to no avail.

MartyG-RealSense, 2021-07-21:
Hi @kevinlinxc  Let's focus first on getting the camera detected in the RealSense Viewer, as that provides a good foundation for then working on ROS problems once librealsense is performing normally.

Which computer / computing device are you using the camera with please (e.g PC, Raspberry Pi, Nvidia Jetson, etc).

Also, which method did you use to install librealsense?

kevinlinxc, 2021-07-21:
Hi @MartyG-RealSense, I am using a persistent USB containing Ubuntu 18.04 LTS on a Lenovo Thinkpad T480. Second, I'm not sure which method I used sorry, is there a command to run to find out?  I seem to remember struggling for a bit with Method 1 and then being surprised that I had realsense2_camera in catkin_ws/src/external_pkgs/realsense

MartyG-RealSense, 2021-07-21:
In this situation I would recommend installing the librealsense SDK first and then building the RealSense ROS wrapper separately from source code with Method 2 once you have established that librealsense is working correctly.

If you are using a PC and are familiar with installing packages then the Debian package installation method for librealsense in the link below may suit you.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

kevinlinxc, 2021-07-22:
I followed the instructions in the link you sent, specifically from `apt-key` to `modinfo`. apt-key had the error `gpg: keyserver receive failed: No name`, maybe because `keys.gnupg.net` is out of service. After doing the rest of the steps, nothing has changed. One thing I notice is that the steps installed librealsense version ~2.48 but when I run realsense-viewer it says realsenseviewer v2.16.0 and librealsense version: 2.16.0. I previously dealt with having two different versions as I said earlier, is there a way to check which versions I have installed, and what are the steps to proceed if this is in fact the issue?

MartyG-RealSense, 2021-07-22:
It definitely should not be installing librealsense 2.16.0, as that version is severely outdated.  Could you try uninstalling all previously installed librealsense packages by inputting the command below into the Ubuntu terminal, please.

**dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge**

There have been recent issues with the keyserver addresses, described in the case linked to below.  

https://github.com/IntelRealSense/librealsense/issues/9289

One RealSense user in that case found that they could install successfully if they replaced keys.gnupg.net with **keyserver.ubuntu.com**, using the same keyserver address in both halves of the instruction (before and after the '| |').

https://github.com/IntelRealSense/librealsense/issues/9289#issuecomment-873457374

kevinlinxc, 2021-07-25:
Hi, after uninstalling via that command, realsense-viewer still boots as version 2.16.0. When I type `which realsense-viewer` it shows me that it's in my `/usr/local/bin/`, and it was not uninstalled. How should I proceed?

kevinlinxc, 2021-07-25:
I tried deleting the realsense-viewer from my bin and rerunning the installation, but after installing realsense-viewer is not found, though I expected it to be installed with the utils

MartyG-RealSense, 2021-07-25:
I would not expect the built executable realsense-viewer to be affected by a Debian package uninstall command.  I would think that the next time that the librealsense SDK was built then the old realsense-viewer version in /usr/local/bin/ would be overwritten by the build compiled by the new version of the SDK.

The SDK is installed from packages using the commands **sudo apt-get install librealsense2-dkms** and
**sudo apt-get install librealsense2-utils**   ... it is the 'utils' package that should install the examples and tools.

Can you confirm please that you registered the server's public key and added the server to the list of repositories before installing the packages?

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

kevinlinxc, 2021-07-25:
Yes, I have done that. I managed to start realsense-viewer 2.48 using sudo realsense-viewer, which starts from /usr/bin/ instead of just realsense=viewer which starts from /usr/local/bin/. I will report back on the ros launch file

kevinlinxc, 2021-07-25:
I did the same with rs-enumerate-devices, and that works now. When I run the ros node, it states
```[ INFO] [1627237515.310981524]: Initializing nodelet with 8 worker threads.
[ INFO] [1627237515.400752973]: RealSense ROS v2.0.3
[ INFO] [1627237515.400836081]: Running with LibRealSense v2.16.0
```
And fails as before. I need to update the ros portion of realsense it seems. When I run `rospack find realsense2_camera`, I see
`catkin_ws/src/external_pkgs/realsense/realsense2_camera`.

How can I update the ros portion of realsense while keeping the package there?

kevinlinxc, 2021-07-25:
Yes, cloning the realsense-repo anew, catkin_make source devel/setup.bash and this new camera is detected as expected. (old camera also still working)

In summary:

To get rs-enumerate-devices or realsense-viewer to work again, I deleted them from /usr/loca/bin (copied them to temp just in case), and then reinstalled using linux instructions.

To get the ros node working, I realized that my predecessors just cloned the realsense-ros repo into our workspace, and I needed to reclone to be-up-to-date. Thanks for your help @MartyG-RealSense

closed 2021-07-25 by kevinlinxc

MartyG-RealSense, 2021-07-25:
You're very welcome, @kevinlinxc  - it's great news that you achieved solutions with librealsense and the ROS wrapper. :)