realsenseai/realsense-ros #1992: Using RealSense d435i IMU to improove Robot Odometry

BrooklynBoy21, 2021-07-21:
I have a problem using imu from realsense d435i camera. I want to fuse IMU data with wheel odometry on a custom robot with **robot_localization**. 

The problem that IMU frame is not coincident with robot base_footprint. I mean axes rotated and this could cause the problem with localization. I will attach screenshot to help understand the problem.

[image]

MartyG-RealSense, 2021-07-21:
Hi @BrooklynBoy21  Is the information in the link below helpful in answering your question, please?

https://answers.ros.org/question/326273/d435i-realsense-and-robot_localization/

BrooklynBoy21, 2021-07-21:
Hi @MartyG-RealSense no(

BrooklynBoy21, 2021-07-21:
I have tried set param odom_frame_id to my odometry frame, but it didn't help

MartyG-RealSense, 2021-07-21:
There was a user of the RealSense T265 (which has the same IMU component as the D435i) that shared advice on how they linked up to robot_localization

https://github.com/IntelRealSense/realsense-ros/issues/749#issuecomment-843086654

BrooklynBoy21, 2021-07-21:
@MartyG-RealSense I wrote comment there.
But the solution provided in that topic is slightly different problem. I am using IMU, not visual odometry.

MartyG-RealSense, 2021-07-21:
The subject of wheel odometry is honestly one that I am not familiar with, so the advice that I can provide on the subject is limited, regretfully.  The research paper in the link below, which discusses D435i and wheel odometry wit ROS, may be worth looking at though.

https://www.researchgate.net/publication/348413335_Survey_and_Experimental_Comparison_of_RGB-D_Indoor_Robot_Navigation_Methods_Supported_by_ROS_and_Their_Expansion_via_Fusion_with_Wheel_Odometry_and_IMU_Data

Apologies that I cannot be of more help on this particular occasion.

MartyG-RealSense, 2021-08-02:
Hi @BrooklynBoy21  Do you require further assistance with this case, please?  Thanks!

BrooklynBoy21, 2021-08-02:
@MartyG-RealSense Hi, I postponed a little bit this problem and work on another things right now. 
I will try some other solutions, so for now I think the issue can be closed.

Thank you

closed 2021-08-02 by BrooklynBoy21

MartyG-RealSense, 2021-08-02:
You are very welcome @BrooklynBoy21 - thanks for the update, and please do feel free to create a new case on this forum when you are ready to resume work on your D435i odometry project if you have further questions.  Good luck!