realsenseai/realsense-ros #1994: ©rostopic list

aatb-ch, 2021-07-21:


MartyG-RealSense, 2021-07-22:
Hi @aatb-ch  Could you provide more information about what you want to know please?  Thanks!

As you mention in the title, you can obtain a full list of the topics supported on a particular RealSense camera model by inputting the command **rostopic list** into your ROS terminal.  For example, the RealSense ROS wrapper documentation has a list of some of the topics published by a **D435i** camera model.

https://github.com/IntelRealSense/realsense-ros#published-topics

aatb-ch, 2021-07-22:
haha, I'm so sorry I typed on the wrong keyboard while setting up my Jetson Nano while reading issues.. you can delete this one!

closed 2021-07-22 by aatb-ch

MartyG-RealSense, 2021-07-22:
No problem at all - thanks for the update   :)