realsenseai/realsense-ros #1996: Camera detected as connected to usb 2.1

sofiane-K, 2021-07-22:
**Hardware**
Realsense d435 camera
origin cable from realsense
Firmware version 5.12.10
**logs**

PARAMETERS
 * /camera/front/decimation/filter_magnitude: 2
 * /camera/front/realsense2_camera/accel_fps: 63
 * /camera/front/realsense2_camera/accel_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/accel_optical_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/align_depth: False
 * /camera/front/realsense2_camera/aligned_depth_to_color_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/aligned_depth_to_fisheye1_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/aligned_depth_to_fisheye2_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/aligned_depth_to_fisheye_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/aligned_depth_to_infra1_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/aligned_depth_to_infra2_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/allow_no_texture_points: False
 * /camera/front/realsense2_camera/base_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/calib_odom_file: 
 * /camera/front/realsense2_camera/clip_distance: 3.3
 * /camera/front/realsense2_camera/color_fps: 15
 * /camera/front/realsense2_camera/color_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/color_height: 480
 * /camera/front/realsense2_camera/color_optical_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/color_width: 640
 * /camera/front/realsense2_camera/confidence_fps: 15
 * /camera/front/realsense2_camera/confidence_height: 480
 * /camera/front/realsense2_camera/confidence_width: 640
 * /camera/front/realsense2_camera/depth_fps: 15
 * /camera/front/realsense2_camera/depth_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/depth_height: 480
 * /camera/front/realsense2_camera/depth_optical_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/depth_width: 640
 * /camera/front/realsense2_camera/device_type: 
 * /camera/front/realsense2_camera/enable_accel: True
 * /camera/front/realsense2_camera/enable_color: True
 * /camera/front/realsense2_camera/enable_confidence: True
 * /camera/front/realsense2_camera/enable_depth: True
 * /camera/front/realsense2_camera/enable_fisheye1: False
 * /camera/front/realsense2_camera/enable_fisheye2: False
 * /camera/front/realsense2_camera/enable_fisheye: False
 * /camera/front/realsense2_camera/enable_gyro: True
 * /camera/front/realsense2_camera/enable_infra1: True
 * /camera/front/realsense2_camera/enable_infra2: True
 * /camera/front/realsense2_camera/enable_infra: False
 * /camera/front/realsense2_camera/enable_pointcloud: True
 * /camera/front/realsense2_camera/enable_pose: False
 * /camera/front/realsense2_camera/enable_sync: False
 * /camera/front/realsense2_camera/filters: decimation,spatia...
 * /camera/front/realsense2_camera/fisheye1_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/fisheye1_optical_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/fisheye2_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/fisheye2_optical_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/fisheye_fps: 15
 * /camera/front/realsense2_camera/fisheye_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/fisheye_height: 480
 * /camera/front/realsense2_camera/fisheye_optical_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/fisheye_width: 640
 * /camera/front/realsense2_camera/gyro_fps: 200
 * /camera/front/realsense2_camera/gyro_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/gyro_optical_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/imu_optical_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/infra1_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/infra1_optical_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/infra2_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/infra2_optical_frame_id: ilogistics_3_0010...
 * /camera/front/realsense2_camera/infra_fps: 15
 * /camera/front/realsense2_camera/infra_height: 480
[... 75 lines cut ...]
[ INFO] [1626968306.212373559]: Expected frequency for infra1 = 15.00000
[ INFO] [1626968306.238721295]: Expected frequency for color = 15.00000
[ INFO] [1626968306.270832906]: setupStreams...
[ INFO] [1626968306.271576437]: insert Depth to Stereo Module
[ INFO] [1626968306.271646324]: insert Color to RGB Camera
[ INFO] [1626968306.271690405]: insert Infrared to Stereo Module
[ INFO] [1626968306.271737922]: insert Gyro to Motion Module
[ INFO] [1626968306.271782272]: insert Accel to Motion Module
[ INFO] [1626968306.342238090]: SELECTED BASE:Depth, 0
[ INFO] [1626968306.358734381]: RealSense Node Is Up!
[ WARN] [1626968306.478290408]: 
[ WARN] [1626968309.191396721]: Hardware Notification:USB SCP overflow,1.62697e+12,Error,Hardware Error
[ WARN] [1626968311.202987899]: Hardware Notification:USB SCP overflow,1.62697e+12,Error,Hardware Error
 22/07 15:38:31,286 WARNING [139704993302272] (backend-v4l2.cpp:1122) Frames didn't arrived within 5 seconds
 22/07 15:38:31,294 WARNING [139704981812992] (backend-v4l2.cpp:1122) Frames didn't arrived within 5 seconds
 22/07 15:38:31,295 WARNING [139704973420288] (backend-v4l2.cpp:1122) Frames didn't arrived within 5 seconds
[ WARN] [1626968313.214674894]: Hardware Notification:USB SCP overflow,1.62697e+12,Error,Hardware Error
[ WARN] [1626968315.226337416]: Hardware Notification:USB SCP overflow,1.62697e+12,Error,Hardware Error

**Issue**

Using a usb 3.2 port realsense camera is detected as using a 2.1 port and this is created an overflow error I guess . If I unplug and replug the camera from the laptop side it disappears and no more overflow (frames arrive normally), if I unplug and replug the camera from the camera side it doesn't (frames don't arrive).

Thank you for your reply and your expertise,
Sofiane

MartyG-RealSense, 2021-07-22:
Hi @sofiane-K   Could you try adding the command **initial_reset:=true** to your roslaunch instruction please to perform an automated reset of the camera at launch instead of having to unplug-replug the USB cable and see whether it makes a positive difference to how the camera is identified.  For example:

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**

sofiane-K, 2021-07-23:
Hello @MartyG-RealSense, I just tried and it didn't solved the issue. I still get the CAM overflow and REC overflow errors.

sofiane-K, 2021-07-23:
After checking the kernel logs, I remark there are some issues happening. 

256198.207313] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[256198.250107] hid-sensor-hub 0003:8086:0B3A.0003: No report with id 0xffffffff found
[256198.250110] hid-sensor-hub 0003:8086:0B3A.0003: No report with id 0xffffffff found
[256198.454115] TCP: request_sock_TCP: Possible SYN flooding on port 33207. Sending cookies.  Check SNMP counters.
[256198.605419] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[256198.656164] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[256198.711336] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[256199.180763] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[256199.235017] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[256199.285640] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[256199.337029] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[256199.388763] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[256199.441017] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).

I notably notice that a possible flooding is logged. Does it bring more clues about what is currently happening ?

MartyG-RealSense, 2021-07-23:
A good debugging rule is to start with the first error, in case subsequent errors are false readings caused by a cascade of glitches set in motion by the earlier error.  Regarding **uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024)**, there is a past case in the link below that discusses it.

https://support.intelrealsense.com/hc/en-us/community/posts/360037100614-D435-Linux-Driver-Message-Issues

MartyG-RealSense, 2021-07-29:
Hi @sofiane-K  Do you require further assistance with this case, please?  Thanks!

sofiane-K, 2021-07-29:
Hello @MartyG-RealSense , since I am not able to reproduce the issue (it doesn't happen all the time), I don't know how to be sure that it will not happen again. I cannot  change the kernel that we use, but I tried external powered usb hub. The problem is that the problem itself is not happening a lot

MartyG-RealSense, 2021-07-29:
The computer that the camera is being attached to can also be a factor to consider when a USB glitch occurs (for example, a problem with the computer's USB port that comes and goes unpredictably).  Do you have another computer that you can test with to see whether reliability improves further, please?

closed 2021-08-10 by MartyG-RealSense