realsenseai/realsense-ros #1997: unable to install realsenseSDK on windows-ROS

05anu, 2021-07-25:
facing bellow error while executing the command :
**catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release**

Error:
NMAKE : fatal error U1077: & #39;&quot;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\bin\HostX64\x64\nmake.exe&quot;& #39; :
return code & #39;0x2& #39;
Invoking &quot;nmake&quot; failed

kindly suggest the solution for above error

MartyG-RealSense, 2021-07-25:
Hi @05anu  The link below has a past case of the nmake error when installing RealSense ROS on Windows using the command **catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release**

https://github.com/IntelRealSense/realsense-ros/issues/1520#issuecomment-736220023

A workaround that worked for that particular user was to use the following command:

**rosdep install --from-paths src --ignore-src -r -y**

05anu, 2021-07-25:
ok.. can you please tell me that how to use this command... 
instead of using   **catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release ** 
I should use this command or what? request you to tell me the exact process....

MartyG-RealSense, 2021-07-25:
My understanding is that you navigate to the catkin workspace folder with **cd ~/catkin_ws/src/** and then run the rosdep command from that folder:

**rosdep install --from-paths src --ignore-src -r -y**

closed 2021-08-07 by MartyG-RealSense

MartyG-RealSense, 2021-08-23:
Did you put a space between install and --from please?

**rosdep install --from-paths src --ignore-src -r -y**

05anu, 2021-08-24:
yes

MartyG-RealSense, 2021-08-24:
Are you running the rosdep command whilst in the catkin workspace folder **/catkin_ws**