realsenseai/realsense-ros #1998: rqt reconfigure crashes on filter configuration tab

daniele-gusmini, 2021-07-26:
With decimation filter active, on selection of decimation tab in dynamic reconfigure rqt crashes, reporting:

Floating point exception (core dumped)

MartyG-RealSense, 2021-07-26:
Hi @daniele-gusmini   Are you able to set the decimation filter's magnitude with a **rosrun dynamic_reconfigure** command after the roslaunch instruction instead of using the decimation option on the dynamic_reconfigure GUI, please?

**rosrun dynamic_reconfigure dynparam set camera/decimation 'filter_magnitude' 8**

[image]

If you would prefer to solve the issue with the GUI crash, there was a similar problem in 2019 in the case below that was diagnosed to be caused if the kernel was unpatched.

https://github.com/IntelRealSense/realsense-ros/issues/607

MartyG-RealSense, 2021-08-02:
Hi @daniele-gusmini  Do you require further assistance with this case, please?  Thanks!

daniele-gusmini, 2021-08-02:
Hi @MartyG-RealSense, thanks for the help! Your method works, I found also an alternative setting up the parameter in the launch file as:

<rosparam>
      /camera/decimation/filter_magnitude: 2
</rosparam>

The GUI crash probably is due to the unpatched kernel as you said

Thank you for the help!

MartyG-RealSense, 2021-08-02:
You are very welcome, @daniele-gusmini - it's excellent news that you were successful.  Thanks for the update and for sharing your method with the RealSense ROS community!

closed 2021-08-08 by MartyG-RealSense