realsenseai/realsense-ros #1999: tf error when running multiple l515 using realsense ros and rtabmap ros

aditdoshi333, 2021-07-27:
Hello,

For better FOV I am using two l515 for mapping using rtabmap. 
Launch file: 
[two_l515.txt](https://github.com/IntelRealSense/realsense-ros/files/6883883/two_l515.txt)

I am able to run the while with two live devices. But If I record .bag file for both the camera and run the launch file I am getting the following error:

`[ WARN] [1627372329.172096917]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom \
   /camera1/rgbd_image \
   /camera2/rgbd_image
[ WARN] [1627372330.020367078]: /rtabmap/rtabmapviz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmapviz subscribed to (approx sync):
   /rtabmap/odom \
   /camera1/rgbd_image \
   /camera2/rgbd_image \
   /rtabmap/odom_info
[ WARN] [1627372333.748568798]: /rtabmap/rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. 
/rtabmap/rgbd_odometry subscribed to (approx sync):
   /camera1/rgbd_image \
   /camera2/rgbd_image
`

I recorded .bag file keeping l515 90 degree apart. I am using two different laptop for recording .bag file

MartyG-RealSense, 2021-07-27:
Hi @aditdoshi333  Which launch instruction are you using to play back the recorded bag, please?  For example, is it **roslaunch realsense2_camera rs_from_file.launch**

aditdoshi333, 2021-07-27:
For camera 1 (terminal 1)
`rosbag play front.bag /camera/aligned_depth_to_color/image_raw:=/camera1/aligned_depth_to_color/image_raw /camera/color/camera_info:=camera1/color/camera_info /camera/color/image_raw:=/camera1/color/image_raw /camera/realsense2_camera_manager/bond:=/camera1/realsense2_camera_manager/bond
`

For camera 2 (terminal 2)
`rosbag play side.bag /camera/aligned_depth_to_color/image_raw:=/camera2/aligned_depth_to_color/image_raw /camera/color/camera_info:=camera2/color/camera_info /camera/color/image_raw:=/camera2/color/image_raw /camera/realsense2_camera_manager/bond:=/camera2/realsense2_camera_manager/bond
`
rtabmap node (terminal 3)

roslaunch two_l515.launch

MartyG-RealSense, 2021-07-27:
It sounds as though you are playing the rosbags with **rosbag play** in order to publish their topic contents and then running **RTABMAP** and launching your custom RealSense ROS launch file **two_l515.launch** in order to listen to the bag topics with RTABMAP and perform mapping with the bag data.  Is that correct, please?

If this is correct then I wonder whether the problem is that **use_sim_time** needs to be defined as True, like in the extract below from the bag play section of Intel's D435i SLAM guide.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

[image]

aditdoshi333, 2021-07-27:
Yes, correct I am playing two .bag files and **two_l515.launch** contains launch command for rtabmap ros. I tried setting use_sim_time as True but still the error is same.

MartyG-RealSense, 2021-07-27:
If **two_l515.launch** is looking for two camera serial numbers but you are using two different laptops with one camera attached to each then I'm not certain how the cameras would be able to be found.

For example' Intel's tutorial of using three RealSense cameras spread over two computers uses **ssh** to define connections between the computers and **ROS_MASTER_URI** to communicate with roscore.

https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines

Thanks very much for your patience during this case!

aditdoshi333, 2021-07-27:
The serial number I would need for live devices. I am not getting any errors with live realsense devices. The above error I am getting for recorded bag files from two different devices. The bag files are stored on the same computer only. If you want to reproduce the error I can share bag files also.

MartyG-RealSense, 2021-07-27:
Is it a requirement of the project to have two separate bag files?  If **rosbag record --all** was used to record all currently active topics (including the topics from both live cameras) into a single accumulated rosbag file then that would simplify playback.

http://wiki.ros.org/rosbag/Tutorials/Recording%20and%20playing%20back%20data#Recording_all_published_topics

MartyG-RealSense, 2021-08-02:
Hi @aditdoshi333  Do you require further assistance with this case, please?  Thanks!

aditdoshi333, 2021-08-02:
No, Thanks

closed 2021-08-02 by aditdoshi333