realsenseai/realsense-ros #2001: ROS2 same frame_ids for multiple cameras

Marenix, 2021-07-27:
Hey, I have a question regarding frame_ids in ROS2. I might've missed or misinterpreted something, so I'd appreciate any explanation/help. 

I've been testing both ROS and ROS2 wrappers and I've noticed that they have different naming convention for frame_ids. Let's take color_frame_id for example. If we connect two cameras and start rs_multiple_devices.launch, one camera will have color_frame_id=camera1_color_frame while the other will have color_frame_id=camera2_color_frame. On the other hand, using ROS2 to launch rs_multi_camera_launch.py, both camera's will have color_frame_id=camera_color_frame.

My question is why? Is there some reason why this was changed in ROS2?
Thanks for the help!

brennand, 2021-07-30:
Looking at the code, the `base_realsense_node.cpp` ignores the "camera_name" param.  If you change the camera name, the frame_id in the header doesn't change.

I think we need to patch this to support multi Realsense's.

Marenix, 2021-07-30:
Yea, I agree. I looked through it too, and saw that frame ID names are made in the following way: camera_STREAM_NAME()_frame which actually makes same named streams of different camera all "blend" with that one name. In ROS wrapper this is solved in launch file/nodelet.launch.xml file where each parameter name is defined as namespace+(rest of the name). This is "convenient" in ROS because in nodelet.launch.xml you can put the prefix of namespace as the parameter that is inputted as namespace for each node. There isn't such an option in ROS2, at least not that I've found, and so, in ROS2, you would have to change the name of each frame by hand. 

I think it would be a good idea to change `base_realsense_node.cpp`, to change the definitions of `FRAME_ID()` `OPTICAL_FRAME_ID` and `ALIGNED_DEPTH_TO_FRAME_ID` so that they don't start with "camera_" but rather the `namespace` of the camera

brennand, 2021-07-30:
I got it working.  Not really a nice solution.  I added all the frame_id's to rs_camera.launch.py  And then my main launch file passed my new frame ID's as params.:

    {"name": "base_frame_id", "default": "camera_link", "description": ""},
    {"name": "depth_frame_id", "default": "camera_depth_frame", "description": ""},
    {"name": "color_frame_id", "default": "camera_color_frame", "description": ""},

eta

Marenix, 2021-08-02:
I made a simple change in base_realsense_node.cpp from:
[image]
to:
[image]
and just wrote namespace in string variable _namespace in BaseRealSenseNode constructor. This resulted in me getting frame IDs like: camera1_color_frame, camera2_color_frame, camera1_depth_frame... except for base_frame_id which was still the same for both camera's, being camera_link.

Your solution works, and I've been doing a similar thing so far. The problem is you need to rename all the frame_ids for each camera you're using, and it's more work the more cameras you have. That's why, I think, something like this would be pretty nice. Having the frame_ids be linked with namespace would remove the need to manually rename all the ids.

Marenix, 2021-08-12:
@doronhi Can you tell us if there is a reason why this was implemented the way it was?