realsenseai/realsense-ros #2002: Cmake cannot find "realsense2"

konnoeric, 2021-07-28:
I installed SDK2.0 and it works well since I can run realsense-viewer. So installing SDK2.0 should be no problem according to the screen shot below.
[image]

I am using Ubuntu18.04 and ROS melodic but could not `catkin_make clear` and the error message below was shown. What should I do with this issue?

> -- +++ processing catkin package: 'realsense2_camera'
-- ==> add_subdirectory(realsense-ros/realsense2_camera)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Warning at realsense-ros/realsense2_camera/CMakeLists.txt:44 (find_package):
  By not providing "Findrealsense2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "realsense2", but CMake did not find one.
Could not find a package configuration file provided by "realsense2"
 (requested version 2.48.0) with any of the following names:
    realsense2Config.cmake
    realsense2-config.cmake
Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.
CMake Error at realsense-ros/realsense2_camera/CMakeLists.txt:48 (message):
     Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases
-- Configuring incomplete, errors occurred!
See also "/home/project/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/project/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

MartyG-RealSense, 2021-07-28:
Hi @konnoeric  May I first confirm that you used **catkin_make clean** and not 'make clear' please?  Thanks!

konnoeric, 2021-07-28:
Yes I am sorry it was my type miss. I did catkin_make clean.

konnoeric, 2021-07-29:
Thank you very much. Yes, this is the exact process I have done. 

And when I did `catkin_init_workspace` at Step4, I encountered error code saying that I have done this process before. But I have to see and check again when I come back home because I am not sure.

Is this the actual problem? Because the error message was shown when I executed `catkin_make clear`, which you can see at the very top of this conversation, saying that I need to pass something to Cmakefile. 

So do I actually have to execute `catkin_init_workspace` without any error message? because 

> It will create the CMakeLists.txt file (actually it just create a symbolic link in the src folder of your workspace pointing to CMakeLists.txt -> /opt/ros/groovy/share/catkin/cmake/toplevel.cmake in the case you're using groovy).

according to [this question to ROS answers.](https://answers.ros.org/question/187084/what-does-the-command-catkin_init_workspace-do/)

MartyG-RealSense, 2021-07-29:
It is necessary to use **catkin_init_workspace**

Some RealSense users have had problems in the past with the **git checkout** command that comes immediately before the catkin_init_workspace command though.  Another way to input this command is to use a simpler version based on the ROS wrapper version number.  For example, if ROS wrapper 2.3.0 was being used then you would use this git checkout command:

**git checkout 2.3.0**

konnoeric, 2021-07-30:
Sorry I found that there is an error when I execute `git checkout git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1` .

error code:
> error: pathspec 'git' did not match any file(s) known to git.
error: pathspec 'tag' did not match any file(s) known to git.

So as you said, I should execute `git checkout 2.3.0 git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1` is it correct?

Though, I dont know how to see the version of the ROS wrapper I got so would you please teach me?

MartyG-RealSense, 2021-07-30:
You would only need to do **git checkout <version number>** - for example, git checkout 2.3.0 if you are building ROS wrapper 2.3.0 from source code.  Change '2.3.0' to whatever the version number is of the wrapper that you are installing.

In regard to the git pathspec error, apparently there are quotation marks surrounding the command that did not get displayed for some reason when I listed the instruction earlier.  Sorry about that.  It should have looked like this:

`git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1``

konnoeric, 2021-07-31:
Thank you very much. I got this message when I typed in 

git checkout  `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1`
※sorry I dont know how to correctly show the command including quotation mark but I did with those quotation mark.

message after executing the command above:
> Note: checking out '2.3.1'.
You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.
If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:
  git checkout -b <new-branch-name>
HEAD is now at 09a263a 2.3.1

So is this ok to move on next ?

MartyG-RealSense, 2021-07-31:
As far as I am aware, the **You are in 'detached HEAD' state** message is not something to be concerned about.

There is the option of deleting the whole **catkin_ws** catkin workspace folder and beginning the ROS wrapper installation process from the beginning if you feel that there is a problem.  Before doing so though, it may be worth skipping to the next step after **catkin_init_workspace** to see whether you are now able to successfully initiate the instruction catkin_make clean

**cd ..
catkin_make clean
catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
catkin_make install**

konnoeric, 2021-08-01:
Thank you. Actually I tried executing until `catkin_make install ` and I am now able to visualise images on Rviz. 

But I cannot see any pointclouds. Is there anything I should care before visualising pointclouds? If needed, I can provide screen shots of my screen after I come back home.

MartyG-RealSense, 2021-08-02:
What method are you using to vsualize a pointcloud in RViz, please?  Is it like **roslaunch realsense2_camera rs_camera.launch filters:=pointcloud**

https://github.com/IntelRealSense/realsense-ros#point-cloud

MartyG-RealSense, 2021-08-08:
Hi @konnoeric  Do you require further assistance with this case, please?  Thanks!

konnoeric, 2021-08-08:
Sorry for being late. I was quite busy these days. So I was just using `roslaunch realsense2_camera rs_camera.launch`, no filters option.

MartyG-RealSense, 2021-08-09:
Thanks very much for the information.  Regarding your earlier statement "I cannot see any pointclouds.  Is there anything I should care before visualising pointclouds?" ... a **pointcloud** filter needs to be defined in the launch in order for a pointcloud to be rendered by the RealSense ROS wrapper, like the process described in https://github.com/IntelRealSense/realsense-ros#point-cloud

MartyG-RealSense, 2021-08-15:
Hi @konnoeric  Do you have an update about this case that you can provide, please?  Thanks!

tky1026, 2021-08-16:
Assuming you install from [Linux Debian Installation Guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages).

Probably would require to install `librealsense2-dev` also, by doing `sudo apt-get install librealsense2-dev`.

This fix my issue `Could not find a package configuration file provided by "realsense2"` when `catkin_make clean`.

MartyG-RealSense, 2021-08-16:
Thanks very much @tky1026 for your advice to @konnoeric   :)

konnoeric, 2021-08-16:
Thank you very much @tky1026 , @MartyG-RealSense . I would like to try your advices and will tell you the results by this weekend.

konnoeric, 2021-08-22:
So was able to view pointclouds by adding option you mentioned. Thank you very much for your kind corporation. It's all ok. You can close this case.

MartyG-RealSense, 2021-08-22:
Great to hear that you were successful, @konnoeric - thanks very much for the update!  As suggested, I will close the case.

closed 2021-08-22 by MartyG-RealSense