realsenseai/realsense-ros #2004: xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device

fiorano10, 2021-07-28:
We have multiple cameras connected to the motherboard, each has its own USB 3.1 Gen 1 port. We randomly see the No such device error when the node is launched. We confirmed that all realsenses are connected as USB 3.2 devices.

Running the following config 
Ubuntu 18.04
ROS Melodic
Librealsense v2.42 
realsense-ros  v2.2.22
Camera fw: 5.12.11

Logs ([camera1-realsense2_camera1-1-stdout.log](https://github.com/IntelRealSense/realsense-ros/files/6896252/camera1-realsense2_camera1-1-stdout.log))
```
[0m[ INFO] [1627477677.310092974]: SELECTED BASE:Depth, 0[0m
[0m[ INFO] [1627477677.318674069]: RealSense Node Is Up![0m
 28/07 09:08:02,913 ERROR [140134976579328] (types.h:315) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 28/07 09:08:02,913 ERROR [140134976579328] (error-handling.cpp:76) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 28/07 09:08:03,339 ERROR [140134783645440] (types.h:315) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 28/07 09:08:03,339 ERROR [140134783645440] (types.h:315) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 28/07 09:08:03,348 ERROR [140134951401216] (types.h:315) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 28/07 09:08:03,349 ERROR [140134951401216] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 28/07 09:08:03,913 ERROR [140134976579328] (types.h:315) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 28/07 09:08:03,913 ERROR [140134976579328] (error-handling.cpp:76) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
```

MartyG-RealSense, 2021-07-29:
Hi @fiorano10  How many RealSense cameras do you have attached, please?  I found serials for two cameras in the log.  Also, what was the roslaunch instruction that you used?

The text doc version of the launch log has so many **xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device** messages generated after the **RealSense Node is Up!** notification that it suggests a serious communication problem.

fiorano10, 2021-07-29:
We have 4 cameras connected. We use a single launch file to launch all cameras and read the serial no. from a yaml file, along the params you see in the logs.

MartyG-RealSense, 2021-07-29:
Does the launch perform any better if you use the launch file **rs_multiple_devices.launch** to define the cameras and their serial numbers in the roslaunch instruction instead of using a yaml file?

https://github.com/IntelRealSense/realsense-ros#work-with-multiple-cameras

fiorano10, 2021-07-29:
It does not, why would having the serial no in a launch file make a difference?

MartyG-RealSense, 2021-07-29:
Having the serial number defined in the launch instruction simplifies the launch process by having one less factor (the yaml file access) to go wrong.

fiorano10, 2021-07-29:
How do we resolve the no such device error? Do we reset the camera when this happens?

MartyG-RealSense, 2021-07-30:
**xioctl(UVCIOC_CTRL_QUERY)** typically indicates a conflict between librealsense and the kernel.  This is a difficult problem to resolve when you cannot use RSUSB due to the need to support a multi-camera setup.  Which kernel version are you using, please?

fiorano10, 2021-07-30:
We're on kernel `5.4.0-65-generic`

MartyG-RealSense, 2021-07-30:
I cannot find a history of problems with RealSense projects that use kernel 5.4.0-65-generic and Ubuntu 18.04, though that does not eliminate the possibility of a conflict in your particular project.

The discussions in the links below look at a range of ways to deal with non-detection fo a camera after boot, such as resetting the entire USB port instead of just the camera, or booting with **rtcwake**

https://github.com/IntelRealSense/librealsense/issues/1615
https://github.com/IntelRealSense/librealsense/issues/1615#issuecomment-501050619
https://github.com/IntelRealSense/librealsense/issues/5913

closed 2021-08-11 by MartyG-RealSense