realsenseai/realsense-ros #2005: get_dev_capabilities: Cannot open '/dev/video27 Last Error: Permission denied

fiorano10, 2021-07-29:
We have multiple cameras connected to the motherboard, each has its own USB 3.1 Gen 1 port. The camera launched fine initially but then stopped with the following errors:
```
get_dev_capabilities: Cannot open '/dev/video"n" Last Error: Permission denied
Asic Temperature value is not valid
```

Running the following config
Ubuntu 18.04
ROS Melodic
Librealsense v2.42
realsense-ros v2.2.22
Camera fw: 5.12.11

[camera4-realsense2_camera4-7-stdout.log](https://github.com/IntelRealSense/realsense-ros/files/6897525/camera4-realsense2_camera4-7-stdout.log)

MartyG-RealSense, 2021-07-29:
Hi @fiorano10  Do the errors still occur if you set the FPS to 15 instead of 6, please?

fiorano10, 2021-07-29:
Why would setting the FPS to 15 matter? Also, I can't really test that since we have the cameras installed on a product that is being used commerically

MartyG-RealSense, 2021-07-29:
In the past, there have been cases where using an FPS of '6' has caused errors that do not occur at '15'.  It is thought that when 6 FPS is being used, frames may be arriving slowly enough that the communications time-out while waiting for the next frame.

closed 2021-08-11 by MartyG-RealSense